Koenkk/zigbee2mqtt #10361: [New device support]: Schneider Electric / ELKO smart pir with dimmer

pab32, 2021-12-25:
### Link

https://proff.elko.no/smartpir-med-dimmer/smartpir-dimmer-rs-ph-article3250-3067.html

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x680ae2fffe106261","nwkAddr":11455,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"NHMOTION/DIMMER/1","epList":[3,37,242],"endpoints":{"3":{"profId":260,"epId":3,"devId":257,"inClusterList":[0,3,4,5,6,8,769,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"NHMOTION/DIMMER/1","manufacturerName":"Schneider Electric","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"","swBuildId":"002.005.024 R"}}},"binds":[],"configuredReportings":[],"meta":{}},"37":{"profId":260,"epId":37,"devId":2080,"inClusterList":[0,3,1024,1030,2821,65305],"outClusterList":[3,6,8,1030],"clusters":{"genBasic":{"attributes":{"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"","swBuildId":"002.005.024 R","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1640472446054,"useImplicitCheckin":true}

### Comments

I have paired the device and it show up as unsupported. As described I have added a *.js file, but I´m stuck when I tro to find what I need in the from Zigbee, toZigbee and exposes fields. fz.occupanzy/e.occupancy show up as ready, but everything else give me an error.

pab32, 2021-12-27:
fromZigbee: fz.on_off
esposes: e.switch()
Switching lights on/off is the only thing working and what i get out of it in addition to linkquality

pab32, 2022-01-27:
💤

pab32, 2022-02-02:
This is the same:
https://github.com/Koenkk/zigbee2mqtt/issues/10988

closed 2022-02-02 by pab32

michaelthunemberglund, 2022-04-17:
@pab32 
Hi there. 10988 is not the same as this. There is two models, one is a switch and the other is a dimmer. 
I have the dimmer version and is "unsupported" in Zigbee2mqtt GUI. I have not managed to make a external converter that is working with Elko smart PIR Dimmer.

fillwe, 2022-04-18:
Following this topic, have ordered a couple of these myself.

michaelthunemberglund, 2022-07-16:
Anyone who is able to look into this?

prinsmikal, 2022-12-22:
has someone found a solution?

torroa, 2022-12-28:
The LK FUGA Wiser wireless PIR with Dimmer (model 545D6111) is working just fine in ZHA, and partly in deCONZ. There shouldn't be a reason why Z2M is lagging behind...

torroa, 2023-03-25:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['NHMOTION/DIMMER/1'],
        model: 'EKO06985',
        vendor: 'Elko',
        description: 'Elko plus - SmartPir dimmer',
        fromZigbee: [fz.on_off, fz.brightness, fz.level_config, fz.schneider_lighting_ballast_configuration, fz.command_recall,
            fz.command_on, fz.command_off, fz.command_move, fz.command_stop],
        toZigbee: [tz.light_onoff_brightness, tz.level_config, tz.ballast_config, tz.schneider_dimmer_mode],
        endpoint: (device) => {
            return {'l1': 3, 's1': 21, 's2': 22, 's3': 23, 's4': 24};
        },
        meta: {multiEndpoint: true},
        exposes: [e.light_brightness().withLevelConfig().withEndpoint('l1'),
            exposes.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
                .withDescription('Specifies the minimum light output of the ballast')
                .withEndpoint('l1'),
            exposes.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
                .withDescription('Specifies the maximum light output of the ballast')
                .withEndpoint('l1'),
            exposes.enum('dimmer_mode', ea.ALL, ['RC', 'RL']).withDescription('Controls Capacitive or Inductive Dimming Mode')
                .withEndpoint('l1'),
            e.action(['on', 'off', 'brightness_move_up', 'brightness_move_down', 'brightness_stop', 'recall_*'])],
        configure: async (device, coordinatorEndpoint, logger) => {
            // Configure the dimmer actuator endpoint
            await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
            const endpoint = device.getEndpoint(3);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingBallastCfg']);
            await reporting.onOff(endpoint);
            await reporting.brightness(endpoint);
            // Configure the four front switches
            device.endpoints.forEach(async (ep) => {
                if (21 <= ep.ID && ep.ID <= 22) {
                    await reporting.bind(ep, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
                } else if (23 <= ep.ID && ep.ID <= 24) {
                    await reporting.bind(ep, coordinatorEndpoint, ['genScenes']);
                }
            });
        },
        onEvent: async (type, data, device) => {
            // Record the factory default bindings for easy removal/change after deviceInterview
            if (type === 'deviceInterview') {
                const dimmer = device.getEndpoint(3);
                device.endpoints.forEach(async (ep) => {
                    if (21 <= ep.ID && ep.ID <= 22) {
                        ep.addBinding('genOnOff', dimmer);
                        ep.addBinding('genLevelCtrl', dimmer);
                    }
                    if (23 <= ep.ID && ep.ID <= 24) {
                        ep.addBinding('genScenes', dimmer);
                    }
                });
            }
        },
            };

module.exports = definition;
`

torroa, 2023-03-25:
Here is an external converter that seems to work:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['NHMOTION/DIMMER/1'],
        model: 'EKO06985',
        vendor: 'Elko',
        description: 'Elko plus - SmartPir dimmer',
        fromZigbee: [fz.on_off, fz.brightness, fz.level_config, fz.schneider_lighting_ballast_configuration, fz.command_recall,
            fz.command_on, fz.command_off, fz.command_move, fz.command_stop],
        toZigbee: [tz.light_onoff_brightness, tz.level_config, tz.ballast_config, tz.schneider_dimmer_mode],
        endpoint: (device) => {
            return {'l1': 3, 's1': 21, 's2': 22, 's3': 23, 's4': 24};
        },
        meta: {multiEndpoint: true},
        exposes: [e.light_brightness().withLevelConfig().withEndpoint('l1'),
            exposes.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
                .withDescription('Specifies the minimum light output of the ballast')
                .withEndpoint('l1'),
            exposes.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
                .withDescription('Specifies the maximum light output of the ballast')
                .withEndpoint('l1'),
            exposes.enum('dimmer_mode', ea.ALL, ['RC', 'RL']).withDescription('Controls Capacitive or Inductive Dimming Mode')
                .withEndpoint('l1'),
            e.action(['on', 'off', 'brightness_move_up', 'brightness_move_down', 'brightness_stop', 'recall_*'])],
        configure: async (device, coordinatorEndpoint, logger) => {
            // Configure the dimmer actuator endpoint
            await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
            const endpoint = device.getEndpoint(3);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingBallastCfg']);
            await reporting.onOff(endpoint);
            await reporting.brightness(endpoint);
            // Configure the four front switches
            device.endpoints.forEach(async (ep) => {
                if (21 <= ep.ID && ep.ID <= 22) {
                    await reporting.bind(ep, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
                } else if (23 <= ep.ID && ep.ID <= 24) {
                    await reporting.bind(ep, coordinatorEndpoint, ['genScenes']);
                }
            });
        },
        onEvent: async (type, data, device) => {
            // Record the factory default bindings for easy removal/change after deviceInterview
            if (type === 'deviceInterview') {
                const dimmer = device.getEndpoint(3);
                device.endpoints.forEach(async (ep) => {
                    if (21 <= ep.ID && ep.ID <= 22) {
                        ep.addBinding('genOnOff', dimmer);
                        ep.addBinding('genLevelCtrl', dimmer);
                    }
                    if (23 <= ep.ID && ep.ID <= 24) {
                        ep.addBinding('genScenes', dimmer);
                    }
                });
            }
        },
            };

module.exports = definition;

torroa, 2023-04-26:
Try this external converter.

It does not include luminance or ocupancy sensor.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
zigbeeModel: ['NHMOTION/DIMMER/1'],
model: 'EKO06985',
vendor: 'Elko',
description: 'Elko plus - SmartPir dimmer',
fromZigbee: [fz.on_off, fz.brightness, fz.level_config, fz.schneider_lighting_ballast_configuration, fz.command_recall,
fz.command_on, fz.command_off, fz.command_move, fz.command_stop],
toZigbee: [tz.light_onoff_brightness, tz.level_config, tz.ballast_config, tz.schneider_dimmer_mode],
endpoint: (device) => {
return {'l1': 3, 's1': 21, 's2': 22, 's3': 23, 's4': 24};
},
meta: {multiEndpoint: true},
exposes: [e.light_brightness().withLevelConfig().withEndpoint('l1'),
exposes.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
.withDescription('Specifies the minimum light output of the ballast')
.withEndpoint('l1'),
exposes.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
.withDescription('Specifies the maximum light output of the ballast')
.withEndpoint('l1'),
exposes.enum('dimmer_mode', ea.ALL, ['RC', 'RL']).withDescription('Controls Capacitive or Inductive Dimming Mode')
.withEndpoint('l1'),
e.action(['on', 'off', 'brightness_move_up', 'brightness_move_down', 'brightness_stop', 'recall_*'])],
configure: async (device, coordinatorEndpoint, logger) => {
// Configure the dimmer actuator endpoint
await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
const endpoint = device.getEndpoint(3);
await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingBallastCfg']);
await reporting.onOff(endpoint);
await reporting.brightness(endpoint);
// Configure the four front switches
device.endpoints.forEach(async (ep) => {
if (21 <= ep.ID && ep.ID <= 22) {
await reporting.bind(ep, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
} else if (23 <= ep.ID && ep.ID <= 24) {
await reporting.bind(ep, coordinatorEndpoint, ['genScenes']);
}
});
},
onEvent: async (type, data, device) => {
// Record the factory default bindings for easy removal/change after deviceInterview
if (type === 'deviceInterview') {
const dimmer = device.getEndpoint(3);
device.endpoints.forEach(async (ep) => {
if (21 <= ep.ID && ep.ID <= 22) {
ep.addBinding('genOnOff', dimmer);
ep.addBinding('genLevelCtrl', dimmer);
}
if (23 <= ep.ID && ep.ID <= 24) {
ep.addBinding('genScenes', dimmer);
}
});
}
},
};

module.exports = definition;`

enoch85, 2023-10-18:
@torroa  Did you figure this one out?

I have the DIMMER version as well, and the SWITCH version doesn't work for me - it complains that L1 doesn't exist. I'm using this specific device: https://www.elbutik.se/product.html/schneider-exxact-wiser-zigbee-rorelsevakt-dimmer

With your config it turns on and off, but that's it. I also get errors in the log like:

```
Zigbee2MQTT:info  2023-10-18 10:30:58: Configuring 'LED Tak garderob HALL'
Zigbee2MQTT:error 2023-10-18 10:30:58: Failed to configure 'LED Tak garderob HALL', attempt 3 (TypeError: Cannot read properties of undefined (reading 'bind')
    at Object.bind (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:44:24)
    at Object.configure (/usr/lib/zigbee2mqtt/data/extension/externally-loaded.js:27:25)
    at Configure.configure (/usr/lib/zigbee2mqtt/lib/extension/configure.ts:117:37)
    at EventEmitter.<anonymous> (/usr/lib/zigbee2mqtt/lib/extension/configure.ts:82:62)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitLastSeenChanged (/usr/lib/zigbee2mqtt/lib/eventBus.ts:52:22)
    at Controller.<anonymous> (/usr/lib/zigbee2mqtt/lib/zigbee.ts:68:27)
    at Controller.emit (node:events:517:28)
    at Controller.selfAndDeviceEmit (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:429:14)
    at Controller.onZclOrRawData (/usr/lib/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:611:18))
Zigbee2MQTT:info  2023-10-18 10:30:58: MQTT publish: topic 'zigbee2mqtt/LED Tak garderob HALL', payload '{"illuminance":null,"illuminance_lux":null,"linkquality":0,"occupancy":null,"occupancy_timeout":null,"state_l1":"OFF"}'
```

Any advice would be great!

skjalggo, 2024-03-26:
Anyone got this working with PIR detection? I tried using the external converters shown above, but can't get the pir detection from the device. any help would be apriciated.