Koenkk/zigbee2mqtt #10364: [New device support]: Climax PRL_00.00.03.04TC Relay

callifo, 2021-12-26:
### Link

https://www.climax.com.tw/new/prl-1zbs.php

### Database entry

{"id":37,"type":"EndDevice","ieeeAddr":"0x00124b001c0cfb2c","nwkAddr":15009,"manufId":0,"manufName":"ClimaxTechnology","powerSource":"Unknown","modelId":"PRL_00.00.03.04TC","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"PRL_00.00.03.04TC","manufacturerName":"ClimaxTechnology","powerSource":0,"zclVersion":1,"appVersion":1,"hwVersion":0,"dateCode":"20161222        "}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"hwVersion":0,"dateCode":"20161222        ","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1640498417225,"useImplicitCheckin":true}

### Comments

Tried to add this new relay through a converter but not had much luck understanding the from/to/exposes part, just seem to kill my zigbee2mqtt completely,

This unit is re-badged by a number of suppliers, mine was sold by Vesta Australia however the unit that arrived is badged as Yale. I obtained the developer manual from the supplier if this is of use.

[image]

Koenkk, 2021-12-26:
Since this seems to be just a simple relay, the following external converter should do it:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['PRL_00.00.03.04TC'],
    model: 'myModel',
    vendor: 'Climax',
    description: 'My super switch!',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

callifo, 2021-12-26:
Thanks, that seems to be working well.

The power status of the device seems to show up as 'Unknown', whereas my Ikea Signal Repeaters which are also mains power, seem to show 'Mains (Single Phase)' and show a plug in Zigbee2mqtt HMI. 

I dont see anything specific in the zigbee-herdsman-converters for the Ikea Signal Repeater for whether it is mains or battery, so is that based on something being sent by the device and therefore the relay not sending something correctly?

callifo, 2021-12-26:
This seems to have fixed it,

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['PRL_00.00.03.04TC'],
    model: 'PRL-1ZBS-12/24V',
    vendor: 'Climax',
    description: 'ZigBee 12-24V Relay Controller',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
};

module.exports = definition;
```

Koenkk, 2021-12-27:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-27 by Koenkk