Koenkk/zigbee2mqtt #10385: [Problem]: TuYa TS011F_plug_1 vs TS011F_plug_3

PM04290, 2021-12-27:
### What happened?

i have 2 kind of TS011F smartplug with 2 kind of pairing : TS011F_plug_1 , TS011F_plug_3
i need to reporting data (seMetering) on each with the same configuration

TS011F_plug_1 (prise_mesure_2) have debug log with :type '**attributeReport**', cluster 'seMetering', data '{"currentSummDelivered":[0,2213]}' from endpoint 1 with groupID 0
**every some minutes** and 100w consuption

TS011F_plug_3 (prise_mesure_3) have debug log with : type '**readResponse**', cluster 'seMetering', data '{"currentSummDelivered":[0,1113]}' from endpoint 1 with groupID 0
**every 10/15 second** without consuption
[image]

### What did you expect to happen?

if i remember well, plug_1 reporting configuration is fully with default values (at pairing time)
[image]
instead of plug_3 was paired with blank configuration; i fill the configuration with some of plug_1 values.
[image]

### How to reproduce it (minimal and precise)

i don't want to change plug_1 default values because i want to keep a plug with a good running.
i try to change plug_3 value but it keep the same running

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20190608

### Adapter

CC2531 zStack12

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7779708/log.txt)

Koenkk, 2021-12-27:
Plugs identified as `TS011F_plug_3` do not support reporting and therefore rely on polling (https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#options).

Whether a TS011F plug is detected as 1 or 3 depends on the `appVersion` (fw version), 69, 68, 65, 64 will be identified as 3, others as 1. 69 is the last known version, downgrading is not possible only upgrading so if your plug is identified as 3 it is not possible to go to 1

PM04290, 2021-12-27:
@Koenkk,  you mean the old version knew how to report and the new one couldn't. strange choice on the part of the developers.
by the way, can you confirm that in my configuration above, plug_3 should only send me a measurement every 5 min, except it does it every 10/15 sec (without consuption)

Koenkk, 2021-12-27:
Depends on the configured `measurement_poll_interval` (default is 60): https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#options

closed 2022-01-24 by PM04290