Koenkk/zigbee2mqtt #10386: [New device support]: Bosch Smart Home Zwischenstecker Kompakt (Smart Plug)

Purii, 2021-12-27:
### Link

https://www.bosch-smarthome.com/de/de/service/hilfe/hilfe-zum-produkt/hilfe-zum-zwischenstecker-kompakt/

### Database entry

{"id":18,"type":"Unknown","ieeeAddr":"0x847127fffe99db9b","nwkAddr":10776,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1640603294509,"useImplicitCheckin":true}

### Comments

Seems to fail on step 1. I'm unable to successfully pair the device initially.
According to the documentation of the device, it supports Zigbee 3.0.

I reset and tried to pair the plug right next to the Zigbee stick.

Do your have any ideas?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
  // Didn't receive a model id, because it's unable to pair
    zigbeeModel: ['bsh.bsp-fz2'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'BSP-FZ2', // Vendor model number, look on the device for a model number
    vendor: 'Bosch Smart Home', // Vendor of the device (only used for documentation and startup logging)
    description: 'Zwischenstecker Kompakt', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off], // We will add this later
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.switch()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

Koenkk, 2021-12-27:
Could you provide the herdman debug log when trying to pair the device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Purii, 2021-12-29:
Hi @Koenkk! Is this sufficient?
I ran it on my prod system, so there might be logs about other devices as well.
Hope I sliced the relevant parts for you.

`ieeeAddr` is `0x847127fffe99db9b`.

```
2021-12-29T17:57:49.549Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-29T17:57:49.552Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,0,4,97,9,2,1,1,60,0,131,160,104,0,0,8,24,106,10,0,0,33,31,19,153,20,28,12]
2021-12-29T17:57:49.552Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,0,4,97,9,2,1,1,60,0,131,160,104,0,0,8,24,106,10,0,0,33,31,19,153,20,28,12]
2021-12-29T17:57:49.552Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,0,4,97,9,2,1,1,60,0,131,160,104,0,0,8,24,106,10,0,0,33,31,19,153,20,28] - 12
2021-12-29T17:57:49.552Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1024,"srcaddr":2401,"srcendpoint":2,"dstendpoint":1,"wasbroadcast":1,"linkquality":60,"securityuse":0,"timestamp":6856835,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,106,10,0,0,33,31,19]}}
2021-12-29T17:57:49.552Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":106,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":33,"attrData":4895}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":2401,"endpoint":2,"linkquality":60,"groupID":0,"wasBroadcast":true,"destinationEndpoint":1}'
2021-12-29T17:57:49.553Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-29T17:57:50.987Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,238,142,155,219,153,254,255,39,113,132,226,21,30]
2021-12-29T17:57:50.987Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,238,142,155,219,153,254,255,39,113,132,226,21,30]
2021-12-29T17:57:50.988Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [238,142,155,219,153,254,255,39,113,132,226,21] - 30
2021-12-29T17:57:50.988Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":36590,"extaddr":"0x847127fffe99db9b","parentaddr":5602}
2021-12-29T17:57:50.988Z zigbee-herdsman:controller:log Device '0x847127fffe99db9b' joined
2021-12-29T17:57:50.988Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-29T17:57:50.988Z zigbee-herdsman:controller:log Device '0x847127fffe99db9b' accepted by handler
2021-12-29T17:57:50.988Z zigbee-herdsman:controller:log Device '0x847127fffe99db9b' is already in database with different networkAddress, updating networkAddress
2021-12-29T17:57:50.990Z zigbee-herdsman:controller:log Not interviewing '0x847127fffe99db9b', completed 'false', in progress 'true'
2021-12-29T17:57:53.409Z zigbee-herdsman:controller:device:log Interview - quirks check for 'undefined'-'undefined'-'Unknown'
2021-12-29T17:57:53.409Z zigbee-herdsman:controller:device:log Interview - quirks did not match
2021-12-29T17:57:53.409Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x847127fffe99db9b', attempt 1
2021-12-29T17:57:53.410Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":36590,"nwkaddrofinterest":36590}
2021-12-29T17:57:53.410Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,238,142,238,142,35]
2021-12-29T17:57:53.425Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2021-12-29T17:57:53.425Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2021-12-29T17:57:53.425Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2021-12-29T17:57:53.425Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2021-12-29T17:57:53.425Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-29T17:58:03.427Z zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '36590' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry
2021-12-29T17:58:03.427Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 36590
2021-12-29T17:58:03.427Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":36590,"options":0,"radius":30}
2021-12-29T17:58:03.427Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,238,142,0,30,26]
2021-12-29T17:58:03.430Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33]
2021-12-29T17:58:03.430Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,69,0,33]
2021-12-29T17:58:03.430Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33
2021-12-29T17:58:03.430Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}
2021-12-29T17:58:03.430Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-29T17:58:06.431Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":36590,"nwkaddrofinterest":36590}
2021-12-29T17:58:06.431Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,238,142,238,142,35]
2021-12-29T17:58:06.441Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2021-12-29T17:58:06.441Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
[... cut ...]
2021-12-29T17:58:30.926Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00158d000758a664:7562/1 (0,0,2)
2021-12-29T17:58:30.926Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":7562,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":90,"options":0,"radius":30,"len":4,"data":{"type":"Buffer","data":[25,30,2,152]}}
2021-12-29T17:58:30.927Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,14,36,1,138,29,1,1,25,0,90,0,30,4,25,30,2,152,120]
2021-12-29T17:58:30.927Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

Koenkk, 2021-12-29:
This is not all the interview logging, I need the logging starting from the point where you put the device in pairing mode until z2m logs that the interview failed

Purii, 2021-12-29:
Like this?

```
[34mZigbee2MQTT:debug[39m 2021-12-29 19:45:58: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"40txo-1","value":true}'
[32mZigbee2MQTT:info [39m 2021-12-29 19:45:58: Zigbee: allowing new devices to join.
2021-12-29T18:45:58.648Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
2021-12-29T18:45:58.648Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
2021-12-29T18:45:58.664Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82]
2021-12-29T18:45:58.664Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82]
2021-12-29T18:45:58.664Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82
2021-12-29T18:45:58.664Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
2021-12-29T18:45:58.665Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-29T18:45:58.665Z zigbee-herdsman:adapter:zStack:znp:SREQ --> UTIL - ledControl - {"ledid":3,"mode":1}
2021-12-29T18:45:58.665Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,2,39,10,3,1,45]
2021-12-29T18:45:58.665Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240]
2021-12-29T18:45:58.666Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
2021-12-29T18:45:58.666Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240
2021-12-29T18:45:58.666Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
2021-12-29T18:45:58.666Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-29T18:45:58.668Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,103,10,0,108]
2021-12-29T18:45:58.668Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,103,10,0,108]
2021-12-29T18:45:58.668Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 7 - 10 - [0] - 108
2021-12-29T18:45:58.668Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- UTIL - ledControl - {"status":0}
2021-12-29T18:45:58.668Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-29T18:45:58.668Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":6,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,7,2,11,254,0]}}
2021-12-29T18:45:58.668Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,6,0,30,6,0,25,7,2,11,254,0,234]
2021-12-29T18:45:58.686Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2021-12-29T18:45:58.686Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2021-12-29T18:45:58.686Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2021-12-29T18:45:58.686Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
2021-12-29T18:45:58.686Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-29T18:45:58.690Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,6,51]
2021-12-29T18:45:58.690Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,6,51]
2021-12-29T18:45:58.690Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,6] - 51
2021-12-29T18:45:58.690Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":6}
2021-12-29T18:45:58.690Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[32mZigbee2MQTT:info [39m 2021-12-29 19:45:58: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"40txo-1"}'
2021-12-29T18:46:03.228Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,150,196,155,219,153,254,255,39,113,132,0,0,219]
2021-12-29T18:46:03.228Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,150,196,155,219,153,254,255,39,113,132,0,0,219]
2021-12-29T18:46:03.229Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [150,196,155,219,153,254,255,39,113,132,0,0] - 219
2021-12-29T18:46:03.229Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":50326,"extaddr":"0x847127fffe99db9b","parentaddr":0}
2021-12-29T18:46:03.229Z zigbee-herdsman:controller:log Device '0x847127fffe99db9b' joined
2021-12-29T18:46:03.229Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-29T18:46:03.229Z zigbee-herdsman:controller:log Device '0x847127fffe99db9b' accepted by handler
2021-12-29T18:46:03.229Z zigbee-herdsman:controller:log New device '0x847127fffe99db9b' joined
2021-12-29T18:46:03.229Z zigbee-herdsman:controller:log Creating device '0x847127fffe99db9b'
[32mZigbee2MQTT:info [39m 2021-12-29 19:46:03: Device '0x847127fffe99db9b' joined
[32mZigbee2MQTT:info [39m 2021-12-29 19:46:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x847127fffe99db9b","ieee_address":"0x847127fffe99db9b"},"type":"device_joined"}'
2021-12-29T18:46:03.333Z zigbee-herdsman:controller:log Interview '0x847127fffe99db9b' start
[32mZigbee2MQTT:info [39m 2021-12-29 19:46:03: Starting interview of '0x847127fffe99db9b'
[... cut ...]
2021-12-29T18:48:50.907Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-12-29T18:48:50.907Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-12-29T18:48:50.908Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-12-29T18:48:50.908Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-12-29T18:48:50.908Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-12-29T18:48

Koenkk, 2021-12-29:
for some reason the plug is constantly rejoining `Device '0x847127fffe99db9b' joined`, not sure why this happens. Do you maybe have the original gateway and can take a sniff when pairing it to that and z2m? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

Purii, 2021-12-30:
Thanks! I will give it a try, but need a few days

BjoernPetersen, 2022-01-30:
Not stale enough for my tastes

Purii, 2022-01-30:
Yep. Sorry for the delay here. Still organizing a second Zigbee Stick to flash

Purii, 2022-03-06:
I won't make it within the next weeks... feel free to take over anyone who likes

BjoernPetersen, 2022-04-07:
Bad bot! Still relevant.

BjoernPetersen, 2022-05-08:
Activity simulation.

closed 2022-06-17 by github-actions[bot]

danieledwardgeorgehitchcock, 2022-09-09:
You might want to check out the activity that has been going on in #13520 - There is now support for adding install codes which should get you passed the interview stage so that you can get a proper DB entry

on-jz, 2022-10-08:
You might want to check my comment on [#13069 ](https://github.com/Koenkk/zigbee2mqtt/issues/13069#issuecomment-1272267851) - The new "add install code" feature did the trick for me. I also added an "External converter" there.