Koenkk/zigbee2mqtt #10387: [Problem]: Z2M bridge stops communicating spontaneously. No connection to any device anymore.

KaHeMu, 2021-12-27:
### What happened?

After several days or weeks of continuous work without any issues Z2M suddenly stops communicating with any device. While the Z2M container still is running it logs the following error:
 
```
debug 2021-12-25 14:42:55: Error: Command 0x5c0272fffe70c472/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
```
This happens for all >50 devices in my system. I can still communicate with Z2M via MQTT and the error occurs with all devices. 

The system can be easily recovered by simply stopping and re-starting the Z2M container.  
(I'm using the docker setup from c't magazine [https://github.com/ct-Open-Source/ct-Smart-Home] )
I remember I had a similar issue isolated to Z2M some weeks ago when I had to power cycle the RasPi because docker stop and docker start did not resolve the issue. 
Any idea?   

### What did you expect to happen?

Normal operation to continue

### How to reproduce it (minimal and precise)

Unfortunately I don't know

### Zigbee2MQTT version

1.22.1 commit: c2b5229

### Adapter firmware version

zStack3x0  coordinator revision 20200715

### Adapter

Slaesh's CC2652RB

### Debug log

```
debug 2021-12-25 14:42:50: Received MQTT message on 'zigbee2mqtt/TF_LED_KUECHE_WINDOW1/set' with data '{"state":"ON"}'
debug 2021-12-25 14:42:50: Publishing 'set' 'state' to 'TF_LED_KUECHE_WINDOW1'
debug 2021-12-25 14:42:50: Received MQTT message on 'zigbee2mqtt/TF_LED_KUECHE_WINDOW2/set' with data '{"state":"ON"}'
debug 2021-12-25 14:42:50: Publishing 'set' 'state' to 'TF_LED_KUECHE_WINDOW2'
debug 2021-12-25 14:42:50: Received MQTT message on 'zigbee2mqtt/TF_LED_KUECHE_WINDOW3/set' with data '{"state":"ON"}'
debug 2021-12-25 14:42:50: Publishing 'set' 'state' to 'TF_LED_KUECHE_WINDOW3'
**** HERE THE ERROR OCCURED AT FIRST. FROM HERE NO COMMUNICATION WITH ANY DEVICE POSSIBLE ANYMORE ****
error 2021-12-25 14:42:55: Publish 'set' 'state' to 'TF_LED_KUECHE_USCHRANK1' failed: 'Error: Command 0x5c0272fffe70c472/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
debug 2021-12-25 14:42:55: Error: Command 0x5c0272fffe70c472/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
info  2021-12-25 14:42:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'TF_LED_KUECHE_USCHRANK1' failed: 'Error: Command 0x5c0272fffe70c472/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"TF_LED_KUECHE_USCHRANK1"},"type":"zigbee_publish_error"}'
error 2021-12-25 14:43:01: Publish 'set' 'state' to 'OSRAM_STD_KUECHE_REGAL' failed: 'Error: Command 0x7cb03eaa00ae6b57/3 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
debug 2021-12-25 14:43:01: Error: Command 0x7cb03eaa00ae6b57/3 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
info  2021-12-25 14:43:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'OSRAM_STD_KUECHE_REGAL' failed: 'Error: Command 0x7cb03eaa00ae6b57/3 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"OSRAM_STD_KUECHE_REGAL"},"type":"zigbee_publish_error"}'
error 2021-12-25 14:43:07: Publish 'set' 'state' to 'TF_LED_KUECHE_WINDOW1' failed: 'Error: Command 0x680ae2fffe83a11c/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
debug 2021-12-25 14:43:07: Error: Command 0x680ae2fffe83a11c/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
info  2021-12-25 14:43:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'TF_LED_KUECHE_WINDOW1' failed: 'Error: Command 0x680ae2fffe83a11c/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"TF_LED_KUECHE_WINDOW1"},"type":"zigbee_publish_error"}'
error 2021-12-25 14:43:13: Publish 'set' 'state' to 'TF_LED_KUECHE_WINDOW2' failed: 'Error: Command 0x680ae2fffe965a58/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
debug 2021-12-25 14:43:13: Error: Command 0x680ae2fffe965a58/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
[... 17 lines cut ...]
debug 2021-12-25 14:45:55: Publishing 'set' 'state' to 'Garage'
debug 2021-12-25 14:45:55: Received MQTT message on 'zigbee2mqtt/Terrasse/l3/set' with data '{"state":"OFF"}'
debug 2021-12-25 14:45:55: Publishing 'set' 'state' to 'Terrasse'
debug 2021-12-25 14:45:55: Received MQTT message on 'zigbee2mqtt/LED_WaLe_Terrasse/set' with data '{"state":"OFF"}'
debug 2021-12-25 14:45:55: Publishing 'set' 'state' to 'LED_WaLe_Terrasse'
error 2021-12-25 14:46:01: Publish 'set' 'state' to 'Garage' failed: 'Error: Command 0x00124b001f319f02/2 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0," [... cut ...]
debug 2021-12-25 14:46:13: Error: Command 0x000d6ffffe44d911/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)

```
**** and so on ...

kpcz, 2021-12-27:
Same issue here, after a while network goes down. Was stable before. 
Current version: 1.22.1-1
stick: cc2652rb_stick_-_slaesh
FW version: 20210708, 'type': 'zStack3x0'
installation: Home Assistant on Rpi4, zigbee stick via 1,5m usb extension cord. 

60 zigbee devices, from which 9 router device (mostly ikea bulbs)

(before had this issue https://github.com/Koenkk/zigbee2mqtt/issues/8711  but FW uprade helped). Since 12. sept was working wihtout issues (v 1.21.1-1), I think month ago started to develop network freezes again, not so often, could be twice a month.

After restarting zigbee2mqtt HA addon, everything goes back to normal.

`error 2021-12-27 16:13:15: Publish 'set' 'state' to '0x680ae2fffe54a35e' failed: 'Error: Command 0x680ae2fffe54a35e/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
info  2021-12-27 16:13:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x680ae2fffe54a35e' failed: 'Error: Command 0x680ae2fffe54a35e/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"0x680ae2fffe54a35e"},"type":"zigbee_publish_error"}'
error 2021-12-27 16:13:23: Publish 'set' 'brightness' to '0x680ae2fffe54a35e' failed: 'Error: Command 0x680ae2fffe54a35e/1 genLevelCtrl.moveToLevelWithOnOff({"level":13,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
info  2021-12-27 16:13:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to '0x680ae2fffe54a35e' failed: 'Error: Command 0x680ae2fffe54a35e/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":13,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"0x680ae2fffe54a35e"},"type":"zigbee_publish_error"}'
error 2021-12-27 16:16:00: Publish 'set' 'state' to '0x680ae2fffe54a35e' failed: 'Error: Command 0x680ae2fffe54a35e/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'`

kpcz, 2021-12-27:
@KaHeMu Do you have any routers (IKEA?)  in your network, if yes what are they?

KaHeMu, 2021-12-27:
@kpcz Yes, I have a number of IKEA Tradfri devices in my network. Among them are the following routers: 
LED1545G12
LED1650R5
LED1623G12
T1828 
ICPSHC24-10EU-IL-1 (2x)
LED1837R5 (3x)

KaHeMu, 2021-12-29:
I followed @Koenkk 's recommendation and flashed the stick with new FW Z-Stack_3.x.0 revision 20211217. I will monitor and report if the error re-occurs or close the issue if not. Thx!

KaHeMu, 2022-01-29:
The error did not re-occur for 1 month since I upgraded to  FW Z-Stack_3.x.0 revision 20211217. I will close the issue. Thanks!

KaHeMu, 2022-01-29:
resolved

closed 2022-01-29 by KaHeMu