Koenkk/zigbee2mqtt #10391: [Problem]: Pairing with Hue Discover impossible

ultimate-tester, 2021-12-27:
### What happened?

I am in the process of migrating from Homey to Home Assistant with Zigbee2MQTT. I have 2 Zigbee devices, the Philips outdoor motion sensor and the Philips Hue Discover floodlight. The outdoor motion sensor connected flawlessly, it has a dedicated reset button at the back. Pressing it was enough to connect it to Zigbee2MQTT.

Now the problems started with the Discover floodlight. I didn't have a hub nor remote, so I went to buy one. Got back, successfully reset the lamp (flashes a few times, then stays on solid) but nothing happens. While I was resetting the lamp, I noticed that Zigbee2MQTT has picked up on the remote and connected to that instead. I resetted it 10+ times, and also removed the power from the device at least 3 times after resetting. It just doesn't want to connect in Zigbee2MQTT and there is no logs whatsoever.

While giving up on connecting, I wanted to setup my backup solution which is to use the remote to turn off the lamp at least. Pairing the remote was also impossible, until I removed and blocked the remote from Zigbee2MQTT. After that, I reset the lamp and was able to connect the remote (without even removing the power). Now the remote works like it should.

### What did you expect to happen?

I expected Zigbee2MQTT to successfully pair to the Philips Hue Discover after resetting it or provide me with appropriate logs.

### How to reproduce it (minimal and precise)

Have a Philips Hue Discover connected to another Zigbee network, reset it using remote, try to pair.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

26720700

### Adapter

ConBee 2

### Debug log

N/A

Koenkk, 2021-12-29:
Could you try to pair the device close to the coordinator?

If that doesn't help and nothing is logged, I'm afraid you need to sniff to figure out what goes wrong: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

daballiemo, 2021-12-31:
I have a Philips Hue Discover that paired without any problems (removed it from the Hue bridge and popped up in Z2M)

```Friendly name
dakterras buitenlamp
Last seen
N/A
avaliability.avaliability
Disabled
Device type
Router
Zigbee Model
1743530P7
Zigbee Manufacturer
Philips
Description
Hue Discover white and color ambiance flood light
Support status
Supported
IEEE Address
0x0017880106xxxyyy
Network address
0xB918
Firmware build date
20210331
Firmware version
1.88.1
Manufacturer
Philips
Model
17435/30/P7
Power
Interview completed
True```

I am using a Sonoff USB 3.0 dongle. 
Maybe you can have the Hue Discover pair to the Hue hub again; remove it from the Hue Hub and then permit join on Z2T?

kr Han

ultimate-tester, 2022-01-06:
Sorry to leave the issue open for so long. I am going to agree that this is probably related to distance. Unfortunately I do not have a way to decrease the distance currently. My plan is to add more Zigbee devices to create a stable path to the Philips hue light. It probably will work then.

closed 2022-01-06 by ultimate-tester