Koenkk/zigbee2mqtt #10396: [Problem]: IKEA lights Brightness:1 behaviour

jeroenvansch, 2021-12-27:
### What happened?

When Brightness of an IKEA light (panel or bulb) is set to 1, the light will turn off and/or wont turn on after being set to off and back to on. (depends on firmware/bulbtype) the latest floalt firmware will result in being turned off.

### What did you expect to happen?

I expected that the lights would go to the lowest brightness without turning off.

### How to reproduce it (minimal and precise)

When sending `{"state": "ON", "brightness": 1}`to an IKEA light the light will turn off and return something like:`{"brightness":1,"color_mode":"color_temp","color_temp":373,"last_seen":"2021-12-27T21:06:00.228Z","linkquality":84,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":false}
`

When I do the same to a Hue light it stays on and returns: `{"brightness":2,"last_seen":"2021-12-27T21:06:49.195Z","linkquality":105,"state":"ON","update":{"state":"idle"},"update_available":false}`

Normally you just wouldn't send this command and be done with it, but you also run into this issue when using "brightness_move".
When using "brightness_move" down until the light stops dimming brightness is also set to 1.
Only when setting the brightness to 2 or higher the lights will turn ON.

Is it possible to get IKEA lights to behave similarly to Hue lights?

### Zigbee2MQTT version

1.22.1-dev commit: 79e5f08

### Adapter firmware version

20201026

### Adapter

zzh!

### Debug log

This is one cycle of using a 5button remote to brightness move down until on the lowest setting, then setting state to OFF, then to ON. (lights are off now) then setting the state to on with brightness 2 (lights are now on)
(I use the longpress on the center button to send the `{"state": "ON", "brightness": 2}` command.)

Received Zigbee message from 'TRADFRI 5B keuken', type 'commandMove', cluster 'genLevelCtrl', data '{"movemode":1,"rate":84}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/keuken/set' with data '{"brightness_move": -40}'
Publishing 'set' 'brightness_move' to 'keuken'
Received Zigbee message from 'TRADFRI 5B keuken', type 'commandStop', cluster 'genLevelCtrl', data '{}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/keuken/set' with data '{"brightness_move": 0}'
Publishing 'set' 'brightness_move' to 'keuken'
Received Zigbee message from 'FLOALT 90 keuken', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Received Zigbee message from 'FLOALT 90 keuken', type 'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":1}' from endpoint 1 with groupID 0
Received Zigbee message from 'TRADFRI 5B keuken', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/keuken/set' with data '{"state": "OFF"}'
Publishing 'set' 'state' to 'keuken'
Received Zigbee message from 'TRADFRI 5B keuken', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/keuken/set' with data '{"state": "ON"}'
Publishing 'set' 'state' to 'keuken'
Received Zigbee message from 'TRADFRI 5B keuken', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/keuken/set' with data '{"state": "OFF"}'
Publishing 'set' 'state' to 'keuken'
Received Zigbee message from 'TRADFRI 5B keuken', type 'commandTradfriArrowRelease', cluster 'genScenes', data '{"value":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'TRADFRI 5B keuken', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":254,"transtime":0}' from endpoint 1 with groupID 0
Received MQTT message on 'zigbee2mqtt/keuken/set' with data '{"state": "ON", "brightness": 3}'
Publishing 'set' 'brightness' to 'keuken'
Received Zigbee message from 'TRADFRI 5B keuken', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":2,"value2":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'TRADFRI 5B keuken', type 'read', cluster 'genBasic', data '["appVersion"]' from endpoint 1 with groupID 0

paulz-gh, 2021-12-29:
Hi,
I noticed the same issue after running the OTA update to firmware 2.3.087 yesterday.  Before this update they reacted as expected.
I'm on Z2M  1.22.0.

jeroenvansch, 2022-01-03:
With the older firmware you only have this issue when you dim the light using brightness_move and then turn them off an on. The light will then stay off. 
With the newer firmware when you set brightness to 1 (method doesn't matter) they will turn off directly without turning off/on.
So the newer firmware has brought a change, but the issue was already there, just not as obvious.

closed 2022-02-10 by github-actions[bot]