Koenkk/zigbee2mqtt #10397: TUYA TS0202 is recognized as unsupported

marcinwrobel, 2021-12-27:
### What happened?

It does not recognize which supported device

### What did you expect to happen?

He will recognize correctly

### How to reproduce it (minimal and precise)

No description

### Zigbee2MQTT version

 1.22.1 

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Debug log

Zigbee2MQTT:info  2021-12-27 23:01:56: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2021-12-27 23:01:56: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"m4lvz-3"}'
Zigbee2MQTT:info  2021-12-27 23:02:25: Device '0x84fd27fffe78fc24' joined
Zigbee2MQTT:info  2021-12-27 23:02:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84fd27fffe78fc24","ieee_address":"0x84fd27fffe78fc24"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-12-27 23:02:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x84fd27fffe78fc24"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-12-27 23:02:25: Starting interview of '0x84fd27fffe78fc24'
Zigbee2MQTT:info  2021-12-27 23:02:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84fd27fffe78fc24","ieee_address":"0x84fd27fffe78fc24","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-12-27 23:02:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x84fd27fffe78fc24"},"type":"pairing"}'
Zigbee2MQTT:info  2021-12-27 23:02:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84fd27fffe78fc24","ieee_address":"0x84fd27fffe78fc24"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-12-27 23:02:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x84fd27fffe78fc24"},"type":"device_announced"}'
Zigbee2MQTT:info  2021-12-27 23:02:32: Successfully interviewed '0x84fd27fffe78fc24', device has successfully been paired
Zigbee2MQTT:warn  2021-12-27 23:02:32: Device '0x84fd27fffe78fc24' with Zigbee model 'TS0202' and manufacturer name '_TYZB01_tv3wxhcz' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:info  2021-12-27 23:02:32: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x84fd27fffe78fc24","ieee_address":"0x84fd27fffe78fc24","status":"successful","supported":false},"type":"device_interview"}'
Zigbee2MQTT:info  2021-12-27 23:02:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x84fd27fffe78fc24","supported":false},"type":"pairing"}'
Zigbee2MQTT:info  2021-12-27 23:02:56: MQTT publish: topic 'zigbee2mqtt/zb_motion_wejscie', payload '{"battery":100,"illuminance":3242,"illuminance_lux":2,"led_indication":null,"linkquality":102,"motion_sensitivity":"medium","occupancy":false,"occupancy_timeout":60,"temperature":7.29,"update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:warn  2021-12-27 23:03:29: Received message from unsupported device with Zigbee model 'TS0202' and manufacturer name '_TYZB01_tv3wxhcz'

retepkirasam, 2021-12-29:
I have the same problem with TS0202 _TYZB01_zwvaj5wy

I wrote an external converter. The device is already supported, it works, but I have a warning with it..

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = { 
        fingerprint: [{ieee_address: '0x680ae2fffe70206e'}],
        zigbeeModel: ['TS0202'],  
        model: 'TS0202',
        vendor: 'TuYa', 
        description: 'Motion sensor',
        fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report],
        toZigbee: [],
        exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery()],
};

module.exports = definition;`

Koenkk, 2021-12-29:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-29 by Koenkk

marcinwrobel, 2021-12-29:
Zigbee model 'TS0202' and manufacturer name '_TYZB01_tv3wxhcz' still  recognized as unsupported

Koenkk, 2021-12-29:
@marcinwrobel added.

Changes will be available in the dev branch tomorrow. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

rednet1, 2023-01-20:
Zigbee model 'TS0202' and manufacturer name '_TYZB01_rwb0hxtf' still recognized as unsupported

Koenkk, 2023-01-21:
@rednet1 added

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

jposouza, 2023-08-03:
Hello: Zigbee model 'TS0202' and manufacturer name '_TZ3000_o4mkahkc' still recognized as unsupported

gutimondra, 2023-08-08:
Zigbee model 'TS0202' and manufacturer name '_TYZB01_j7iyqfcs' still recognized as unsupported

Koenkk, 2023-08-08:
Added the `_TYZB01_j7iyqfcs`

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

GavriKos, 2024-07-03:
Same thing with _TZ3000_lf56vpxj