Koenkk/zigbee2mqtt #10400: [Problem]: OTA of Ikea Repeater E1746

newlund, 2021-12-28:
### What happened?

I have six Ikea E1746 repeaters. Three of them I have updated from 2.3.070 to 2.3.086 sucessfully but the other three appear to be stuck at 2.3.070 even if the update seems to get applied.  See log
Perhaps this is not an issue with zigbee2mqtt but with the repeaters?
I friend of mine have the same issue with his Ikea E1746 repeaters

### What did you expect to happen?

Firmware version should be updated in the OTA tab and no more updates should be available

### How to reproduce it (minimal and precise)

Click update device firmware in the OTA tab

### Zigbee2MQTT version

1.22.1-1

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Debug log

info  2021-12-28 08:54:03: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T08:54:03+01:00","linkquality":255,"update":{"progress":97.51,"remaining":25,"state":"updating"},"update_available":false}'
info  2021-12-28 08:54:08: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T08:54:08+01:00","linkquality":255,"update":{"progress":97.51,"remaining":25,"state":"updating"},"update_available":false}'
debug 2021-12-28 08:54:08: Device 'Repeater 3 Kök' announced itself
info  2021-12-28 08:54:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Repeater 3 Kök","ieee_address":"0x680ae2fffe3b83c5"},"type":"device_announce"}'
info  2021-12-28 08:54:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Repeater 3 Kök"},"type":"device_announced"}'
debug 2021-12-28 08:54:09: Received Zigbee message from 'Repeater 3 Kök', type 'attributeReport', cluster 'genBasic', data '{"appVersion":33}' from endpoint 1 with groupID null
info  2021-12-28 08:54:09: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T08:54:09+01:00","linkquality":255,"update":{"progress":97.51,"remaining":25,"state":"updating"},"update_available":false}'
debug 2021-12-28 08:54:10: Retrieving state of 'Repeater 3 Kök' after reconnect
info  2021-12-28 08:55:33: Update of 'Repeater 3 Kök' at 100.00%
info  2021-12-28 08:55:33: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T08:54:09+01:00","linkquality":255,"update":{"progress":100,"remaining":25,"state":"updating"},"update_available":false}'
info  2021-12-28 08:55:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Repeater 3 Kök' at 100.00%","meta":{"device":"Repeater 3 Kök","progress":100,"status":"update_progress"},"type":"ota_update"}'
debug 2021-12-28 08:55:33: Received Zigbee message from 'Repeater 3 Kök', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20190311","swBuildId":"2.3.070"}' from endpoint 1 with groupID null
info  2021-12-28 08:55:33: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T08:55:33+01:00","linkquality":255,"update":{"progress":100,"remaining":25,"state":"updating"},"update_available":false}'
info  2021-12-28 08:55:33: Finished update of 'Repeater 3 Kök', from '{"dateCode":"20190311","softwareBuildID":"2.3.070"}' to '{"dateCode":"20190311","softwareBuildID":"2.3.070"}'
info  2021-12-28 08:55:33: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T08:55:33+01:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
info  2021-12-28 08:55:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"id":"Repeater 3 Kök","transaction":"1whhe-1"},"meta":{"device":"Repeater 3 Kök","from":{"dateCode":"20190311","softwareBuildID":"2.3.070"},"status":"update_succeeded","to":{"dateCode":"20190311","softwareBuildID":"2.3.070"}},"type":"ota_update"}'
info  2021-12-28 08:55:33: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20190311","software_build_id":"2.3.070"},"id":"Repeater 3 Kök","to":{"date_code":"20190311","software_build_id":"2.3.070"}},"status":"ok","transaction":"1whhe-1"}'
debug 2021-12-28 08:56:20: Received Zigbee message from 'Repeater 3 Kök', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587662897,"imageType":4354,"manufacturerCode":4476}' from endpoint 1 with groupID null
info  2021-12-28 08:56:20: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T08:56:20+01:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
debug 2021-12-28 08:56:20: Device 'Repeater 3 Kök' requested OTA
debug 2021-12-28 08:56:23: Received Zigbee message from 'Repeater 3 Kök', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587662897,"imageType":4354,"manufacturerCode":4476}' from endpoint 1 with groupID null
info  2021-12-28 08:56:23: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T08:56:23+01:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
debug 2021-12-28 08:56:23: Device 'Repeater 3 Kök' requested OTA
debug 2021-12-28 08:56:26: Received Zigbee message from 'Repeater 3 Kök', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587662897,"imageType":4354,"manufacturerCode":4476}' from endpoint 1 with groupID null
info  2021-12-28 08:56:26: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T08:56:26+01:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
debug 2021-12-28 08:56:26: Device 'Repeater 3 Kök' requested OTA
debug 2021-12-28 08:56:29: Received Zigbee message from 'Repeater 3 Kök', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587662897,"imageType":4354,"manufacturerCode":4476}' from endpoint 1 with groupID null
info  2021-12-28 08:56:29: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T08:56:29+01:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
debug 2021-12-28 08:56:29: Device 'Repeater 3 Kök' requested OTA
debug 2021-12-28 08:56:32: Received Zigbee message from 'Repeater 3 Kök', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587662897,"imageType":4354,"manufacturerCode":4476}' from endpoint 1 with groupID null
info  2021-12-28 08:56:32: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T08:56:32+01:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
debug 2021-12-28 08:56:32: Device 'Repeater 3 Kök' requested OTA
debug 2021-12-28 09:01:50: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Repeater 3 Kök","transaction":"1whhe-4"}'
info  2021-12-28 09:01:50: Checking if update available for 'Repeater 3 Kök'
info  2021-12-28 09:01:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Repeater 3 Kök'","meta":{"device":"Repeater 3 Kök","status":"checking_if_available"},"type":"ota_update"}'
debug 2021-12-28 09:01:52: Received Zigbee message from 'Repeater 3 Kök', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587662897,"imageType":4354,"manufacturerCode":4476}' from endpoint 1 with groupID null
info  2021-12-28 09:01:52: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T09:01:52+01:00","linkquality":255,"update":{"state":"idle"},"update_available":false}'
debug 2021-12-28 09:01:52: Device 'Repeater 3 Kök' requested OTA
info  2021-12-28 09:01:53: Update available for 'Repeater 3 Kök'
info  2021-12-28 09:01:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update available for 'Repeater 3 Kök'","meta":{"device":"Repeater 3 Kök","status":"available"},"type":"ota_update"}'
info  2021-12-28 09:01:53: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T09:01:52+01:00","linkquality":255,"update":{"state":"available"},"update_available":true}'
info  2021-12-28 09:01:53: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Repeater 3 Kök","updateAvailable":true},"status":"ok","transaction":"1whhe-4"}'
debug 2021-12-28 09:01:55: Received Zigbee message from 'Repeater 3 Kök', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587662897,"imageType":4354,"manufacturerCode":4476}' from endpoint 1 with groupID null
info  2021-12-28 09:01:55: MQTT publish: topic 'zigbee2mqtt/Repeater 3 Kök', payload '{"last_seen":"2021-12-28T09:01:55+01:00","linkquality":255,"update":{"state":"available"},"update_available":true}'
debug 2021-12-28 09:01:55: Device 'Repeater 3 Kök' requested OTA

copystring, 2021-12-28:
I have this too. All 4 repeaters I have are on 2.3.070 and can't upgrade to 2.3.086. Update completes to 100%. z2m logs successfully updated. Version is still the same. Confirmed by swBuildId in dev console of each E1746

hoggerz, 2021-12-28:
I have exactly the same issue, however at some point one of my other repeaters successfully updated to 2.3.086 before somethings broken somewhere. I also have the same problem with my FYRTUR blinds (two are stuck on 2.3.073, one updated to 2.3.088), and an Ikea puck remote (still stuck on 2.3.014). They go through the process, get to 100%, and the message is like the above, telling me its updated from the "old version" to the "old version". I'm using the troublesome Sonoff zigbee plus dongle though, so perhaps that breaks everything, as its a bit odd and temperamental compared to my ZZH.

hoggerz

MattWestb, 2021-12-29:
I have one E174 and its downloading and rebooting after 100% but coming back with the original version so its not flashing the new firmware.
I have disabling OTA updates after have it running over night without success so i think some device dont like the new updates but some looks working OK.

copystring, 2021-12-30:
So I bought 2 used Tradfri Routers. One was on 2.2.003 and the other on 2.2.005.
Both of them updated to 2.3.086 just fine first try. The other 4 Tradfri Routers did not update even after about 10 attempts on each. Feels like 2.3.070 is fishy preventing the update.

groenmarsmannetje, 2022-01-04:
Same issue here. Did somebody already try to update it with the IKEA gateway itself? Just to make sure that it is an issue of the firmware version for that specific device and not related to zigbee2mqtt OTA.

copystring, 2022-01-04:
I tried to update via official gateway but none of my routers are able to pair with the gateway.

groenmarsmannetje, 2022-01-04:
IKEA mentioned in [their release notes](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html) for version 2.3.070:

[image]

Whatever that might mean, but they did change something in the OTA procedure.

jhf2442, 2022-01-04:
Same here, 2x E1746 updated to 2.3.086, one further stuck at 2.3.070. Update to 2.3.070 already done through OTA some months ago. Therefore excluding issues with Zigbee adapter (ITEAD Zigbee 3.0 USB Dongle Plus)
All 3 repeaters bought at same time (same batch ?)
`payload '{"message":"Failed to check if update available for 'tradfri_repeater_3' (Device didn't respond to OTA request)","meta":{"device":"tradfri_repeater_3","status":"check_failed"},"type":"ota_update"}'`
Ikea testserver NOT used

groenmarsmannetje, 2022-01-04:
> Same here, 2x E1746 updated to 2.3.086, one further stuck at 2.3.070. Update to 2.3.070 already done through OTA some months ago. Therefore excluding issues with Zigbee adapter (ITEAD Zigbee 3.0 USB [...]

Where all 3 devices on 2.3.070 before you updated them?

copystring, 2022-01-04:
I remembered I have tried this also via home assistant and iobroker. Both were not able to update any of the routers.

klada, 2022-01-05:
BTW: I have also tried updating the routers using DeCONZ on a Conbee II. They also did not update there. So this might not only be a Z2M issue unfortunately.

copystring, 2022-01-05:
Maybe we can use this new feature (https://github.com/Koenkk/zigbee2mqtt/commit/16842a3e72b741811c18de3ab00b2848a625dc6f) to downgrade to an older version and then upgrade to the most recent version?
I tried but does not work for me. See my comment.

jhf2442, 2022-01-07:
> > Same here, 2x E1746 updated to 2.3.086, one further stuck at 2.3.070. Update to 2.3.070 already done through OTA some months ago. Therefore excluding issues with Zigbee adapter (ITEAD Zigbee 3.0 USB Dongle Plus) All 3 repeaters bought at same time (same batch ?) `payload '{"message":"Failed to check if update available for 'tradfri_repeater_3' (Device didn't respond to OTA request)","meta":{"device":"tradfri_repeater_3","status":"check_failed"},"type":"ota_update"}'` Ikea testserver NOT used
> 
> Where all 3 devices on 2.3.070 before you updated them? And I think your situation is different because the OTA check doesn't fail for the problem reported here. It will update upto 100% without error but after restart it is still has previous version installed and OTA check shows that there is still an update available.
> 
> Your error message is a timeout because the device did not respond. Power cycle it and try again.

this actually solved the issue. sorry for the noise !

Koenkk, 2022-01-17:
This issue should be fixed now in the latest-dev. 

If you've already updated your device, you can refresh the version by: 
- go the the frontend, click on your device
- read the swBuildId and dateCode via the dev console (endpoint number can differ from the screenshot below):
[image]
- restart z2m

copystring, 2022-01-17:
Thank you @Koenkk but I think the issue you solved is a different one.
The issue we have the device is actually not updating the firmware and not just reporting the old version.
I've tried to update via z2m, homeassistant, iobroker and zigpy. Neither were able to update the firmware coming from 2.3.070.
Coming from 2.2.003 or 2.2.005 to 2.3.086 works just fine. I'm confident this is an issue with the firmware or upgrade path. Feels like IKEA has to fix this unless we can find a way to downgrade via OTA.

BUT! It is actually possible to flash and backup the firmware by directly connecting a debugger to the device. However I guess backing up the firmware from an already update device and flashing it to one which does not update itself would probably also transfer the IEEE Address to the other device which is well.... not good? xD That's if it is like that.
I could spend some time to find a way but I don't have much time for this at the moment so I'd rather wait for IKEA to fix this.

But hey I tried anyway with latest dev. Firmware stays on the old version from 2019:
`Info Read result of 'genBasic': {"swBuildId":"2.3.070","dateCode":"20190311"}`

MattWestb, 2022-01-17:
Silabs ERE32MG chips have the IEEE burned in the hardware config aria that cant being erased and the OTA files is normally only updating the main flash so its very safe flashing one older extracted OTA file like we have doing  with all controllers that was getting bad firmware for some month ago.
If you is dumping one mainflash for one device and it was joined to your network you is also getting the network information to the new device until you is resetting it and joining one new network.
I have one E1746 that was updated to later firmware then i was buying it for 2 months ago and its rolling back the new OTS update all the time so i have disabling IKEA OTA until i have extracting one old working OTA file andreflashing it with SWD probe.

copystring, 2022-01-17:
> Silabs ERE32MG chips have the IEEE burned in the hardware config aria that cant being erased and the OTA files is normally only updating the main flash so its very safe flashing one older extracted OTA file
The backup file contains more than the ota file. I compared both. Tha backup file is a few kb larger than the OTA file. Whether the IEEE is part of the backup or not I can't tell since I did not look for it.

MattWestb, 2022-01-17:
Then the device is being reseted its initiating storage in the higher part of the main flash and its used for saving link keys, counters and all other "runtime storage" like power on state.
You can looking in the OTA file and comparing (its dont have bootloader and is flashed at 0x04000) and is having some singing data before and after but all between is in clear.

You can deleting the NVM from the dumped flash but its easier extracting one "pure" flash file from the OTA file and its safer if making it as one S37 like we was doing with the controllers and is 100% factors made.

The IEEE is burned in the device hardware config aria like the radio calibration is made and cant being eraser but the Zigbee stack can reading it and its also possible buring one new IEEE in `user data` that can being erased with one chip erase and then the device is going back the the factory one.

copystring, 2022-01-20:
I wish I had more time to have a look this. Although it does not seem very practical. Many users of z2m won't be able to or don't want to flash their devices.
The actual problem feels like it's on IKEAs side. So I'll wait and hope they will fix this.

j6s33m, 2022-01-31:
> 

@copystring @Koenkk - I'm having this exact same problem with all my IKEA devices. Here's an example from my IKEA Signal Repeater:

`Warning Firmware on '0x588e81fffe3f2ac6' is newer than latest firmware online.`

and yet...

`Info Read result of 'genBasic': {"swBuildId":"2.3.070","dateCode":"20190311"}`

**Other items not updating:**

FYRTUR block-out roller blind (stuck on: 2.3.073)
SYMFONISK sound controller (stuck on: 2.1.024)
TRADFRI Signal Repeater (stuck on: 2.3.070)
TRADFRI open/close remote (stuck on: 2.3.079)
TRADFRI bulb E26 WS opal 1000lm (stuck on: 2.0.029)

**Note I have a Conbee II stick running my Z2MQTT. Other products like Hue have updated with no issue.

chrisrock1984, 2022-02-18:
same problem for me - stuck at 2.3.070

MattWestb, 2022-02-18:
= returning it to IKEA if you not have opening it with force like i have doing.

hoggerz, 2022-03-22:
Just to add, my 2 Ikea repeaters are still stuck on 2.3.070, and two of my Fyrtur blinds are stuck on the last firmware variant 2.3.073, one updated successfully to 2.3.088. Any ideas or something I can try to help solve it?

hoggerz

clyso, 2022-04-14:
same here - 1 repeater stuck at 2.3.070

AdmiralStipe, 2022-05-25:
Same issue here, one repeater updated, two just install the update, but nothing changes and update is immediateky available again. Please, re-open the issue.

tbarnekov, 2022-10-24:
@Koenkk 
I have the same issue with updating IKEA repeaters on 2.3.070 as well as FYRTUR blinds on 2.3.073 (the firmware of both introduced Secure OTA according to the release notes).

When searching for a solution I came across a thread on the DeCONZ where users are having the same issue. I see the deCONZ teams has recently managed to get it working with some changes to the OTA update process. I'm thinking their solution might work for z2m as well.

I tracked down the commit that fixed the issue for deCONZ users: https://github.com/dresden-elektronik/deconz-ota-plugin/commit/4732fea004328b0efd7a677d75f0e138dd5e7db2

newlund, 2022-10-25:
> @Koenkk I have the same issue with updating IKEA repeaters on 2.3.070 as well as FYRTUR blinds on 2.3.073 (the firmware of both introduced Secure OTA according to the release notes).
> 
> When searching for a solution I came across a thread on the DeCONZ where users are having the same issue. I see the deCONZ teams has recently managed to get it working with some changes to the OTA update process. I'm thinking their solution might work for z2m as well.
> 
> I tracked down the commit that fixed the issue for deCONZ users: [dresden-elektronik/deconz-ota-plugin@4732fea](https://github.com/dresden-elektronik/deconz-ota-plugin/commit/4732fea004328b0efd7a677d75f0e138dd5e7db2)

Nice find! :)

Perhaps you could create a new issue with this information since the current one is closed?

Koenkk, 2022-10-25:
@manup could you explain what part of https://github.com/dresden-elektronik/deconz-ota-plugin/commit/4732fea004328b0efd7a677d75f0e138dd5e7db2 fixed the ikea updates?

manup, 2022-10-25:
The problem in our case was that we have parsed the tags and section data and than incompletely recreated the OTA file (to support OTA file editor in the UI). This worked fine in the past but newer files contain more tags that we didn't take into account.

So the change is simply to use the plain original OTA file from the start where the header is found.

Koenkk, 2022-10-26:
@manup thanks for the explanation!

Zigbee2MQTT already handles these extra tags (https://github.com/Koenkk/zigbee-herdsman-converters/blob/a5abd98e98719c1f5ce55712d50ed1edc16a3395/lib/ota/common.js#L57) nor does z2m modify the OTA file before sending so I don't think it is the same issue.

@manup reading the ZCL spec I saw this:

[image]

Does Deconz set the device ieeeAddr in the upgrade file destination field?

manup, 2022-10-26:
We do parse the fields but as far as I can tell these are not used further. Since the whole file is send from the start of the header (if requested with offset 0 from the device) the fields are also available by the receiving device.

https://github.com/dresden-elektronik/deconz-ota-plugin/blob/c69e8f96da192e04fbef53a017be8a61f422ab8d/otau_file.cpp#L125

The total image size which is also used in responses is taken from the header (not calculated from file size). Another thing is that we cap max data size to 40 bytes as we had problems with some devices requesting larger sizes but couldn't handle it :)

Koenkk, 2022-10-27:
@manup thanks for the explanation

@ all others: I've checked the code but I cannot spot the issue yet. Can someone provide the herdsman debug log starting from the point where starting the upgrade until it finishes?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

tbarnekov, 2022-10-27:
@Koenkk Knock yourself out :)
[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/9883021/debug.txt)

Koenkk, 2022-10-29:
Can you try to reduce the blocksize from 50 to 40? 

- If running via docker first do a `docker exec -it CONTAINER_ID /bin/bash`
- `nano node_modules/zigbee-herdsman-converters/lib/ota/common.js`
- Change line 207 from `const maximumDataSize = imageBlockRequest.payload.manufacturerCode === 4474 ? 40 : 50;` to `const maximumDataSize =  40;`
- Restart z2m (or the container with `docker restart CONTAINER_ID`)
- Try OTA update again.

tbarnekov, 2022-10-29:
Unfortunately it made no difference.

[debug2.txt](https://github.com/Koenkk/zigbee2mqtt/files/9894398/debug2.txt)

Koenkk, 2022-10-30:
@manup

>  This worked fine in the past but newer files contain more tags that we didn't take into account.

AFAIK this is not the case, e.g. the header of `10037603-3.1-TRADFRI-signal-repeater-2.3.086.ota.ota.signed ` has `otaHeaderFieldControl` `0` so the extra fields like `otaHeaderFieldControl` are not included:

[image]

@tbarnekov do you have a Conbee? If yes can you check if updating these plugs with the latest Deconz version?

tbarnekov, 2022-10-31:
Dammit! It turns out DeCONZ can't perform the upgrade either :(

I guess we're back to sqaure one then...

manup, 2022-10-31:
Did you use the latest versions deCONZ v2.19.1 and firmware 0x26780700 ?

I did update my plug with the 10005777-TRADFRI-control-outlet-2.3.089.ota.ota.signed
Heres how the basic cluster looks after the update.

[image]

tbarnekov, 2022-10-31:
I'm running firmware 0x26780700 and just downloaded deCONZ for Windows today (it says it's 2.19.1).

I've never used deCONZ before so I may have done something wrong. It did appear to transfer the file to the repeater though - it took about 6 minutes. The version number reported by the repeater didn't change afterwards. I repaired it with z2m after the update and got the same old 2.3.070.

tbarnekov, 2022-10-31:
I looked at the image for the IKEA outlet (10005777-TRADFRI-control-outlet-2.3.089.ota.ota.signed) and I don't see any significant differences in the OTA header between the outlet and the repeater. The outlet seems to have 0 in otaHeaderFieldControl as well. It also looks like the final bytes of the image (based on totalImageSize) are the same in both images which leads me to believe we're sending the right span of data.

I had an unused IKEA repeater lying around with FW 2.2.005 and I could update that to 2.3.088 without issue. So I think we can also rule out a bad image for the repeater.

manup, 2022-10-31:
I just remembered that for some other devices from Philips and Ubisys it was needed to update not right away to the latest version but to the next higher version, so that updated are applied in order. Might be worth a try.

Would be good to compare the versions "from -> to"  which peaople have problems to update with the ones that worked.

newlund, 2022-11-02:
Yes, I can confirm that. All devices were on the same fw level before I started updating to 2.3.086 (from 2.3.070). Three of them updated fine to 2.3.086 and three got stuck on 2.3.070.
Strange yes... Perhaps different hw revisions?

Koenkk, 2022-11-02:
If someone has an IKEA hub, can you try updating this device via the IKEA hub and make a sniff while doing so? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

newlund, 2022-12-03:
Nice that you now can do OTA update via HA! :)

However, now I always have three repeaters requiring update and they cannot be updated due to the problems in this thread.
Possible to hide these somehow?

[image]

MattWestb, 2022-12-03:
If you have SWD probe you can flashing one working firmware file with version 2.2.005 that is extracted from the original OTA with Zipy-CLI. Some user have this problems (including my) with that the firmware is being rolling back and after flashing the firmware its doing the OTA is working OK so i think IKEA have doing somthing bad in earlier updates that cant being solved with OTA.

https://github.com/zigpy/zigpy/discussions/660#discussioncomment-2256254

PS: For 2 weeks I was baying one more and it was shipped with  the last firmware so no go for more testing.

newlund, 2022-12-03:
Don't have SWD probe and don't feel like messing with opening them up. No way to solve it using OTA I guess?
If not, I think I would rather get rid of them or continue using them with 2.3.070. Possible to hide the updates in HA?

closed 2023-01-10 by github-actions[bot]

groenmarsmannetje, 2023-01-13:
Reopen - I just checked with latest Z2M version and coordinator firmware. Problem still exists

jbliesener, 2023-08-15:
+1, same here.

zikson, 2024-02-01:
Checked with latest dev version and coordinator firmware. Problem still exists.

Zigbee2MQTT version
    1.35.1-dev commit: [630b4ec5](https://github.com/Koenkk/zigbee2mqtt/commit/630b4ec5)

Coordinator type
    zStack3x0

Coordinator revision
    20230507

Coordinator IEEE Address
    0x00124b0005a97b2b

Frontend version
    0.6.152

Zigbee-herdsman-converters version
    18.25.0

Zigbee-herdsman version
    0.33.5

Jim-Noble, 2025-01-11:
This is still an issue.
Relevant log entries of failed update:

`[2025-01-11 22:05:37] info:     z2m: tradfri_repeater_4 (0x000d6ffffe7ad2e8): E1746 - IKEA TRADFRI signal repeater (Router)
[2025-01-11 22:05:37] info:     z2m: tradfri_repeater_1 (0x000d6ffffe8e7eab): E1746 - IKEA TRADFRI signal repeater (Router)
[2025-01-11 22:05:37] info:     z2m: tradfri_repeater_2 (0x000d6ffffe789913): E1746 - IKEA TRADFRI signal repeater (Router)
[2025-01-11 22:05:37] info:     z2m: tradfri_repeater_3 (0x000d6ffffe7c1019): E1746 - IKEA TRADFRI signal repeater (Router)
[2025-01-11 22:05:38] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tradfri_repeater_4', payload '{"identify":null,"linkquality":25,"update":{"installed_version":587753009,"latest_version":587753009,"state":"idle"},"update_available":false}'
[2025-01-11 22:05:38] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tradfri_repeater_1', payload '{"identify":null,"linkquality":47,"update":{"installed_version":587662897,"latest_version":587753009,"state":"available"},"update_available":false}'
[2025-01-11 22:05:38] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tradfri_repeater_2', payload '{"identify":null,"linkquality":94,"update":{"installed_version":587662897,"latest_version":587753009,"state":"available"}}'
[2025-01-11 22:05:38] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tradfri_repeater_3', payload '{"identify":null,"linkquality":83,"update":{"installed_version":587662897,"latest_version":587753009,"state":"available"},"update_available":false}'
[2025-01-11 22:08:08] info:     zhc:ota: [0x000d6ffffe7ad2e8 | TRADFRI signal repeater] Firmware release notes: https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html
[2025-01-11 22:08:08] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tradfri_repeater_4', payload '{"identify":null,"linkquality":40,"update":{"installed_version":587662897,"latest_version":587753009,"state":"available"},"update_available":false}'
[2025-01-11 22:08:08] info:     z2m: Update available for 'tradfri_repeater_4'
[2025-01-11 22:09:14] info:     z2m: Updating 'tradfri_repeater_4' to latest firmware
[2025-01-11 22:09:14] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tradfri_repeater_4', payload '{"identify":null,"linkquality":40,"update":{"installed_version":587662897,"latest_version":587753009,"state":"available"},"update_available":false}'
[2025-01-11 22:09:15] info:     zhc:ota: [0x000d6ffffe7ad2e8 | TRADFRI signal repeater] Getting latest image, meta: {"fileName":"tradfri-signal-repeater_release_prod_v587753009_3ce8f096-bd12-4b2c-b66e-51dc9f2637dc.ota","fileVersion":587753009,"fileSize":197052,"originalUrl":"https://fw.ota.homesmart.ikea.com/files/tradfri-signal-repeater_release_prod_v587753009_3ce8f096-bd12-4b2c-b66e-51dc9f2637dc.ota","url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/IKEA/tradfri-signal-repeater_release_prod_v587753009_3ce8f096-bd12-4b2c-b66e-51dc9f2637dc.ota","imageType":4354,"manufacturerCode":4476,"sha512":"f5166c7fb478b574269092b1086424cbc726bb05fa5512d5dcbb62f53b0fa2f184533d9467977e1e6ebc83e1ec6998b5097bfb65238b226b24a13ab54a9de731","otaHeaderString":"GBL GBL_tradfri_zigbee_repeater\u0000","releaseNotes":"https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html"}
[2025-01-11 22:09:24] info:     z2m: Update of 'tradfri_repeater_4' at 0.00%
[2025-01-11 22:09:24] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tradfri_repeater_4', payload '{"identify":null,"linkquality":36,"update":{"installed_version":587662897,"latest_version":587753009,"progress":0,"state":"updating"},"update_available":false}'
[2025-01-11 22:09:54] info:     z2m: Update of 'tradfri_repeater_4' at 2.89%, <E2><89><88> 22 minutes remaining
[2025-01-11 22:09:54] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tradfri_repeater_4', payload '{"identify":null,"linkquality":36,"update":{"installed_version":587662897,"latest_version":587753009,"progress":2.89,"remaining":1305,"state":"updating"},"update_available":false}'
[2025-01-11 22:10:24] info:     z2m: Update of 'tradfri_repeater_4' at 5.94%, <E2><89><88> 18 minutes remaining
[2025-01-11 22:10:24] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tradfri_repeater_4', payload '{"identify":null,"linkquality":40,"update":{"installed_version":587662897,"latest_version":587753009,"progress":5.94,"remaining":1092,"state":"updating"},"update_available":false}'
[2025-01-11 22:10:54] info:     z2m: Update of 'tradfri_repeater_4' at 8.63%, <E2><89><88> 17 minutes remaining
[2025-01-11 22:10:54] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tradfri_repeater_4', payload '{"identify":null,"linkquality":43,"update":{"installed_version":587662897,"latest_version":587753009,"progress":8.63,"remaining":1048,"state":"updating"},"update_available":false}'
[2025-01-11 22:11:24] info:     z2m: Update of 'tradfri_repeater_4' at 11.67%, <E2><89><88> 16 minutes remaining
[2025-01-11 22:11:24] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tradfri_repeater_4', payload '{"identify":null,"linkquality":47,"update":{"installed_version":587662897,"latest_version":587753009,"progress":11.67,"remaining":976,"state":"updating"},"update_available":false}'
[2025-01-11 22:11:54] info:     z2m: Update of 'tradfri_repeater_4' at 14.72%, <E2><89><88> 15 minutes remaining
[... cut ...]
[2025-01-11 22:28:37] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/tradfri_repeater_4', payload '{"identify":null,"linkquality":54,"update":{"installed_version":587662897,"latest_version":587753009,"state":"available"},"update_available":false}'
[2025-01-11 22:28:37] info:     z2m: Update available for 'tradfri_repeater_4'
`

It looks like it's worked - the reported installed firmware is correct, but the buildID and date code are still the old one, but then it reports an update is available, and suddenly the installed firmware version reported has reverted back to the old one... :-?

tbarnekov, 2025-01-13:
@Jim-Noble I've personally come to the conclusion that the problem is not in Z2M, but rather a firmware issue with the 2.3.x firmwares.

Every device I've owned and updated to 2.3.070 (or 2.3.073) has been unable to accept further updates. I've had a few devices on older FW versions and updated those to newer FW than 2.3.x (skipping the 2.3.x versions) and all of those devices still accepts FW updates.

MattWestb, 2025-01-13:
@tbarnekov I think you is being more then right.
Then IKEA was roiling out update pre 2.3.x many controllers was not able flashing it and was rolling back to the current ruining firmware.
We was extracting the signed OTA files and flashing it to the device with SWD and then all was working OK and also later OTA files looks working,
Also i have never seen the problem with real old FW like first gen ZLL updated to the last like the old motion sensor.

groenmarsmannetje, 2025-05-15:
@newlund I did see that you downgraded the firmware with OTA of an IKEA styrbar in [this message](https://github.com/Koenkk/zigbee2mqtt/issues/25546#issuecomment-2629398593)

Did you or somebody else did something similar with trying to downgrade the repeater firmware?

newlund, 2025-05-15:
> [@newlund](https://github.com/newlund) I did see that you downgraded the firmware with OTA of an IKEA styrbar in [this message](https://github.com/Koenkk/zigbee2mqtt/issues/25546#issuecomment-2629398593)
> 
> Did you or somebody else did something similar with trying to downgrade the repeater firmware?

I have tried but did not succeed :(