Koenkk/zigbee2mqtt #10403: LEDVANCE Smart+ Outdoor Plug (AC26940)

machv, 2021-12-28:
### What happened?

I am not sure whether this is common behavior of these LEDVANCE plugs, but when I manually press their power button to toggle its state, it changes the state (on/off) of that plug, but that state isn't propagated back to the Z2M and Z2M still thinks that plug is in previous state.

### What did you expect to happen?

I would expect that the plug's state would be synced properly in Z2M, e. g. when I turn the plug on, it would show in Z2M that plug is turned on.

Is this common behavior of these LEDVANCE plugs? I tried to upgrade firmware via available OTA to the latest available `00203045`, but it still behaves in the same way. For example, Xiaomi plugs that I also have, are reporting state back immediately after a power button is pressed.

As a workaround I am currently periodically sending `zigbee2mqtt/ledvance-plug/get {"state": ""}` message to refresh the state, but it depends on active polling of the device, so if there would be a way to avoid this it would be better.

### How to reproduce it (minimal and precise)

Change the plug's state using physical button.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20210708

### Adapter

zzh!

### Debug log

When I press the plug's button nothing appears in debug log at all.

Koenkk, 2021-12-29:
Can you check if it works after going to the frontend -> click on the device -> Bind, configure it like this and click bind:

[image]

If it works, please share a screenshot of the bind tab.

machv, 2021-12-29:
For that combination of parameters (Source endpoint, Destination, Destination endpoint) I have only available Cluster `Ota`, I am missing the `OnOff`:
[image]

Second available Source endpoint in dropdown is `242` with a cluster `greenPower`:
[image]

Koenkk, 2021-12-29:
Can you provide the data/database.db entry of this device?

machv, 2021-12-29:
Of course, this is the corresponding line from `database.db` file for that plug:
```
{"id":9,"type":"Router","ieeeAddr":"0xf0d1b800001bdf1b","nwkAddr":1379,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Unknown","modelId":"Plug Value","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Plug Value","manufacturerName":"LEDVANCE","powerSource":0,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":2,"swBuildId":"02056550"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":2,"swBuildId":"00203045","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1640800201953,"useImplicitCheckin":true}
```

Koenkk, 2021-12-29:
Can you check with the following external converter?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Outdoor Plug', 'Plug Value'],
    model: 'AC26940/AC31266',
    vendor: 'LEDVANCE',
    description: 'Smart Zigbee outdoor plug EXTERNAL CONVERTER',
    extend: extend.switch(),
    ota: ota.ledvance,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

See https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device how to add the external converter, to make sure it is loaded the description in the frontend should contain EXTERNAL CONVERTER

machv, 2021-12-29:
Perfect, thank you!

With this external converter, plug's state is immediately reported back and in Frontend, and Bind tab looks like this:
[image]

Koenkk, 2021-12-29:
Great, added the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-29 by Koenkk

machv, 2021-12-29:
Great, thank you very much!