Koenkk/zigbee2mqtt #10404: [Problem]: IKEA devices leaves the network

knutasund, 2021-12-28:
### What happened?

Some battery operated IKEA switches stopped working. 
Unable to repair. 
"Device '-----f' left the network"

### What did you expect to happen?

Pair IKEA devices E1524, E1743

### How to reproduce it (minimal and precise)

Permit join
Reset IKEA E1524, E1743 device close to coordinator

### Zigbee2MQTT version

1.22.1 

### Adapter firmware version

20210708

### Adapter

ZZH!

### Debug log

[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/7784827/debug.txt)

jeroenvansch, 2022-01-02:
What firmware are those devices running? if it's not 2.3.079/2.3.080 or higher you should update them:
[From Ikea firmware release notes:
](https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html)◉ TRÅDFRI remote, SYMFONISK sound remote, TRÅDFRI shortcut button (V-2.3.080)
◉ TRÅDFRI on/off switch, open/close remote (V-2.3.079)
◆ Fixed bug: Some remotes and sensors losing connection due to low battery
◆ Increased security regarding upgrade of firmware

knutasund, 2022-01-03:
I guess I have no way of checking and updating without pairing to Trådfri hub (which I do not have at the moment). Strange thing is that some devices work, and some not. 
BTW, I recently upgraded to ZZH! from CC2531, no other major changes.

claymen, 2023-06-26:
This might be a bit late, but I ran into the same issue, and found the fix was to force the device to pair only with the coordinator rather than pair to any router. This allowed it to join, and could update the firmware after.