Koenkk/zigbee2mqtt #10407: [Problem]: Smart plug TS011F _TZ3000_gjnozsaz was not recognized

peetereczek, 2021-12-28:
### What happened?

Smart plug {modelID: 'TS011F', manufacturerName: '_TZ3000_gjnozsaz', applicationVersion: 68} was not recognized

### What did you expect to happen?

That it will be recognized as model: 'TS011F_plug_3', description: 'Smart plug (with power monitoring by polling)',

### How to reproduce it (minimal and precise)

just try to pair new device

### Zigbee2MQTT version

1.22.1-dev commit: 28cb8b0

### Adapter firmware version

0x26700700

### Adapter

ConBee2

### Debug log

Info Device '0xa4c138744505194b' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138744505194b","ieee_address":"0xa4c138744505194b"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c138744505194b"},"type":"device_connected"}'
Debug Received Zigbee message from '0xa4c138744505194b', type 'raw', cluster '57344', data '{"data":[8,136,10,1,208,72,2,0,6],"type":"Buffer"}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_gjnozsaz'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138744505194b', type 'raw', cluster '57344', data '{"data":[8,137,10,2,208,72,2,0,10],"type":"Buffer"}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_gjnozsaz'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138744505194b', type 'attributeReport', cluster 'genOnOff', data '{"20480":1,"32768":0,"moesStartUpOnOff":2,"offWaitTime":0,"onOff":1,"onTime":0,"tuyaBacklightMode":1}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_gjnozsaz'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138744505194b', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"53264":2,"switchType":0}' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_gjnozsaz'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Debug Received Zigbee message from '0xa4c138744505194b', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null
Warning Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_gjnozsaz'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

peetereczek, 2021-12-28:
database entry:
`{"id":2,"type":"Router","ieeeAddr":"0xa4c138744505194b","nwkAddr":64627,"manufId":4417,"manufName":"_TZ3000_gjnozsaz","powerSource":"Mains (single phase)","modelId":"TS011F",
    "epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],
        "clusters":{
            "genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_gjnozsaz","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},
            "genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},
            "manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"53264":2,"switchType":0}},
            "haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},
            "seMetering":{"attributes":{"divisor":100,"multiplier":1}}
        },
        "binds":[
            {"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07e4e1","endpointID":1},
            {"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00212effff07e4e1","endpointID":1},
            {"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00212effff07e4e1","endpointID":1}
        ],
        "configuredReportings":[
            {"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},
            {"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},
            {"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},
            {"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}
        ],
        "meta":{}},
        "242":
            {"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},
                "appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,
                "meta":{"configured":680236614},"lastSeen":1640696170740,"useImplicitCheckin":true}`

peetereczek, 2021-12-28:
Herdsman debug:

> 2021-12-28T13:34:58.914Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 24
2021-12-28T13:34:58.916Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 24 srcAddr: 0xfc73 destAddr: 0x0 profile id: 0x104 cluster id: 0x6 lqi: 255
2021-12-28T13:34:58.917Z zigbee-herdsman:deconz:frameParser response payload: 24,3,1,0,0,0,16,1
2021-12-28T13:34:58.923Z zigbee-herdsman:deconz:adapter resolve data request with transSeq Nr.: 3
2021-12-28T13:34:58.929Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":3,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":0,"status":0,"dataType":16,"attrData":1}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":64627,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2021-12-28T13:34:58.940Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-12-28T13:34:58.941Z zigbee-herdsman:deconz:adapter response received

peetereczek, 2021-12-28:
After setting up custom device manually, from main branch tuya.js file:

> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const definition = {
    fingerprint: [
        {modelID: 'TS011F', manufacturerName: '_TZ3000_gjnozsaz', applicationVersion: 68},
    ],
    model: 'TS011F_plug_3',
    description: 'Smart plug (with power monitoring by polling)',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    options: [exposes.options.measurement_poll_interval()],
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
        e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
            .withDescription('Recover state after power outage')],
    onEvent: tuya.onEventMeasurementPoll,
};
module.exports = definition;

Device is reporting empty values:

[image]

> {
    "current": null,
    "energy": null,
    "linkquality": 255,
    "power": null,
    "power_outage_memory": "restore",
    "state": "ON",
    "voltage": null
}

Setting up state and power outage memory seems fine

porzino, 2021-12-28:
Same problem here.
I bought it here: https://es.aliexpress.com/item/1005003444957152.html

[image]

danielvandenberg95, 2021-12-29:
I just re-pulled the latest image, and that got it working:

Before docker pull:
```
Zigbee2MQTT:error 2021-12-29 11:23:50: Not connected to MQTT server!
Zigbee2MQTT:error 2021-12-29 11:23:50: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":{"friendly_name":"0xa4c13876161f82b5"},"type":"device_connected"}
Zigbee2MQTT:error 2021-12-29 11:23:50: Not connected to MQTT server!
Zigbee2MQTT:error 2021-12-29 11:23:50: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":"interview_started","meta":{"friendly_name":"0xa4c13876161f82b5"},"type":"pairing"}
Zigbee2MQTT:error 2021-12-29 11:23:50: Not connected to MQTT server!
Zigbee2MQTT:error 2021-12-29 11:23:50: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":"announce","meta":{"friendly_name":"0xa4c13876161f82b5"},"type":"device_announced"}
Zigbee2MQTT:info  2021-12-29 11:23:54: Successfully interviewed '0xa4c13876161f82b5', device has successfully been paired
Zigbee2MQTT:warn  2021-12-29 11:23:54: Device '0xa4c13876161f82b5' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_gjnozsaz' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:error 2021-12-29 11:23:54: Not connected to MQTT server!
Zigbee2MQTT:error 2021-12-29 11:23:54: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":"interview_successful","meta":{"friendly_name":"0xa4c13876161f82b5","supported":false},"type":"pairing"}

```

after docker pull:

```
Zigbee2MQTT:info  2021-12-29 12:54:44: 0xa4c13876161f82b5 (0xa4c13876161f82b5): TS011F_plug_3 - TuYa Smart plug (with power monitoring by polling) (Router)

Zigbee2MQTT:info  2021-12-29 12:54:45: Configuring '0xa4c13876161f82b5'

Zigbee2MQTT:info  2021-12-29 12:54:45: Successfully configured '0xa4c13876161f82b5'

Zigbee2MQTT:info  2021-12-29 12:55:45: MQTT publish: topic 'zigbee2mqtt/0xa4c13876161f82b5', payload '{"current":0.43,"linkquality":39,"power":61,"voltage":233}'

Zigbee2MQTT:info  2021-12-29 12:55:45: MQTT publish: topic 'zigbee2mqtt/0xa4c13876161f82b5', payload '{"current":0.43,"energy":0.04,"linkquality":39,"power":61,"voltage":233}'
```

porzino, 2021-12-29:
> 

What is your setup? 
I have sonoff zigbee 3.0 dongle with firmware 20210708, zigbee2mqtt add-on in 1.22.1-1 version and the smart plug with applicationVersion = 68.

danielvandenberg95, 2021-12-29:
Raspberry PI with C2531, and if I remember correctly the firmware was already installed. Zigbee2mqtt pulled from docker-hub, tag latest-dev (!).

I'd assume it's fixed on the dev branch or so.

porzino, 2021-12-29:
I have installed Zigbee2MQTT version 1.22.1-dev (commit #73a6ad4) and now it works fine.

Thanks for your help!!!

> Raspberry PI with C2531, and if I remember correctly the firmware was already installed. Zigbee2mqtt pulled from docker-hub, tag latest-dev (!).
> 
> I'd assume it's fixed on the dev branch or so.

porzino, 2021-12-29:
This smart plug says that the model is NEO NAS-WR01B and zigbee2mqtt recognizes it as TuYa TS011F_plug_3.

Zigbee2MQTT should not recognize it as TuYa TS011F_plug_1?
The difference is about the electric measurement.

Koenkk, 2021-12-29:
`_TZ3000_gjnozsaz` is already supported in the dev branch. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-29 by Koenkk

peetereczek, 2021-12-29:
@Koenkk Hmm, I'm on Edge version of HA add-on, which is supposed to be development branch. Why it's not recognized?

Koenkk, 2021-12-29:
@peetereczek make sure you are on the latest dev branch, uninstall the addon and install the addon again.

porzino, 2021-12-29:
@Koenkk just one question.

Zigbee2MQTT should not recognize it as TuYa TS011F_plug_1? My smart plug says NEO NAS-WR01B.

Koenkk, 2021-12-29:
@porzino it is the same hardware, `NAS-WR01B` is a whitelabel of it.

porzino, 2021-12-29:
> @Koenkk Hmm, I'm on Edge version of HA add-on, which is supposed to be development branch. Why it's not recognized?

I have installed the edge version this afternoon and it works. Try to reinstall again.

Zigbee2MQTT version 1.22.1-dev (commit #73a6ad4)

porzino, 2021-12-29:
> @porzino it is the same hardware, `NAS-WR01B` is a whitelabel of it.

@Koen Ok. I was saying it for the difference over the electrical measure.

peetereczek, 2021-12-29:
Okey, now it works perfectly. Thank you for your advice :)

hppyworld, 2022-01-06:
I am trying to make this work but I have this error in start up of zigbee2mqtt.
zigbee2mqtt    | Error: Cannot find module '../lib/exposes'

I have added new lines in configuration.yaml
  log_level: debug
external_converters:
  - tuya.js

after that my docker container fails to run with above error. can you please help ? what other configuration i need to add in configuration.yaml? 
Thanks

hppyworld, 2022-01-06:
> Okey, now it works perfectly. Thank you for your advice :)

@peetereczek Can you please share your configuration.yaml, i have issue zigbee2mqtt | Error: Cannot find module '../lib/exposes'.

Thanks

ivanleovitanza, 2022-01-20:
> @peetereczek I have installed the edge version this afternoon and it works. Try to reinstall again.
> 
> Zigbee2MQTT version 1.22.1-dev (commit #73a6ad4)

Can you please share the config string? I keep seeing Unsupported