Koenkk/zigbee2mqtt #10408: Make the colorMode attribute (8) available

asgothian, 2021-12-28:
### Is your feature request related to a problem? Please describe

Restoring a previous state of lights which support different color modes (color temp vs hsv or RGB) is hampered by being unable to check the current color mode of the device before setting either the ct attribute or the color attribute(s) to bring the light back to its former state. 

### Describe the solution you'd like

It seems that the zigbee cluster lightingColorCtrl (768) provides an attribute colorMode (8) which, if it can be queried or is either set to be reported by the lamp or queried after changing color related items can provide said information.

A test of a number of lights with color capabilities was successful as in all lights which supported color did support reading said attribute. Note that the sample size itself was low (10 different device types)

An ideal case has an exposed datapoint which at all points correctly reports the current color mode of the lamp, allowing the client application to change behavior depending on the lights mode, e.g. change the UI to reflect the mode the light is in.

### Describe alternatives you've considered

The ability to query the color mode as device specific configuration would be an alternative which - while not as powerful as the automatic exposed datapoint - would still allow the client application to check the color modes and change UI / operation modes as appropriate for the light in question.

### Additional context

./.

sjorge, 2021-12-28:
You can already read this attribute for bulbs that support it, when setting colors using z2m it is also updated.

If you want it to be updated when you use a remote to change the color, that also works if you configure reporting. (Well for Tradfri and Innr at least, Hue does not allow you to configure reporting for that attribute)

asgothian, 2021-12-29:
Thank you for the information. I could not find this in the documentation, so i was unaware that it was already possible.

I will take a closer look at it.

closed 2021-12-29 by asgothian