Koenkk/zigbee2mqtt #10412: [Problem]: supervisor warning, unknown options

tetsuo55, 2021-12-28:
### What happened?

My supervisor log has a lot of copies of this message:
I've updated to latest home assistant and Zigbee2mqtt

21-12-28 18:16:17 WARNING (MainThread) [supervisor.addons.options] Unknown option 'new_api' for Zigbee2mqtt (45df7312_zigbee2mqtt)
21-12-28 18:16:17 WARNING (MainThread) [supervisor.addons.options] Unknown option 'restartdelay' for Zigbee2mqtt (45df7312_zigbee2mqtt)
21-12-28 18:16:17 WARNING (MainThread) [supervisor.addons.options] Unknown option 'initialdelay' for Zigbee2mqtt (45df7312_zigbee2mqtt)

### What did you expect to happen?

No warnings

### How to reproduce it (minimal and precise)

Open supervisor logs.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20201026

### Adapter

slaesh

### Debug log

Not applicable

Koenkk, 2021-12-29:
Should be fixed by removing these options from the configuration.

tetsuo55, 2021-12-29:
> Should be fixed by removing these options from the configuration.

I reset the settings to default and accidentally lost my network.
However i noticed in the documentation it tells me to add the new API thing.

Also setting availability to false doesn't disable it

Koenkk, 2021-12-29:
> However i noticed in the documentation it tells me to add the new API thing.

`new_api` isn't needed anymore, could you link me these docs?

> Also setting availability to false doesn't disable it
I guess you think this because `Using experimental new availability feature`, this was a bug and is fixed in the next z2m version (only a cosmetic thing, availability is disbabled when it is set to false)

tetsuo55, 2021-12-29:
> `new_api` isn't needed anymore, could you link me these docs?

Ok thanks for the info.

Here is the documentation https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/blob/b870fd2c0b92ee4944659c403fc0bf739fbdd6af/zigbee2mqtt/DOCS.md?plain=1#L11

Koenkk, 2021-12-29:
Thanks, updated

tetsuo55, 2021-12-29:
Awesome thanks

closed 2021-12-29 by tetsuo55