Koenkk/zigbee2mqtt #10415: [Problem]: Unable to pair  Miboxer FUT038Z

cndian, 2021-12-28:
### What happened?

I want to pair the  Miboxer FUT038Z [but](https://www.zigbee2mqtt.io/devices/FUT038Z.html) I get this error;

` 2021-12-28 19:06:33.507 Zigbee2mqtt: (Zigbee2mqtt) Received available devices list from bridge
2021-12-28 19:06:33.515 Zigbee2mqtt: (Zigbee2mqtt) 0x84fd27fffe635fba: device definiton not found, skipping...
2021-12-28 19:06:33.680 Zigbee2mqtt: (Zigbee2mqtt) Received available devices list from bridge
2021-12-28 19:06:33.688 Zigbee2mqtt: (Zigbee2mqtt) 0x84fd27fffe635fba: device definiton not found, skipping...

2021-12-28 19:06:33.508 Error: Zigbee2mqtt: (Zigbee2mqtt) Lamp voor: can not process numeric item "color_temp_startup"
2021-12-28 19:06:33.508 Error: Zigbee2mqtt: (Zigbee2mqtt) Lamp voor: can not process feature type "composite"
2021-12-28 19:06:33.509 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot motor rechts: can not process feature type "composite"
2021-12-28 19:06:33.509 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 3: can not process numeric item "color_temp_startup"
2021-12-28 19:06:33.509 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 3: can not process feature type "composite"
2021-12-28 19:06:33.509 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 4: can not process numeric item "color_temp_startup"
2021-12-28 19:06:33.509 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 4: can not process feature type "composite"
2021-12-28 19:06:33.510 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot motor midden: can not process feature type "composite"
2021-12-28 19:06:33.510 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 1: can not process numeric item "color_temp_startup"
2021-12-28 19:06:33.510 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 1: can not process feature type "composite"
2021-12-28 19:06:33.510 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 2: can not process numeric item "color_temp_startup" `

Even I get every time the other errors like color temp start up.

I do see the Miboxer in the Zigbee Tab, but there are no switches or LQ or whatever

### What did you expect to happen?

I hope to control my LED strip with this controller.

### How to reproduce it (minimal and precise)

Everytime when I want to pair the device

### Zigbee2MQTT version

v.1.22.1 

### Adapter firmware version

zStack3x0 20200925

### Adapter

Slaesh 2652

### Debug log

` 2021-12-28 19:06:33.507 Zigbee2mqtt: (Zigbee2mqtt) Received available devices list from bridge
2021-12-28 19:06:33.515 Zigbee2mqtt: (Zigbee2mqtt) 0x84fd27fffe635fba: device definiton not found, skipping...
2021-12-28 19:06:33.680 Zigbee2mqtt: (Zigbee2mqtt) Received available devices list from bridge
2021-12-28 19:06:33.688 Zigbee2mqtt: (Zigbee2mqtt) 0x84fd27fffe635fba: device definiton not found, skipping...

2021-12-28 19:06:33.508 Error: Zigbee2mqtt: (Zigbee2mqtt) Lamp voor: can not process numeric item "color_temp_startup"
2021-12-28 19:06:33.508 Error: Zigbee2mqtt: (Zigbee2mqtt) Lamp voor: can not process feature type "composite"
2021-12-28 19:06:33.509 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot motor rechts: can not process feature type "composite"
2021-12-28 19:06:33.509 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 3: can not process numeric item "color_temp_startup"
2021-12-28 19:06:33.509 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 3: can not process feature type "composite"
2021-12-28 19:06:33.509 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 4: can not process numeric item "color_temp_startup"
2021-12-28 19:06:33.509 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 4: can not process feature type "composite"
2021-12-28 19:06:33.510 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot motor midden: can not process feature type "composite"
2021-12-28 19:06:33.510 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 1: can not process numeric item "color_temp_startup"
2021-12-28 19:06:33.510 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 1: can not process feature type "composite"
2021-12-28 19:06:33.510 Error: Zigbee2mqtt: (Zigbee2mqtt) Spot keuken 2: can not process numeric item "color_temp_startup" `

Koenkk, 2021-12-29:
- Can you control it from the frontend?
- Where does this logging come from?

cndian, 2021-12-29:
Hello,

I can't control it. The log is from Domoticz.

When I go to the Zigbee tab in Domoticz I do see the Miboxer by Devices. But when I click on it I see no tabel devices below like link qualitie, switch, color etc.

cndian, 2021-12-29:
[image]

And this is all I got when I want to pair it, I get no message that's connected succesfully

Koenkk, 2021-12-29:
Please provide the zigbee2mqtt debug log when trying to pair the device.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

cndian, 2021-12-30:
` 2021-12-30 14:09:44.646 Zigbee2mqtt: (Zigbee2mqtt) Coordinator address is 0x00124b0021cbef2f
2021-12-30 14:09:44.646 Zigbee2mqtt: (Zigbee2mqtt) {"access": 7, "description": "Color temperature after cold power on of this light", "name": "color_temp_startup", "presets": [{"description": "Coolest temperature supported", "name": "coolest", "value": 150}, {"description": "Cool temperature (250 mireds / 4000 Kelvin)", "name": "cool", "value": 250}, {"description": "Neutral temperature (370 mireds / 2700 Kelvin)", "name": "neutral", "value": 370}, {"description": "Warm temperature (454 mireds / 2200 Kelvin)", "name": "warm", "value": 454}, {"description": "Warmest temperature supported", "name": "warmest", "value": 500}, {"description": "Restore previous color_temp on cold power on", "name": "previous", "value": 65535}], "property": "color_temp_startup", "type": "numeric", "unit": "mired", "value_max": 500, "value_min": 150}
2021-12-30 14:09:44.647 Zigbee2mqtt: (Zigbee2mqtt) {"description": "Color of this light expressed as hue/saturation", "features": [{"access": 7, "name": "hue", "property": "hue", "type": "numeric"}, {"access": 7, "name": "saturation", "property": "saturation", "type": "numeric"}], "name": "color_hs", "property": "color", "type": "composite"}
2021-12-30 14:09:44.647 Zigbee2mqtt: (Zigbee2mqtt) {"description": "Color of this light expressed as hue/saturation", "features": [{"access": 7, "name": "hue", "property": "hue", "type": "numeric"}, {"access": 7, "name": "saturation", "property": "saturation", "type": "numeric"}], "name": "color_hs", "property": "color", "type": "composite"}
2021-12-30 14:09:44.647 Zigbee2mqtt: (Zigbee2mqtt) {"access": 7, "description": "Color temperature after cold power on of this light", "name": "color_temp_startup", "presets": [{"description": "Coolest temperature supported", "name": "coolest", "value": 150}, {"description": "Cool temperature (250 mireds / 4000 Kelvin)", "name": "cool", "value": 250}, {"description": "Neutral temperature (370 mireds / 2700 Kelvin)", "name": "neutral", "value": 370}, {"description": "Warm temperature (454 mireds / 2200 Kelvin)", "name": "warm", "value": 454}, {"description": "Warmest temperature supported", "name": "warmest", "value": 500}, {"description": "Restore previous color_temp on cold power on", "name": "previous", "value": 65535}], "property": "color_temp_startup", "type": "numeric", "unit": "mired", "value_max": 500, "value_min": 150}
2021-12-30 14:09:44.647 Zigbee2mqtt: (Zigbee2mqtt) {"description": "Color of this light expressed as hue/saturation", "features": [{"access": 7, "name": "hue", "property": "hue", "type": "numeric"}, {"access": 7, "name": "saturation", "property": "saturation", "type": "numeric"}], "name": "color_hs", "property": "color", "type": "composite"}
2021-12-30 14:09:44.648 Zigbee2mqtt: (Zigbee2mqtt) {"access": 7, "description": "Color temperature after cold power on of this light", "name": "color_temp_startup", "presets": [{"description": "Coolest temperature supported", "name": "coolest", "value": 150}, {"description": "Cool temperature (250 mireds / 4000 Kelvin)", "name": "cool", "value": 250}, {"description": "Neutral temperature (370 mireds / 2700 Kelvin)", "name": "neutral", "value": 370}, {"description": "Warm temperature (454 mireds / 2200 Kelvin)", "name": "warm", "value": 454}, {"description": "Warmest temperature supported", "name": "warmest", "value": 500}, {"description": "Restore previous color_temp on cold power on", "name": "previous", "value": 65535}], "property": "color_temp_startup", "type": "numeric", "unit": "mired", "value_max": 500, "value_min": 150}
2021-12-30 14:09:44.648 Zigbee2mqtt: (Zigbee2mqtt) {"description": "Color of this light expressed as hue/saturation", "features": [{"access": 7, "name": "hue", "property": "hue", "type": "numeric"}, {"access": 7, "name": "saturation", "property": "saturation", "type": "numeric"}], "name": "color_hs", "property": "color", "type": "composite"}
2021-12-30 14:09:44.648 Zigbee2mqtt: (Zigbee2mqtt) {"description": "Color of this light expressed as hue/saturation", "features": [{"access": 7, "name": "hue", "property": "hue", "type": "numeric"}, {"access": 7, "name": "saturation", "property": "saturation", "type": "numeric"}], "name": "color_hs", "property": "color", "type": "composite"}
2021-12-30 14:09:44.649 Zigbee2mqtt: (Zigbee2mqtt) {"access": 7, "description": "Color temperature after cold power on of this light", "name": "color_temp_startup", "presets": [{"description": "Coolest temperature supported", "name": "coolest", "value": 150}, {"description": "Cool temperature (250 mireds / 4000 Kelvin)", "name": "cool", "value": 250}, {"description": "Neutral temperature (370 mireds / 2700 Kelvin)", "name": "neutral", "value": 370}, {"description": "Warm temperature (454 mireds / 2200 Kelvin)", "name": "warm", "value": 454}, {"description": "Warmest temperature supported", "name": "warmest", "value": 500}, {"description": "Restore previous color_temp on cold power on", "name": "previous", "value": 65535}], "property": "color_temp_startup", "type": "numeric", "unit": "mired", "value_max": 500, "value_min": 150}
2021-12-30 14:09:44.649 Zigbee2mqtt: (Zigbee2mqtt) {"description": "Color of this light expressed as hue/saturation", "features": [{"access": 7, "name": "hue", "property": "hue", "type": "numeric"}, {"access": 7, "name": "saturation", "property": "saturation", "type": "numeric"}], "name": "color_hs", "property": "color", "type" [... cut ...]
2021-12-30 14:09:46.184 Error: Zigbee2mqtt: (Zigbee2mqtt) 0x0017880109209f82: can not process feature type "composite"
2021-12-30 14:09:46.185 Error: Zigbee2mqtt: (Zigbee2mqtt) 0x00178801085be83e: can not process numeric item "color_temp_startup"
2021-12-30 14:09:46.185 Error: Zigbee2mqtt: (Zigbee2mqtt) 0x00178801085be83e: can not process feature type "composite"
2021-12-30 14:09:46.185 Error: Zigbee2mqtt: (Zigbee2mqtt) Lamp computerkamer: can not process numeric item "color_temp_startup"
2021-12-30 14:09:46.186 Error: Zigbee2mqtt: (Zigbee2mqtt) Lamp computerkamer: can not process feature type "composite" `

cndian, 2021-12-30:
` 2021-12-30 14:09:46.109 Zigbee2mqtt: (Zigbee2mqtt) 0x84fd27fffe635fba: device definiton not found, skipping...
2021-12-30 14:09:46.123 Zigbee2mqtt: (Zigbee2mqtt) MqttClient::publish zigbee2mqtt/bridge/config/devices/get ()
2021-12-30 14:09:46.178 Zigbee2mqtt: (Zigbee2mqtt) Received available devices list from bridge
2021-12-30 14:09:46.178 Zigbee2mqtt: (Zigbee2mqtt) Coordinator address is 0x00124b0021cbef2f
2021-12-30 14:09:46.179 Zigbee2mqtt: (Zigbee2mqtt) {"access": 7, "description": "Color temperature after cold power on of this light", "name": `

Koenkk, 2021-12-30:
I don't recognize this logging, it does not come from zigbee2mqtt. You can find the log in the zigbee2mqtt directory under `data/log`. Also what does your setup look like?

cndian, 2022-01-09:
I finally found the log file; I'm using Domoticz and the Slaesh stick on a Raspberry PI

`info  2022-01-09 10:55:09: Device '0x84fd27fffe635fba' joined
info  2022-01-09 10:55:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84fd27fffe635fba","ieee_address":"0x84fd27fffe635fba"},"type":"device_joined"}'
info  2022-01-09 10:55:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x84fd27fffe635fba"},"type":"device_connected"}'
info  2022-01-09 10:55:10: Starting interview of '0x84fd27fffe635fba'
info  2022-01-09 10:55:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84fd27fffe635fba","ieee_address":"0x84fd27fffe635fba","status":"started"},"type":"device_interview"}'
info  2022-01-09 10:55:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x84fd27fffe635fba"},"type":"pairing"}'
info  2022-01-09 10:55:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x84fd27fffe635fba","ieee_address":"0x84fd27fffe635fba"},"type":"device_announce"}'
info  2022-01-09 10:55:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x84fd27fffe635fba"},"type":"device_announced"}'

info  2022-01-09 10:55:11: Successfully interviewed '0x84fd27fffe635fba', device has successfully been paired
warn  2022-01-09 10:55:11: Device '0x84fd27fffe635fba' with Zigbee model 'TS0503B' and manufacturer name '_TZ3210_778drfdt' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-01-09 10:55:11: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x84fd27fffe635fba","ieee_address":"0x84fd27fffe635fba","status":"successful","supported":false},"type":"device_interview"}'
info  2022-01-09 10:55:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x84fd27fffe635fba","supported":false},"type":"pairing"}'`

Koenkk, 2022-01-09:
Your device was not supported yet, added it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-09 by Koenkk