Koenkk/zigbee2mqtt #10417: [Problem]: Unable to bind Tradfri remote E1810 to Ikea LED bulb

Stamper-repo, 2021-12-28:
### What happened?

Setup a new cluster, paired the tradfri remote and bulb and tried to bind the remote to the bulb. Unfortunately a non-descriptive error arose.

### What did you expect to happen?

Bulb and remote are successfully bound so the remote may control the bulb without the presence of zigbee2mqtt.

### How to reproduce it (minimal and precise)

1. Setup new cluster
2. Pair remote control E1810 (fw version 2.3.080)
3. Pair remote lamp (fw version 2.3.086)
4. Send MQTT message to bind remote control to remote lamp ` mosquitto_pub -t "zigbee2mqtt/bridge/request/device/bind" -h localhost -p 1883 -m '{"from": "<remote>", "to": "<lamp>"}'`
5. See error: `Failed to bind cluster 'genOnOff' from '<remote>' to '<bulb>' (Error: Bind <remote>/1 genLevelCtrl from '<bulb>/1' failed (AREQ - ZDO - bindRsp after 10000ms))`

### Zigbee2MQTT version

1.22.1 (commit #c2b5229)

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7786286/log.txt)

Stamper-repo, 2021-12-28:
Possibly related to https://github.com/Koenkk/zigbee2mqtt/issues/10352 ?

Koenkk, 2021-12-29:
Make sure to wakeup the remote right before sending a command by clicking a button on it.

Stamper-repo, 2021-12-29:
@Koenkk  That worked perfectly, thank you for the quick assist! I created a pull request to add this helpful tip: https://github.com/Koenkk/zigbee2mqtt.io/pull/1096

closed 2021-12-29 by Stamper-repo