Koenkk/zigbee2mqtt #10422: [Problem]: TS0601_thermostat stopped synchronizing time. In automatic mode (shedule) ea different batteries at different times.

iksrd, 2021-12-29:
### What happened?

TS0601_thermostat stopped synchronizing time. In automatic mode (shedule) ea different batteries at different times. If the time is set manually, it is reset again when syncing with zigbee2mqtt

### What did you expect to happen?

from 22.00-04.00 22 S from 04.00-22.00 21 S. This is set by shedule in auto mode. I was waiting for the time to be constant and the thermostat to set the temperature correctly by time

### How to reproduce it (minimal and precise)

Set the time on the thermostat manually, set up a schedule with different temperatures by the hour, check compliance with the set temperature depending on the time

### Zigbee2MQTT version

1.22.1-dev commit: b058517

### Adapter firmware version

 20210320

### Adapter

zStack3x0

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7788157/log.txt)
[log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/7788158/log.1.txt)

mrybas, 2021-12-29:
i have the same issue

WildeRNS, 2021-12-29:
Also have this issue

Minims, 2022-01-02:
Same issue with last zigbee2mqtt version

Ivano62, 2022-01-02:
I've the same problem on BRT-100-TRV thermostat (TS0601 - _TZE200_b6wax7g0) with zigbee2mqtt v1.22.2 - RaspBee 2.
Thanks

groenmarsmannetje, 2022-01-02:
Seems that this was an early problem report on edge version for this same issue in latest release #10489

Koenkk, 2022-01-02:
@groenmarsmannetje seems indeed the same, lets continue in #10489

closed 2022-01-02 by Koenkk