Koenkk/zigbee2mqtt #10424: Wrong state in HA

ShayGus, 2021-12-29:
### What happened?

The problem
Version 2021.12.6
Zegbee2MQTT
Tasmota flashed sonoff ZigBee brdge as a coordinator.

Since update to 2021.12.6 it's only possible to turn ZigBee switches on.
HA stopped reporting the correct state of the switch, when a switch is turned on, it immediately goes back to off on the HA dashboard, while reporting the correct "ON" state in zigbee2mqtt.
Restarting z2m or hass.io didn't fix the issue.
Downgrading to version 2021.12.5 fixed this behavior.

What version of Home Assistant Core has the issue?
2021.12.6

What was the last working version of Home Assistant Core?
2021.12.5

What type of installation are you running?
Home Assistant OS

Integration causing the issue
Zigbee2MQTT

### What did you expect to happen?

I expect switches to show the correct state.
Also to be able to turn off the switches.

### How to reproduce it (minimal and precise)

upgrade to HA 2021.12.6 
Try to operate switches

### Zigbee2MQTT version

1.22.1-1

### Adapter firmware version

6.7.9.0 build 405

### Adapter

sonoff zigbee bridge with tasmota

### Debug log

Unable to provide, downgraded HA.

Koenkk, 2021-12-29:
Please provide the debug log when controlling the switch.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

daballiemo, 2021-12-31:
This is an issue with HA 2021.12.6; please upgrade to 2021.12.7 which was released 2 days ago:

https://github.com/home-assistant/frontend/issues/11039#issuecomment-1002309735

closed 2022-02-07 by github-actions[bot]