Koenkk/zigbee2mqtt #10429: [Problem]: Tuya temperature/Humidity sensor not detected correctly

HA-TB303, 2021-12-29:
### What happened?

When adding this sensor it get's detected as a  TuYa TS0201
[image]
But it is a different type Tuya Zigbee 3.0 Temperature/humidity sensor:
[image]

### What did you expect to happen?

I was not expecting it to show up at all, since it is not supported yet.

### How to reproduce it (minimal and precise)

Pair the device.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20210708

### Adapter

Slaesch zStack3x0

### Debug log

Info Device '0xa4c138ff35516d72' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138ff35516d72","ieee_address":"0xa4c138ff35516d72"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xa4c138ff35516d72"},"type":"device_connected"}'
Debug Device '0xa4c138ff35516d72' announced itself
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138ff35516d72","ieee_address":"0xa4c138ff35516d72"},"type":"device_announce"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xa4c138ff35516d72"},"type":"device_announced"}'
Debug Retrieving state of '0xa4c138ff35516d72' after reconnect
Debug Received Zigbee message from '0xa4c138ff35516d72', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2525}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/0xa4c138ff35516d72', payload '{"battery":null,"device":{"applicationVersion":69,"dateCode":"","friendlyName":"0xa4c138ff35516d72","hardwareVersion":1,"ieeeAddr":"0xa4c138ff35516d72","manufacturerID":4417,"manufacturerName":"_TZ3000_bguser20","model":"TS0201","networkAddress":53599,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"humidity":null,"linkquality":105,"temperature":25.25,"voltage":null}'
Debug Received Zigbee message from '0xa4c138ff35516d72', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5554}' from endpoint 1 with groupID 0

Koenkk, 2021-12-29:
Please provide a link to this device + model number

oh2xx, 2021-12-29:
> Please provide a link to this device + model number

https://expo.tuya.com/product/897026

https://zigbee.blakadder.com/Tuya_WSD500A.html 

I have these too but they were not detected/paired at all. 

My setup:
Zigbee2MQTT v. 1.22.1 (docker)
CC2531 Xiaomi with Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20211115.hex FW

I have also a Sonoff Zigbee 3.0 here but it seems to have other issues so I do not know if it is worth trying it out yet.

HA-TB303, 2021-12-29:
Thats it exactly!

Koenkk, 2021-12-29:
Added, picture will show up with new release.

@oh2xx if nothing is logged when trying to pair make sure to check the points from https://www.zigbee2mqtt.io/guide/faq/#why-does-my-device-not-or-fail-to-pair

closed 2021-12-29 by Koenkk

jeykodev, 2021-12-29:
I have the same sensor (_TZ3000_bguser20), so version without display. Anyway the sensor unfortunately do not send data often enough. Same as mentioned here: https://github.com/Koenkk/zigbee2mqtt/issues/4202

@HA-TB303 do you experience the same?

HA-TB303, 2021-12-29:
I will test it out tomorrow.

HA-TB303, 2021-12-29:
It sends temperature and humidity data every 20 mins, which seems quit acceptable:

[image]

oh2xx, 2021-12-30:
> 

> Added, picture will show up with new release. (1 January)
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)
> 
> @oh2xx if nothing is logged when trying to pair make sure to check the points from https://www.zigbee2mqtt.io/guide/faq/#why-does-my-device-not-or-fail-to-pair

Yes, extension cable solved the pairing problem for me. I had 10 devices paired before so this was the first one that needed something extra.

oh2xx, 2022-01-01:
> 

@Koenkk just FWIW. It seems that the extension cable did not actually solve the pairing problem after all. Pairing was successful after disconnecting/reconnecting the USB stick and restarting the zigbee2mqtt docker. Had to do that for my two other new TuYa thermometers (they all are of this same type).

jeykodev, 2022-01-01:
> It sends temperature and humidity data every 20 mins, which seems quit acceptable:

Your graph more looks like a update interval of 30 minutes. I can confirm 30 minutes for all my devices too: 

[image]

Here is an interesting note from the manual: 

[image]

So for my understanding the device internally checks the sensor values every 5 minutes. If no significant changes were detected, it will send the data after 30 minutes. 

I planned to use the data for climate automation, with this high reporting interval it wont work. I think it will be a show stopper for a lot of usecases. 

Are there any chances to change this behaviour based on software? I saw other hardware related workarounds on radiator valves, where people "simulated" pressing the pairing button, to force data to be send.