Koenkk/zigbee2mqtt #10431: [Problem]: TS004F not recognised

Mearman, 2021-12-29:
### What happened?

I have a [TuYa TS004F](https://www.zigbee2mqtt.io/devices/TS004F.html) which will pair but is not recognised. The serial number and model match.

### What did you expect to happen?

Should be paired and recognised

### How to reproduce it (minimal and precise)

Attempt to pair the TS004F found here: [4 Gang Wireless Scene Switch](https://amzn.to/3sIuAn0)

### Zigbee2MQTT version

1.18.1-1

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Debug log

```
Zigbee2MQTT:warn  2021-12-29 15:07:35: Received message from unsupported device with Zigbee model 'TS004F' and manufacturer name '_TZ3000_xabckq1v'
Zigbee2MQTT:warn  2021-12-29 15:07:35: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

Koenkk, 2021-12-29:
Already supported in the latest z2m version. 

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2021-12-29 by Koenkk

Mearman, 2021-12-29:
Ah it looks like I was pointing at an old repo. Just switching over now.