Koenkk/zigbee2mqtt #10436: [Problem]: MQTT does not retry connection on startup using npm start on Windows

cvb941, 2021-12-29:
### What happened?

When starting up the program using npm start on Windows and my mosquito container running on Docker is not ready yet, the script exits.

### What did you expect to happen?

The script should retry the connection to the MQTT broker, similar to what running in a docker container with restart=always does.

### How to reproduce it (minimal and precise)

Use zigbee2mqtt on windows using npm, attempt to start without a running MQTT broker.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20190608

### Adapter

CC2530/CC2531

### Debug log

debug 2021-12-29 19:48:02: Loaded state from file C:\Server\zigbee2mqtt\data\state.json
info  2021-12-29 19:48:02: Logging to console and directory: 'C:\Server\zigbee2mqtt\data\log\2021-12-29.19-48-01' filename: log.txt
debug 2021-12-29 19:48:02: Removing old log directory 'C:\Server\zigbee2mqtt\data\log\2021-12-20.20-45-38'
info  2021-12-29 19:48:02: Starting Zigbee2MQTT version 1.22.1 (commit #89996835)
info  2021-12-29 19:48:02: Starting zigbee-herdsman (0.13.176)
debug 2021-12-29 19:48:02: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"C:\\Server\\zigbee2mqtt\\data\\coordinator_backup.json","databaseBackupPath":"C:\\Server\\zigbee2mqtt\\data\\database.db.backup","databasePath":"C:\\Server\\zigbee2mqtt\\data\\database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"\\\\.\\COM3"}}'
info  2021-12-29 19:48:03: zigbee-herdsman started (resumed)
info  2021-12-29 19:48:03: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
debug 2021-12-29 19:48:03: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
info  2021-12-29 19:48:03: Currently 2 devices are joined:
info  2021-12-29 19:48:03: Barborkine Stlacitko (0x00158d0004188250): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
info  2021-12-29 19:48:03: Vianocny Stromcek (0x540f57fffef92262): TS0121_plug - TuYa 10A UK or 16A EU smart plug (Router)
warn  2021-12-29 19:48:03: `permit_join` set to  `true` in configuration.yaml.
warn  2021-12-29 19:48:03: Allowing new devices to join.
warn  2021-12-29 19:48:03: Set `permit_join` to `false` once you joined all devices.
info  2021-12-29 19:48:03: Zigbee: allowing new devices to join.
info  2021-12-29 19:48:03: Connecting to MQTT server at mqtt://localhost
debug 2021-12-29 19:48:03: Using MQTT login with username: cvb941
error 2021-12-29 19:48:03: MQTT failed to connect: connect ECONNREFUSED 127.0.0.1:1883
error 2021-12-29 19:48:03: Exiting...
info  2021-12-29 19:48:03: Stopping zigbee-herdsman...
info  2021-12-29 19:48:04: Stopped zigbee-herdsman

Koenkk, 2021-12-29:
This has been explicitly chosen for (see https://github.com/Koenkk/zigbee2mqtt/issues/8956). Either add a `depends_on` dependency in the docker-compose file or run the container with `--restart=unless=stopped`

cvb941, 2021-12-29:
Well I'm not running zigbee2mqtt in docker, so I cannot add those anywhere.

Could there be a new setting in the configuration file that would allow a similar behaviour when running zigbee2mqtt using npm?

Koenkk, 2021-12-29:
> Could there be a new setting in the configuration file that would allow a similar behaviour when running zigbee2mqtt using npm?

I'll reopen this, lets see how many people are interested in such feature.

closed 2022-02-05 by github-actions[bot]