Koenkk/zigbee2mqtt #10448: [Problem]: LEDVANCE AC03360 (similar to LEDVANCE AC26940/AC31266) identified as OSRAM AB3257001NJ

nohn, 2021-12-30:
### What happened?

LEDVANCE AC03360 (that is similar to LEDVANCE AC26940/AC31266) is identified as OSRAM AB3257001NJ

### What did you expect to happen?

LEDVANCE AC03360 is identidied as LEDVANCE AC26940/AC31266/AC03360

### How to reproduce it (minimal and precise)

Pair a LEDVANCE AC03360

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20210708

### Adapter

Slaesh's CC2652RB Stick

### Debug log

```
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:53:57: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"06qqt-2","value":true}'
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:53:57: Zigbee: allowing new devices to join.
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:53:57: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"06qqt-2"}'
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:08: Device '0x7cb03eaa0a082233' reconnected
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:08: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa0a082233/availability', payload 'online'
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:08: Device '0x7cb03eaa0a082233' announced itself
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x7cb03eaa0a082233","ieee_address":"0x7cb03eaa0a082233"},"type":"device_announce"}'
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:24: Starting interview of '0x7cb03eaa0a082233'
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x7cb03eaa0a082233","ieee_address":"0x7cb03eaa0a082233","status":"started"},"type":"device_interview"}'
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:24: Device '0x7cb03eaa0a082233' announced itself
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x7cb03eaa0a082233","ieee_address":"0x7cb03eaa0a082233"},"type":"device_announce"}'
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:35: Received Zigbee message from '0x7cb03eaa0a082233', type 'readResponse', cluster 'genBasic', data '{"modelId":"Plug 01"}' from endpoint 3 with groupID 0
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:35: Skipping message, definition is undefined and still interviewing
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:36: Received Zigbee message from '0x7cb03eaa0a082233', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"OSRAM"}' from endpoint 3 with groupID 0
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:36: Skipping message, definition is undefined and still interviewing
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:36: Received Zigbee message from '0x7cb03eaa0a082233', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 3 with groupID 0
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:36: Skipping message, definition is undefined and still interviewing
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:36: Received Zigbee message from '0x7cb03eaa0a082233', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 3 with groupID 0
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:36: Skipping message, definition is undefined and still interviewing
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:36: Received Zigbee message from '0x7cb03eaa0a082233', type 'readResponse', cluster 'genBasic', data '{"appVersion":9}' from endpoint 3 with groupID 0
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:36: Skipping message, definition is undefined and still interviewing
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:37: Received Zigbee message from '0x7cb03eaa0a082233', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 3 with groupID 0
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:37: Skipping message, definition is undefined and still interviewing
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:37: Received Zigbee message from '0x7cb03eaa0a082233', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 3 with groupID 0
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:37: Skipping message, definition is undefined and still interviewing
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:37: Received Zigbee message from '0x7cb03eaa0a082233', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20140331DEOS****"}' from endpoint 3 with groupID 0
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:37: Skipping message, definition is undefined and still interviewing
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:37: Received Zigbee message from '0x7cb03eaa0a082233', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"V1.05.09"}' from endpoint 3 with groupID 0
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:37: Skipping message, definition is undefined and still interviewing
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:37: Successfully interviewed '0x7cb03eaa0a082233', device has successfully been paired
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:37: Device '0x7cb03eaa0a082233' is supported, identified as: OSRAM Smart+ plug (AB3257001NJ)
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart+ plug","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"AB3257001NJ","options":[],"supports_ota":true,"vendor":"OSRAM"},"friendly_name":"0x7cb03eaa0a082233","ieee_address":"0x7cb03eaa0a082233","status":"successful","supported":true},"type":"device_interview"}'
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:37: Configuring '0x7cb03eaa0a082233'
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:53:57: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"06qqt-2","value":true}'
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:53:57: Zigbee: allowing new devices to join.
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:53:57: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"06qqt-2"}'
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:08: Device '0x7cb03eaa0a082233' reconnected
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:08: MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa0a082233/availability', payload 'online'
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:08: Device '0x7cb03eaa0a082233' announced itself
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x7cb03eaa0a082233","ieee_address":"0x7cb03eaa0a082233"},"type":"device_announce"}'
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:24: Starting interview of '0x7cb03eaa0a082233'
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x7cb03eaa0a082233","ieee_address":"0x7cb03eaa0a082233","status":"started"},"type":"device_interview"}'
zigbee2mqtt                | Zigbee2MQTT:debug 2021-12-30 15:54:24: Device '0x7cb03eaa0a082233' announced itself
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:24: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x7cb03eaa0a082233","ieee_address":"0x7cb03eaa0a082233"},"type":"device_announce"}'
[... cut ...] itch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"AB3257001NJ","options":[],"supports_ota":true,"vendor":"OSRAM"},"friendly_name":"0x7cb03eaa0a082233","ieee_address":"0x7cb03eaa0a082233","status":"successful","supported":true},"type":"device_interview"}'
zigbee2mqtt                | Zigbee2MQTT:info  2021-12-30 15:54:37: Configuring '0x7cb03eaa0a082233'
```
[image]

Koenkk, 2021-12-30:
please provide the data/database.db entry of this device

nohn, 2021-12-30:
```
{
  "id": 20,
  "type": "Router",
  "ieeeAddr": "0x7cb03eaa0a082233",
  "nwkAddr": 41307,
  "manufId": 48042,
  "manufName": "OSRAM",
  "powerSource": "Mains (single phase)",
  "modelId": "Plug 01",
  "epList": [
    3
  ],
  "endpoints": {
    "3": {
      "profId": 49246,
      "epId": 3,
      "devId": 16,
      "inClusterList": [
        4096,
        0,
        3,
        4,
        5,
        6,
        2820,
        64527
      ],
      "outClusterList": [
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "Plug 01",
            "manufacturerName": "OSRAM",
            "powerSource": 1,
            "zclVersion": 1,
            "appVersion": 9,
            "stackVersion": 2,
            "hwVersion": 1,
            "dateCode": "20140331DEOS****",
            "swBuildId": "V1.05.09"
          }
        },
        "genOnOff": {
          "attributes": {
            "onOff": 1
          }
        }
      },
      "binds": [
        {
          "cluster": 6,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b002390c2bd",
          "endpointID": 1
        }
      ],
      "configuredReportings": [
        {
          "cluster": 6,
          "attrId": 0,
          "minRepIntval": 0,
          "maxRepIntval": 3600,
          "repChange": 0
        }
      ],
      "meta": {}
    }
  },
  "appVersion": 9,
  "stackVersion": 2,
  "hwVersion": 1,
  "dateCode": "20140331DEOS****",
  "swBuildId": "V1.05.09",
  "zclVersion": 1,
  "interviewCompleted": true,
  "meta": {
    "configured": 1
  },
  "lastSeen": 1640889832334,
  "useImplicitCheckin": true
}
```

Koenkk, 2021-12-31:
Looking at the data/database.db entry we cannot distinguish this device from the OSRAM `AB3257001NJ`, even the manufacturer name is `OSRAM`. Note that OSRAM is a brand of LEDVANCE so this is not really strange. With regards to the `AC26940/AC31266`, this device uses a different modelId (`Outdoor Plug` or `Plug Value`). I've added `AC03360` as a whitelabel of the `AB3257001NJ` so it will appear on the supported devices page.

closed 2021-12-31 by Koenkk