Koenkk/zigbee2mqtt #10449: CC2538+CC2592 not recognize CC2531 with ROUTER firmware

manny199, 2021-12-30:
### What happened?

I have a CC2538+CC2592 coordinator with firmware 20201010 and a CC2531 with router firmware CC2531_router_2020_09_29.
Unfortunately CC2538+CC2592 does not detect CC2531 router and not pair it.

### What did you expect to happen?

I'd like to understand what the problem could be. I want to pair my CC2351 router with CC2538+CC2592 coordinator

### How to reproduce it (minimal and precise)

I tried to change router firmware on the CC2531 but with no success

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20201010

### Adapter

CC2538+CC2592

### Debug log

Nothing in log because not recognized

kaweksl, 2022-01-03:
Did you use CC2531 as coordinator in same instance of zigbee2mqtt before ?

kaweksl, 2022-01-03:
Did you use CC2531 as coordinator in same instance of zigbee2mqtt before ?

Today i had issue because i wanted to use usb stick (egony) as router but before i have used it as coordinator ( replaced it with sonoff usb dongle ) . Turns out zigbee2mqtt is cloning ieee address to new coordinator so we won't need to re-pair  all devices after switching coordinator hardware. Downside is that now two devices have same ieee address. To resolve that i have assigned different secondary address to old usb stick and now it works as router.

closed 2022-02-10 by github-actions[bot]