Koenkk/zigbee2mqtt #10454: [New device support]: Philips Hue Motion Sensor Outdoor 929003067401

vkernelblog, 2021-12-30:
### Link

https://www.philips-hue.com/nl-nl/p/hue-buitensensor/8719514342262

### Database entry

{"id":45,"type":"EndDevice","ieeeAddr":"0x001788010bd109ec","nwkAddr":65292,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Battery","modelId":"SML004","epList":[2],"endpoints":{"2":{"profId":260,"epId":2,"devId":263,"inClusterList":[0,1,3,1030,1024,1026],"outClusterList":[25,0,3,6],"clusters":{"genBasic":{"attributes":{"modelId":"SML004","manufacturerName":"Signify Netherlands B.V.","powerSource":3,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20210830","swBuildId":"2.53.6"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20210830","swBuildId":"2.53.6","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1640893729815,"useImplicitCheckin":true}

### Comments

I am unable to add this device into zigbee2mqtt. 
Can you please help me to add this device to zigbee2mqtt

sjorge, 2021-12-30:
Seems to already be supported in the latest dev version: https://github.com/Koenkk/zigbee-herdsman-converters/blob/8b922b27dca90261d0d25216467f15d18661d999/devices/philips.js#L1926

You can wait for the next release or switch to the dev branch.

vkernelblog, 2021-12-30:
Thank you very much for the update

Sent from Nine<http://www.9folders.com/>

Koenkk, 2021-12-31:
FYI: next release is tomorrow

closed 2021-12-31 by Koenkk

pmknowles, 2022-01-03:
I have 2 which have paired fine (although one appears in the Map without a link to the co-ordinator). Worked on the old release and the new. Need to hold the button for about 15 seconds until green LED flashes