Koenkk/zigbee2mqtt #10455: [Problem]: Wrong device identification in dev branch - Nous E6

stomko11, 2021-12-30:
### What happened?

I am using Zigbee2MQTT HA plugin - the edge version. I am using Nous E6 Temp and humidity sensor with display which was added 16 days ago according to https://github.com/Koenkk/zigbee-herdsman-converters/tree/master/devices 
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3503

However when I pair the device, it is identified as TS0601 - _TZE200_nnrfa68v - Curtain motor/roller blind motor/window pusher/tubular motor

[image]

Appreciate the help. 

### What did you expect to happen?

To be identified as Nous E6 sensor

### How to reproduce it (minimal and precise)

Pair Nous E6 sensor

### Zigbee2MQTT version

1.22.1-dev commit: e8526b7

### Adapter firmware version

0x26700700

### Adapter

Conbee II

### Debug log

[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/7794589/debug.txt)

1igeza, 2021-12-30:
I have similar problem with tuya dimmer. It is also identified as Curtain motor.

- Friendly name
-     Sciemniacz lazienka
- Last seen
-     just now
- Availability
-     Online
- Device type
-     Router
- Zigbee Model
-     TS0601
- Zigbee Manufacturer
-     _TZE200_9i9dt8is
- Description
-     Curtain motor/roller blind motor/window pusher/tubular motor
- Support status
- 
-     Supported
- IEEE Address
-     0xbc33acfffee939ee
- Network address
-     0xB331
- Manufacturer
-     TuYa
- Model
-     TS0601_cover
- Power
- Interview completed
-     True

```
Zigbee2MQTT:debug 2021-12-30 22:11:20: Received Zigbee message from 'Sciemniacz lazienka', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-12-30 22:11:20: MQTT publish: topic 'zigbee2mqtt/Sciemniacz lazienka', payload '{"last_seen":"2021-12-30T22:11:20+01:00","linkquality":76,"position":10,"running":false,"state":"OPEN"}'
  zigbee-herdsman:controller:endpoint ReadResponse 0xbc33acfffee939ee/1 genTime({"localTime":694217481}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +2s
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xbc33acfffee939ee:65131/1 (0,0,1) +2s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":65131,"destendpoint":1,"srcendpoint":1,"clusterid":10,"transid":9,"options":0,"radius":30,"len":11,"data":{"type":"Buffer","data":[24,118,1,7,0,0,35,9,235,96,41]}} +2s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,21,36,1,107,254,1,1,10,0,9,0,30,11,24,118,1,7,0,0,35,9,235,96,41,83] +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,9,207] +40ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,9,207] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,9] - 207 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":9} +63ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,7,69,196,107,254,2,107,89,227,202,10] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,7,69,196,107,254,2,107,89,227,202,10] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 7 - 2 - 5 - 196 - [107,254,2,107,89,227,202] - 10 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":65131,"relaycount":2,"relaylist":[22891,51939]} +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,30,68,129,0,0,0,239,107,254,1,1,0,72,0,136,245,183,0,0,10,9,119,1,0,1,1,1,0,1,1,227,202,27,101] +49ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,30,68,129,0,0,0,239,107,254,1,1,0,72,0,136,245,183,0,0,10,9,119,1,0,1,1,1,0,1,1,227,202,27,101] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 30 - 2 - 4 - 129 - [0,0,0,239,107,254,1,1,0,72,0,136,245,183,0,0,10,9,119,1,0,1,1,1,0,1,1,227,202,27] - 101 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":65131,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":72,"securityuse":0,"timestamp":12055944,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[9,119,1,0,1,1,1,0,1,1]}} +49ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":119,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"seq":256,"dp":1,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[1]}},"Command":{"ID":1,"parameters":[{"name":"seq","type":33},{"name":"dp","type":32},{"name":"datatype","type":32},{"name":"fn","type":32},{"name":"data","type":65}],"name":"dataResponse"}},"address":65131,"endpoint":1,"linkquality":72,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}' +1s
Zigbee2MQTT:debug 2021-12-30 22:11:21: Received Zigbee message from 'Sciemniacz lazienka', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1,"fn":0,"seq":256}' from endpoint 1 with groupID 0
  zigbee-herdsman:controller:endpoint DefaultResponse 0xbc33acfffee939ee/1 61184(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +1s
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xbc33acfffee939ee:65131/1 (0,0,1) +1s
[... cut ...]
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":49,"manufacturerCode":null,"commandIdentifier":0},"Payload":[{"attrId":0}],"Command":{"ID":0,"name":"read","parameters":[{"name":"attrId","type":33}],"response":1}},"address":58552,"endpoint":1,"linkquality":156,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}' +3s

```

stomko11, 2021-12-30:
I tried to manually create js file based on https://github.com/Koenkk/zigbee-herdsman-converters/pull/3503 and configured it as external converter, that gives me proper names, fields, etc. However I am not able to see the reported data correctly. There is "N/A" for all (humidity, temperature, battery) and in logs, I see:
```
Debug Received Zigbee message from 'Nous E6', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,251],"type":"Buffer"},"datatype":2,"dp":1,"fn":0,"seq":44032}' from endpoint 1 with groupID null
Debug No converter available for 'E6' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"data":{"data":[0,0,0,251],"type":"Buffer"},"datatype":2,"dp":1,"fn":0,"seq":44032}'
Debug Received Zigbee message from 'Nous E6', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,47],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"seq":44288}' from endpoint 1 with groupID null
Debug No converter available for 'E6' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"data":{"data":[0,0,0,47],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"seq":44288}'
Debug Received Zigbee message from 'Nous E6', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9,"fn":0,"seq":44544}' from endpoint 1 with groupID null
Debug No converter available for 'E6' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9,"fn":0,"seq":44544}'
Debug Received Zigbee message from 'Nous E6', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9,"fn":0,"seq":44800}' from endpoint 1 with groupID null
Debug No converter available for 'E6' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":9,"fn":0,"seq":44800}'
Debug Received Zigbee message from 'Nous E6', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null
Debug Received Zigbee message from 'Nous E6', type 'attributeReport', cluster 'genBasic', data '{"65506":34,"65508":0,"appVersion":64}' from endpoint 1 with groupID null
Debug Received Zigbee message from 'Nous E6', type 'attributeReport', cluster 'genBasic', data '{"65503":"��`)i��`)\u0011��`)i��`)\u0011��`)\u0011��`)i��`)\u0011%�`)i+�`)\u0011��`)\u0011"}' from endpoint 1 with groupID null
Debug Received Zigbee message from 'Nous E6', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,100,36,176,0],"type":"Buffer"}' from endpoint 1 with groupID null
Debug No converter available for 'E6' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,100,36,176,0],"type":"Buffer"}'
Debug Received Zigbee message from 'Nous E6', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,101,36,177,0],"type":"Buffer"}' from endpoint 1 with groupID null
Debug No converter available for 'E6' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,101,36,177,0],"type":"Buffer"}'
```

Coninox, 2021-12-30:
I have the same problem with my Tuya radiator valves. They are seen as curtain motor.

csowada, 2021-12-31:
Same here with zwo devices:

_TZE200_d0yu2xgi (NEO Alarm sirene)
_TZE200_s8gkrkxk (Lidl Xmas Stripe)

Both reported as "TS0601_cover"

Koenkk, 2021-12-31:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

csowada, 2021-12-31:
Yes, it is fixed with latest docker container. After a restart both devices are back as usual.

closed 2021-12-31 by Koenkk