Koenkk/zigbee2mqtt #10462: Please add this device : NodOn SIN-4-RS-20

Balooforever, 2021-12-31:
### Link

https://nodon.fr/nodon/module-volet-roulant-zigbee/

### Database entry

{"id":105,"type":"Router","ieeeAddr":"0x540f57fffe44aeed","nwkAddr":52369,"manufId":4747,"manufName":"NodOn","powerSource":"Mains (single phase)","modelId":"SIN-4-RS-20\u0000\u0000\u0000\u0000\u0014\u0000\u0000\u0000�\u0000\u0000\u0000=\u0001\u0001\u0000\u0001\u0000\u0000��","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,3,4,5,258,4096,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SIN-4-RS-20\u0000\u0000\u0000\u0000\u0014\u0000\u0000\u0000�\u0000\u0000\u0000=\u0001\u0001\u0000\u0001\u0000\u0000��","manufacturerName":"NodOn","powerSource":1,"zclVersion":3,"appVersion":39,"stackVersion":103,"hwVersion":50,"dateCode":"2020"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":39,"stackVersion":103,"hwVersion":50,"dateCode":"2020","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1640944798672,"useImplicitCheckin":true}

### Comments

Hello,

Can you integrate this device please. Thanks :-)

Balooforever, 2022-01-01:
Hi @Koenkk 

Working with : 
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['SIN-4-RS-20'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SIN-4-RS-20', // Vendor model number, look on the device for a model number
    vendor: 'NodOn', // Vendor of the device (only used for documentation and startup logging)
    description: 'Nodon Roller Shutter', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.cover_position_tilt], // We will add this later
    toZigbee: [tz.cover_state, tz.cover_position_tilt], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
    },
    exposes: [e.cover_position()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;`

But in debug, a lot of stat reporting :
Received Zigbee message from '0x540f57fffe44aeed', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":23,"currentPositionTiltPercentage":24}' from endpoint 1 with groupID null

Koenkk, 2022-01-01:
@Balooforever thanks, added!

> But in debug, a lot of stat reporting :

How often?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Balooforever, 2022-01-01:
Every second my two device update here status (infos and debug).

i don t understand why ..

Koenkk, 2022-01-02:
Does it work correctly after that?

Balooforever, 2022-01-02:
Yep, no problem.
Can I "control" the "attributeReport" every second ?

Koenkk, 2022-01-02:
Could you provide me the debug log of those reports?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Balooforever, 2022-01-03:
Hello,

`Zigbee2MQTT:debug 2022-01-03 09:15:05: Received Zigbee message from 'Volet Avant', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionTiltPercentage":0}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2022-01-03 09:15:05: MQTT publish: topic 'zigbee2mqtt/Volet Avant', payload '{"linkquality":255,"position":0,"tilt":100}'
Zigbee2MQTT:debug 2022-01-03 09:15:10: Received Zigbee message from 'Volet Avant', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionTiltPercentage":0}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2022-01-03 09:15:10: MQTT publish: topic 'zigbee2mqtt/Volet Avant', payload '{"linkquality":255,"position":0,"tilt":100}'
Zigbee2MQTT:debug 2022-01-03 09:15:15: Received Zigbee message from 'Volet Avant', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionTiltPercentage":0}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2022-01-03 09:15:15: MQTT publish: topic 'zigbee2mqtt/Volet Avant', payload '{"linkquality":255,"position":0,"tilt":100}'
Zigbee2MQTT:debug 2022-01-03 09:15:20: Received Zigbee message from 'Volet Avant', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionTiltPercentage":0}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2022-01-03 09:15:20: MQTT publish: topic 'zigbee2mqtt/Volet Avant', payload '{"linkquality":244,"position":0,"tilt":100}'
Zigbee2MQTT:debug 2022-01-03 09:15:25: Received Zigbee message from 'Volet Avant', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionTiltPercentage":0}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2022-01-03 09:15:25: MQTT publish: topic 'zigbee2mqtt/Volet Avant', payload '{"linkquality":255,"position":0,"tilt":100}'`
...

Koenkk, 2022-01-03:
Can you try to add below `await reporting.currentPositionLiftPercentage(endpoint);` 👍 
```js
await reporting.currentPositionTiltPercentage(endpoint);
```

Balooforever, 2022-01-03:
Oups, my bad : work perfectly.

If i remove "await reporting.currentPositionLiftPercentage(endpoint);" : all work too (and no more log), i don't understand if needed.

Koenkk, 2022-01-03:
> i don't understand if needed.

it is needed for one time setup (so if you remove it afterwards the setup already took place).

added the fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-03 by Koenkk

PEGARifrouf, 2022-01-07:
Hello,

I had setup the device, do the calibration, add it trough external converters thanks to your last definition.
Il played with it, it can fully open, closed and do partial position with good estimation. Very responsive on commands (HA and hardwired switch).

But it often stay stuck after a move it need several tries of do a full closure before re-open. When it happens I hear it click no movement of the RS and the position seems to report fully close or open according to the command instantly. Same behavior when commanded manually with a 3 button switch (UP/STOP/DOWN).

Definitely not usable daily in this state.

Did you think this an HA fix of more Nodon firmware ? What can I do to investigate ?

See attached log
Best regards,

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7828661/log.txt)
.

PEGARifrouf, 2022-01-07:
In this sequence, the RS has never moved:

```
debug 2022-01-07 14:00:23: Received MQTT message on 'zigbee2mqtt/VR_salle_a_manger/set' with data '{ "position": 47 }'
debug 2022-01-07 14:00:23: Publishing 'set' 'position' to 'VR_salle_a_manger'
info  2022-01-07 14:00:23: MQTT publish: topic 'zigbee2mqtt/VR_salle_a_manger', payload '{"linkquality":75,"position":47,"tilt":0}'
debug 2022-01-07 14:00:25: Received Zigbee message from 'VR_salle_a_manger', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100,"currentPositionTiltPercentage":98}' from endpoint 1 with groupID 0
info  2022-01-07 14:00:25: MQTT publish: topic 'zigbee2mqtt/VR_salle_a_manger', payload '{"linkquality":72,"position":0,"tilt":2}'
debug 2022-01-07 14:00:30: Received MQTT message on 'zigbee2mqtt/VR_salle_a_manger/set' with data 'OPEN'
debug 2022-01-07 14:00:30: Publishing 'set' 'state' to 'VR_salle_a_manger'
debug 2022-01-07 14:00:32: Received Zigbee message from 'VR_salle_a_manger', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0,"currentPositionTiltPercentage":100}' from endpoint 1 with groupID 0
info  2022-01-07 14:00:32: MQTT publish: topic 'zigbee2mqtt/VR_salle_a_manger', payload '{"linkquality":72,"position":100,"tilt":0}'
debug 2022-01-07 14:00:43: Received MQTT message on 'zigbee2mqtt/VR_salle_a_manger/set' with data '{ "position": 54 }'
debug 2022-01-07 14:00:43: Publishing 'set' 'position' to 'VR_salle_a_manger'
info  2022-01-07 14:00:43: MQTT publish: topic 'zigbee2mqtt/VR_salle_a_manger', payload '{"linkquality":72,"position":54,"tilt":0}'
debug 2022-01-07 14:00:44: Received Zigbee message from 'VR_salle_a_manger', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":100,"currentPositionTiltPercentage":98}' from endpoint 1 with groupID 0
info  2022-01-07 14:00:44: MQTT publish: topic 'zigbee2mqtt/VR_salle_a_manger', payload '{"linkquality":78,"position":0,"tilt":2}'
debug 2022-01-07 14:00:48: Received MQTT message on 'zigbee2mqtt/VR_salle_a_manger/set' with data 'OPEN'
debug 2022-01-07 14:00:48: Publishing 'set' 'state' to 'VR_salle_a_manger'
debug 2022-01-07 14:00:49: Received Zigbee message from 'VR_salle_a_manger', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":0,"currentPositionTiltPercentage":100}' from endpoint 1 with groupID 0
info  2022-01-07 14:00:49: MQTT publish: topic 'zigbee2mqtt/VR_salle_a_manger', payload '{"linkquality":75,"position":100,"tilt":0}'
debug 2022-01-07 14:00:50: Received MQTT message on 'zigbee2mqtt/VR_salle_a_manger/set' with data 'STOP'
debug 2022-01-07 14:00:50: Publishing 'set' 'state' to 'VR_salle_a_manger'
```

MathieuDmbc, 2022-01-09:
Hi, will there be a possibility to set the tilt position for blinds in the future ? Curently I see I can only set the position with Z2M Edge. According to the manufacturer, the device should suit for venetian blinds also. Thanks for any advice :)