Koenkk/zigbee2mqtt #10464: Smoke sensor TS0601 (_TZE200_uebojraa) doesn't report smoke

Roelski, 2021-12-31:
I have 2 smoke sensors (TS0601 = _TZE200_uebojraa).
( https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/TS0601_smoke.md )

They are paired fine, but as far as i can test, they never report smoke.

Pressing test button on side gives:
zigbee2mqtt             | 2021-12-24T18:18:14.024254149Z Zigbee2MQTT:debug 2021-12-24 19:18:14: Received Zigbee message from 'Smoke-Woonkamer', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":11520}' from endpoint 1 with groupID null
zigbee2mqtt             | 2021-12-24T18:18:14.035912285Z Zigbee2MQTT:debug 2021-12-24 19:18:14: No converter available for 'TS0601_smoke' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":11520}'
zigbee2mqtt             | 2021-12-24T18:18:14.058850853Z Zigbee2MQTT:info  2021-12-24 19:18:14: MQTT publish: topic 'zigbee2mqtt/Smoke-Woonkamer', payload '{"battery_low":null,"last_seen":"2021-12-24T19:18:14+01:00","linkquality":255,"smoke":null}'
zigbee2mqtt             | 2021-12-24T18:18:15.482588348Z Zigbee2MQTT:debug 2021-12-24 19:18:15: Received Zigbee message from 'Smoke-Woonkamer', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":11776}' from endpoint 1 with groupID null
zigbee2mqtt             | 2021-12-24T18:18:15.493820609Z Zigbee2MQTT:debug 2021-12-24 19:18:15: No converter available for 'TS0601_smoke' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":11776}'
zigbee2mqtt             | 2021-12-24T18:18:15.510105098Z Zigbee2MQTT:info  2021-12-24 19:18:15: MQTT publish: topic 'zigbee2mqtt/Smoke-Woonkamer', payload '{"battery_low":null,"last_seen":"2021-12-24T19:18:15+01:00","linkquality":255,"smoke":null}'

Pressing again for second test
zigbee2mqtt             | 2021-12-24T18:23:32.603980894Z Zigbee2MQTT:debug 2021-12-24 19:23:32: Received Zigbee message from 'Smoke-Woonkamer', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":12032}' from endpoint 1 with groupID null
zigbee2mqtt             | 2021-12-24T18:23:32.619597884Z Zigbee2MQTT:debug 2021-12-24 19:23:32: No converter available for 'TS0601_smoke' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":12032}'
zigbee2mqtt             | 2021-12-24T18:23:32.637235688Z Zigbee2MQTT:info  2021-12-24 19:23:32: MQTT publish: topic 'zigbee2mqtt/Smoke-Woonkamer', payload '{"battery_low":null,"last_seen":"2021-12-24T19:23:32+01:00","linkquality":255,"smoke":null}'
zigbee2mqtt             | 2021-12-24T18:23:34.066938920Z Zigbee2MQTT:debug 2021-12-24 19:23:34: Received Zigbee message from 'Smoke-Woonkamer', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":12288}' from endpoint 1 with groupID null
zigbee2mqtt             | 2021-12-24T18:23:34.077227908Z Zigbee2MQTT:debug 2021-12-24 19:23:34: No converter available for 'TS0601_smoke' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":12288}'
zigbee2mqtt             | 2021-12-24T18:23:34.089136312Z Zigbee2MQTT:info  2021-12-24 19:23:34: MQTT publish: topic 'zigbee2mqtt/Smoke-Woonkamer', payload '{"battery_low":null,"last_seen":"2021-12-24T19:23:34+01:00","linkquality":255,"smoke":null}'

Generating real smoke alarm with piece of smoking paper (and clearing it again by waving it in fresh air):
Echte smoke alarm:
zigbee2mqtt             | 
zigbee2mqtt             | 2021-12-24T18:48:40.621187380Z Zigbee2MQTT:debug 2021-12-24 19:48:40: Received Zigbee message from 'Smoke-Woonkamer', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":12544}' from endpoint 1 with groupID null
zigbee2mqtt             | 2021-12-24T18:48:40.631655864Z Zigbee2MQTT:debug 2021-12-24 19:48:40: No converter available for 'TS0601_smoke' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":12544}'
zigbee2mqtt             | 2021-12-24T18:48:40.645097338Z Zigbee2MQTT:info  2021-12-24 19:48:40: MQTT publish: topic 'zigbee2mqtt/Smoke-Woonkamer', payload '{"battery_low":null,"last_seen":"2021-12-24T19:48:40+01:00","linkquality":255,"smoke":null}'
zigbee2mqtt             | 2021-12-24T18:48:46.141742694Z Zigbee2MQTT:debug 2021-12-24 19:48:46: Received Zigbee message from 'Smoke-Woonkamer', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":12800}' from endpoint 1 with groupID null
zigbee2mqtt             | 2021-12-24T18:48:46.149343460Z Zigbee2MQTT:debug 2021-12-24 19:48:46: No converter available for 'TS0601_smoke' with cluster 'manuSpecificTuya' and type 'commandDataReport' and data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"seq":12800}'
zigbee2mqtt             | 2021-12-24T18:48:46.158930820Z Zigbee2MQTT:info  2021-12-24 19:48:46: MQTT publish: topic 'zigbee2mqtt/Smoke-Woonkamer', payload '{"battery_low":null,"last_seen":"2021-12-24T19:48:46+01:00","linkquality":239,"smoke":null}'

By the way: according to tuya site it should also be possible to silence the siren when there is a smoke alarm. We still need a piece of the puzzle for complete definition of this device. I dont have a hub, so unfortunately I cant sniff the protocol.

Do I need to arm the alarm or something? The beeper inside is working perfectly, but I never receive something I can use in my home automation. Have paired them several times, but always same result. Am I missing something?

_Originally posted by @Roelski in https://github.com/Koenkk/zigbee2mqtt/discussions/10373_

gmbnomis, 2022-01-03:
It looks like the device reports the change (`"data":[0]` vs. `"data":[1]`) using the expected data point id 1. Strangely, it uses the `commandDataReport` command instead of `commandDataResponse` (formerly known as `commandGetData`) that was present in the logs when the device was added in the past.

Did this work for you in a previous version of zigbee2mqtt?

Roelski, 2022-01-03:
I recently started with zigbee2mqtt and zigbee devices. I just switched from master to dev, but thats about it. So I cant say if it worked with previous version, because this is my first version in which I try.

gmbnomis, 2022-01-03:
> I recently started with zigbee2mqtt and zigbee devices. I just switched from master to dev, but thats about it. So I cant say if it worked with previous version, because this is my first version in which I try.

Ok, could you test the following external converter with a current dev version of zigbee2mqtt? (see https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device for how to configure it)

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const myFz = {
    tuya_smoke: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            for (const dpValue of msg.data.dpValues) {
		const dp = dpValue.dp;
		const value = tuya.getDataValue(dpValue);
		switch (dp) {
		case tuya.dataPoints.state:
                    result.smoke = value === 0;
		    break;
		default:
                    meta.logger.warn(`zigbee-herdsman-converters:tuya_smoke: Unrecognized DP #${ dp} with data ${JSON.stringify(dpValue)}`);
		}
            }
	    return result;
        },
    },
};

const definition = {
    zigbeeModel: ['5p1vj8r'],
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_t5p1vj8r'}, {modelID: 'TS0601', manufacturerName: '_TZE200_uebojraa'}],
    model: 'TS0601_smoke',
    vendor: 'TuYa',
    description: 'Smoke sensor',
    fromZigbee: [myFz.tuya_smoke],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.smoke(), e.battery_low()],
};

module.exports = definition;
```

Roelski, 2022-01-03:
Not sure if I did everything correct.
Have to double check log if advanced js was correctly loaded.
First attempt (with freshly pulled dev docker) and pressing test button:

zigbee2mqtt | 2022-01-03T17:53:18.128189482Z Zigbee2MQTT:debug 2022-01-03 18:53:18: Received Zigbee message from 'Smoke-Woonkamer', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":17152}' from endpoint 1 with groupID null
zigbee2mqtt | 2022-01-03T17:53:18.152083127Z Zigbee2MQTT:info  2022-01-03 18:53:18: MQTT publish: topic 'zigbee2mqtt/Smoke-Woonkamer', payload '{"battery_low":null,"last_seen":"2022-01-03T18:53:18+01:00","linkquality":255,"smoke":true}'
zigbee2mqtt | 2022-01-03T17:53:19.538945686Z Zigbee2MQTT:debug 2022-01-03 18:53:19: Received Zigbee message from 'Smoke-Woonkamer', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1}],"seq":17408}' from endpoint 1 with groupID null
zigbee2mqtt | 2022-01-03T17:53:19.560047661Z Zigbee2MQTT:info  2022-01-03 18:53:19: MQTT publish: topic 'zigbee2mqtt/Smoke-Woonkamer', payload '{"battery_low":null,"last_seen":"2022-01-03T18:53:19+01:00","linkquality":255,"smoke":false}'

Do I need to delete and re-pair the smoke alarm to actually use the advanced js config?

gmbnomis, 2022-01-03:
Actually, this looks like it is doing something sensible:

zigbee2mqtt publishes an MQTT message with smoke: false

```
zigbee2mqtt             | 2022-01-03T18:14:27.679421690Z Zigbee2MQTT:info  2022-01-03 19:14:27: MQTT publish: topic 'zigbee2mqtt/0xa4c1381a9336550e', payload '{"battery_low":null,"last_seen":"2022-01-03T19:14:27+01:00","linkquality":255,"smoke":false}'
```

and then one with "smoke: true" and finally one with "smoke: false" again. These are the messages your automation should subscribe to.

The warning message indicates that the device sends another data point (number 14, enum) which is unknown yet. Looking at other Tuya devices, this could be a battery indication.

You said that a Tuya hub is on the way to you? Once you have it, you could try to find out more about DP 14 using https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html.

Roelski, 2022-01-03:
Lol, i missed the smoke part. But does this mean the definition is wrong or changed for _TZE200_uebojraa?

gmbnomis, 2022-01-03:
Yes, I will post a PR to address this. Thanks for testing!

Roelski, 2022-01-03:
> Yes, I will post a PR to address this. Thanks for testing!

No, thank YOU!! :-)

If I have the hub and have the dev account. Whats the best place to share information (and I probably need some assistance)?  This site, but discussions part?

gmbnomis, 2022-01-04:
> If I have the hub and have the dev account. Whats the best place to share information (and I probably need some assistance)? This site, but discussions part?

I am quite new to zigbee2mqtt development myself. But I suppose discussion is the right place to ask for assistance.

closed 2022-01-04 by Koenkk