Koenkk/zigbee2mqtt #10465: [Problem]: Crash when joining legrand netatmo shutter command

fariouche, 2021-12-31:
### What happened?

I got a legrand netatmo 600087 shutter command, and when I try to join it, I have z2m that always crashes at the end and the configuration. After restarting z2m, the device does not work properly even if z2m shows it as joined.
The exact error is "Error: Cluster 'genBasic' has no attribute '61440'" and full log is below.

### What did you expect to happen?

No crash, and device should work

### How to reproduce it (minimal and precise)

Get the netatmo legrand device (https://www.legrand.fr/pro/catalogue/42798-version-dooxie-with-netatmo/interrupteur-filaire-connecte-dooxie-with-netatmo-pour-volet-roulant-blanc)

Enter joining mode, press the join button on the device.

After a while, device is discovered, configured and a few seconds after, z2m crashes

### Zigbee2MQTT version

1.22.0

### Adapter firmware version

20210708

### Adapter

zzh

### Debug log

Zigbee2MQTT:info  2021-12-31 12:44:11: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2021-12-31 12:44:11: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"9ycrb-1"}'
Zigbee2MQTT:info  2021-12-31 12:44:21: Device '0x0004740000c36ec7' joined
Zigbee2MQTT:info  2021-12-31 12:44:21: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000c36ec7","ieee_address":"0x0004740000c36ec7"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-12-31 12:44:21: Starting interview of '0x0004740000c36ec7'
Zigbee2MQTT:info  2021-12-31 12:44:21: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000c36ec7","ieee_address":"0x0004740000c36ec7","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-12-31 12:44:21: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0004740000c36ec7","ieee_address":"0x0004740000c36ec7"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-12-31 12:44:27: Successfully interviewed '0x0004740000c36ec7', device has successfully been paired
Zigbee2MQTT:info  2021-12-31 12:44:27: Device '0x0004740000c36ec7' is supported, identified as: Legrand Wireless shutter switch (067646)
Zigbee2MQTT:info  2021-12-31 12:44:27: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Wireless shutter switch","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["identify","open","close","stop","moving","stopped"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"067646","options":[],"supports_ota":false,"vendor":"Legrand"},"friendly_name":"0x0004740000c36ec7","ieee_address":"0x0004740000c36ec7","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2021-12-31 12:44:27: Configuring '0x0004740000c36ec7'
Zigbee2MQTT:info  2021-12-31 12:44:27: MQTT publish: topic 'homeassistant/sensor/0x0004740000c36ec7/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0004740000c36ec7"],"manufacturer":"Legrand","model":"Wireless shutter switch (067646)","name":"0x0004740000c36ec7","sw_version":"Zigbee2MQTT 1.22.0"},"device_class":"battery","enabled_by_default":true,"name":"0x0004740000c36ec7 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x0004740000c36ec7","unique_id":"0x0004740000c36ec7_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:info  2021-12-31 12:44:27: MQTT publish: topic 'homeassistant/sensor/0x0004740000c36ec7/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0004740000c36ec7"],"manufacturer":"Legrand","model":"Wireless shutter switch (067646)","name":"0x0004740000c36ec7","sw_version":"Zigbee2MQTT 1.22.0"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"0x0004740000c36ec7 action","state_topic":"zigbee2mqtt/0x0004740000c36ec7","unique_id":"0x0004740000c36ec7_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Zigbee2MQTT:info  2021-12-31 12:44:27: MQTT publish: topic 'homeassistant/sensor/0x0004740000c36ec7/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0004740000c36ec7"],"manufacturer":"Legrand","model":"Wireless shutter switch (067646)","name":"0x0004740000c36ec7","sw_version":"Zigbee2MQTT 1.22.0"},"enabled_by_default":false,"icon":"mdi:signal","name":"0x0004740000c36ec7 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x0004740000c36ec7","unique_id":"0x0004740000c36ec7_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2021-12-31 12:44:29: Successfully configured '0x0004740000c36ec7'
Error: Cluster 'genBasic' has no attribute '61440'
    at Object.getAttribute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/zcl/utils.ts:93:19)
    at Device.onZclData (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/device.ts:215:48)
    at Controller.onZclOrRawData (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:690:20)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:135:70)
    at ZStackAdapter.emit (node:events:390:28)
    at ZStackAdapter.onZnpRecieved (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:772:30)
    at Znp.emit (node:events:390:28)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:108:18)
    at Parser.emit (node:events:390:28)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)

Koenkk, 2021-12-31:
Same as https://github.com/Koenkk/zigbee-herdsman/issues/472 , already fixed in the dev branch (and in tomorrows new release)  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2021-12-31 by Koenkk

fariouche, 2021-12-31:
That's why I don't reproduce it with the master branch but only with my production version! Ok, will wait till next release, thank you

fariouche, 2021-12-31:
I confirm, it works with dev branch and I was able to bind my shutters to it :)