Koenkk/zigbee2mqtt #10467: [New device support]: innr RS 227 T Gu10

Desaydrone, 2021-12-31:
### Link

https://www.innr.com/fr/produit/smart-spot-comfort-gu10/

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0x9c6937000070debe","nwkAddr":18471,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"RS 227 T","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,64642],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":35,"stackVersion":21,"hwVersion":2,"dateCode":"20211013","swBuildId":"2.03.00","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1640958901757,"useImplicitCheckin":true}

### Comments

Hello,

I try to activate debug mode and link the converter. But nothing appear in the log. So i'm stuck at point 2, don't know if it works well :(

Sorry for not really capable to give you more informations :(

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['inrr'],
    model: 'RS 227 T',
    vendor: 'innr',
    description: 'GU10 spot 420 lm, dimmable, white spectrum',
    // Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    // - extend.light_onoff_brightness
    // - extend.light_onoff_brightness_colortemp
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    extend: extend.light_onoff_brightness_colortemp(),
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint,
            ['genBasic', 'genGroups', 'genScenes', 'genOnOff', 'genLevelCtrl']);
        for (const ep of [3, 4, 5, 6, 7, 8]) {
            const endpoint = device.getEndpoint(ep);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        }
    },
};

module.exports = definition;
```

### Supported color modes

brightness, colortemp

### Color temperature range

{"colorTempPhysicalMin":200,"colorTempPhysicalMax":454}

Koenkk, 2021-12-31:
Added, device will be supported out-of-the-box in the tomorrows new release.

closed 2021-12-31 by Koenkk

Desaydrone, 2021-12-31:
Hoo thanks a lot this awesome !