Koenkk/zigbee2mqtt #10469: [Problem]:

kender65, 2021-12-31:
### What happened?

[error] [zigbee-shepherd:zig-a-zig-ah!] Failed to setup reporting for 0x842e14fffe3b8881 Silvercrest LED-Bulb - ConfigureReporting 0x842e14fffe3b8881/1 lightingColorCtrl

### What did you expect to happen?

setup reporting without errors

### How to reproduce it (minimal and precise)

pair device and set up configuration and reporting

### Zigbee2MQTT version

none

### Adapter firmware version

2.7.1 20210708

### Adapter

Electrolama zig-a-zig-ah!

### Debug log

27 Dec 20:46:58 - [error] [zigbee-shepherd:zig-a-zig-ah!] Failed to setup reporting for 0x842e14fffe3b8881 Silvercrest LED-Bulb - ConfigureReporting 0x842e14fffe3b8881/1 lightingColorCtrl([{"attribute":"colorTemperature","minimumReportInterval":3,"maximumReportInterval":3600,"reportableChange":1},{"attribute":"currentX","minimumReportInterval":3,"maximumReportInterval":3600,"reportableChange":1},{"attribute":"currentY","minimumReportInterval":3,"maximumReportInterval":3600,"reportableChange":1},{"attribute":"currentSaturation","minimumReportInterval":3,"maximumReportInterval":3600,"reportableChange":1},{"attribute":"enhancedCurrentHue","minimumReportInterval":3,"maximumReportInterval":3600,"reportableChange":1},{"attribute":"colorMode","minimumReportInterval":3,"maximumReportInterval":3600,"reportableChange":1}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')

kender65, 2021-12-31:
I'm currently using @hobbyquaker's node-red-contrib-zigbee (based on zigbee2mqtt) instead of native zigbee2mqtt.

It looks like @sjorge's is correct with his assumption  (https://github.com/Koenkk/zigbee2mqtt/issues/6921#issuecomment-812932627) about all Lidl (Livarno) bulbs provide the same behavior is coming true.
I was able to reproduce it with "non RGB devices" Livarno Lux E27 bulb CCT (https://www.zigbee2mqtt.io/devices/HG06492C.html) and Livarno Lux GU10 spot CCT (https://www.zigbee2mqtt.io/devices/HG06492A.html).
Reading attribute "enhancedCurrentHue" leads to the above mentioned error for both devices.
I checked lidl.js for devices HG06492A, HG06492B and HG06492C for meta attribute "enhancedHue: false" but it is missing.

Since I'm not familiar how to conduct a PR to fix it I'd like to ask @sjorge or @koenkk to do it.
Thx for your support and the great work with zigbee2mqtt. I think I'll switch to the native version very soon.

Koenkk, 2021-12-31:
The reporting feature has been deprecated, I recommend to migrate away from it, see second point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.17.1 (disabling it will solve this error)

closed 2021-12-31 by Koenkk

kender65, 2022-04-22:
Hello

As I mentioned in my thread I'm currently running node-red-contrib-zigbee.
For a newbie in node-red, zigbee and even linux it was the easiest way to
get a network running.

It seems node-red-contrib-zigbee, at least right now, is not longer
maintained. Therefore I think about switching to your native zigbee2mqtt.

Prior to the start of this 'migration' I would like to get some additional
info:
Would you recommend to upgrade to the latest firmware version for my zigbee
stick (zzh!)?
Can I preserve the panId and the network key of the current configuration?
Do I have to re-pair the devices? Just 15, but re-pairing would take some
time.

Thanks in advance for your support.

Kind regards
Gerald