Koenkk/zigbee2mqtt #10471: [Problem]: RTCGQ13LM does not send "occupancy false"

serje73, 2021-12-31:
### What happened?

Due need of support of QBKG39LM I cnange the firmware of my cc2652p E72 to 20211114
And from that firmware my RTCGQ13LM stopped sent "occupancy false" event. It freezes at "occupancy true" state.
I see the event of "occupancy true". But no "occupancy false" after the "occupancy_timeout" seconds.

The same thing was with my cc2531 and firmware 20211115.

### What did you expect to happen?

I expect the occupancy_timeout will force the sensor to send "occupancy false" event.

### How to reproduce it (minimal and precise)

Install that firmware and re-pair the sensor

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211114

### Adapter

cc2652p E72

### Debug log

Received Zigbee message from '0x54ef4410002dc00f', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0

Koenkk, 2022-01-01:
- First make sure to use the latest firmware from https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin
- After that make sure to repair the RTCGQ13LM (since it needs to detect the change in the coordinator firmware)

serje73, 2022-01-01:
Did it!
Upgraded z2m and the firmware on the stick.
Now:
Zigbee2MQTT version    1.22.2 commit: unknown
Coordinator type    zStack3x0
Coordinator revision    20211217
Coordinator IEEE Address    0x00124b00257922e1
Frontend version    0.6.67

and the sensor state is:
{
    "battery": 100,
    "linkquality": 160,
    "motion_sensitivity": "high",
    "occupancy": true,
    "occupancy_timeout": 60,
    "state": "OFF",
    "temperature": 25,
    "voltage": 3216
}
And occupancy timeout does not push "occupancy false" event.

Koenkk, 2022-01-01:
@z-x-f could you maybe help with this?

Koenkk, 2022-01-02:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/3600

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-02 by Koenkk