Koenkk/zigbee2mqtt #10481: [Problem]: Xiaomi MFKZQ01LM no battery

kwetnico, 2022-01-01:
### What happened?

Hello, I have a problem with the battery information feedback on fronted with my xiaomi cube.
The battery information goes up correctly except on the main page
[image]

[image]
.

Thanks

### What did you expect to happen?

The battery information goes back correctly like other devices

### How to reproduce it (minimal and precise)

.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

 0x26680700

### Adapter

ConBee2/RaspBee2

### Debug log

No debug log

kwetnico, 2022-01-08:
No idea ? thanks

Koenkk, 2022-01-08:
Can you upgrade to 1.22.2 and provide the herdsman debug logging when force removing the device and pairing it again?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

kwetnico, 2022-01-08:
> Can you upgrade to 1.22.2 and provide the herdsman debug logging when force removing the device and pairing it again?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

I am already in 1.22.2

closed 2022-02-17 by github-actions[bot]