Koenkk/zigbee2mqtt #10482: [Problem]: Node version for Z2M on PI4 ?

kwetnico, 2022-01-01:
### What happened?

Hello, I am using a PI4 with Debian.
i have the latest version of Z2M: 1.22.2 but when i installed i got messages for outdated versions of node and npm.

Which version should I install with Z2M? and what are the commands to update under Debian? 
Thank you

Error : 
[image]

NPM et NODE Version : 
[image]

### What did you expect to happen?

Update version.

### How to reproduce it (minimal and precise)

.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

 0x26680700

### Adapter

ConBee2/RaspBee2

### Debug log

.

lux73, 2022-01-01:
https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing - there are all Information you need.

btw. i'm running z2m under Debian 11 Buster - no problems so far

```
[21:05:57] root@apu:~# node -v
v14.18.1
[21:06:00] root@apu:~# npm -v
8.1.4
[21:06:05] root@apu:~#
```

kwetnico, 2022-01-01:
> https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing - there are all Information you need.
> 
> btw. i'm running z2m under Debian 11 Buster - no problems so far
> 
> ```
> [21:05:57] root@apu:~# node -v
> v14.18.1
> [21:06:00] root@apu:~# npm -v
> 8.1.4
> [21:06:05] root@apu:~#
> ```

kwetnico, 2022-01-01:
> > https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing - there are all Information you need.
> > btw. i'm running z2m under Debian 11 Buster - no problems so far
> > ```
> > [21:05:57] root@apu:~# node -v
> > v14.18.1
> > [21:06:00] root@apu:~# npm -v
> > 8.1.4
> > [21:06:05] root@apu:~#
> > ```

Ok but my z2m has been running for over a year but how can I manually update node and npm?

lux73, 2022-01-01:
https://docs.npmjs.com/cli/v7/commands/npm-update/

`npm update -g` should do

node should be updated via apt update manager - package called nodejs

i could not furnish a guarantee that the cmd's above work without Problems on your System - use it carefully please & **do a backup before proceeding!**

kwetnico, 2022-01-01:
> https://docs.npmjs.com/cli/v7/commands/npm-update/
>
> `npm update -g` should do
>
> node should be updated via apt update manager - package called nodejs
>
> i could not furnish a guarantee that the cmd's [...]

After npm update-g :
[image]

kwetnico, 2022-01-01:
I manually install nodejs 14.xx and npm
[image]

Koenkk, 2022-01-02:
The warnings from your first post can be ignored, recommended version can be found here: https://www.zigbee2mqtt.io/guide/installation/01_linux.html#determine-location-of-the-adapter-and-checking-user-permissions 

> node --version  # Should output v10.X, v12.X, v14.X, v15.X or V16.X

closed 2022-01-02 by Koenkk

kwetnico, 2022-01-02:
> The warnings from your first post can be ignored, recommended version can be found here: https://www.zigbee2mqtt.io/guide/installation/01_linux.html#determine-location-of-the-adapter-and-checking-user-permissions
> 
> > node --version  # Should output v10.X, v12.X, v14.X, v15.X or V16.X

Thanks but I can't find which version you recommend? V12? V14 or V16 (even if z2M works with all)

Koenkk, 2022-01-02:
16 then, since it is the latest lts

kwetnico, 2022-01-02:
> 16 then, since it is the latest lts

I updated to the latest V16.13.1 but apparently you compiled with an older version, Z2M refused to start.
I reinstall node V14 and it is very good for the moment.