Koenkk/zigbee2mqtt #10487: [TS011F_plug_3 _TZ3000_gjnozsaz] Power outage memory not working

majkers, 2022-01-01:
### What happened?

I am using Home Assistant plugin version 1.22.2-1. In this version socket TS011F_plug_3 _TZ3000_gjnozsaz is supported but when I set power outage memory to restore or on it is not working. Every time after power outage, socket sets its state to off.

### What did you expect to happen?

I excpected socket to set its state as in power_outage_memory setting

### How to reproduce it (minimal and precise)

1. Set Power_outage_memory setting to restore.
2. Take socket off from wall socket 
3. Put it back

### Zigbee2MQTT version

1.22.2-1

### Adapter firmware version

0x26660700

### Adapter

ConBee2/RaspBee2

### Debug log

n/a

majkers, 2022-01-01:
Updated conbee 2 to 0x26720700 and still no change

frederikbove, 2022-01-02:
Ik have the same problem with the _TZ3000_v1pdxuqq, switching works but no power measurements

[image]

flatsiedatsie, 2022-01-03:
Ah, so that's it. I have seen this too. Suddenly had a defrosted freezer :-)

- Also conbee2 (with most recent firmware). 
- No Home Assistant though (using Webthings Gateway)

I bought four of these devices, and two of them cause a power outage whenever I plug them in.. They're not great in my opinion. Maybe the device database could also have a rating or opinion system, or just some warnings from the community.

sergey-1976, 2022-01-18:
> ### What happened?
> I am using Home Assistant plugin version 1.22.2-1. In this version socket TS011F_plug_3 _TZ3000_gjnozsaz is supported but when I set power outage memory to restore or on it is not working. Every time after power outage, socket sets its state to off.

I have the same problem with these sockets.

paszczaq, 2022-01-24:
Same problem with deconz and zha. Seems that there is firmware update that fixes problem with setting after power loss and also meter report instead of polling https://forum.phoscon.de/t/new-tuya-smart-plug-nas-wr01b-or-tz3000-w0qqde0g/819

closed 2022-03-03 by github-actions[bot]

bogorad, 2022-05-21:
Same problem here, with [this socket](https://www.aliexpress.com/item/1005003611211316.html). 

**FIXED it** by pairing it with a TUYA-compatible hub and updating the firmware to the current one, then re-pairing back to z2m.

BombusAlpinus, 2023-12-23:
hi, i know this thread is old but i still have the problem that the power outage memory is not working. also had an defrosted feezer one day. can anyone confirm that the fix with an tuya compatible hub for the nous a1z works? where do i get the current firmware informations out of the plug? thanks in advance!