Koenkk/zigbee2mqtt #10489: Wrong time synchronization  and no periodical time synchronization on thermostats MOES TS0601, AVATTO ME81H, HKGK  BAC-002-ALZB

shymchenko, 2022-01-01:
### What happened?

After update to version 1.22.2  synchronization of times  occurs only after restart zibgee2mqtt but with the wrong random time!
On the version 1.22.1 time synchronization worked correct. 
If restore backup with previous Zigbee2Mqtt, time synchronisation work is good.
P.S.
Same problem was on my test HA server with zigbee2mqtt EDGE v1.22.0-3

### What did you expect to happen?

Correct periodic time synchronization

### How to reproduce it (minimal and precise)

Update zigbee2mqtt version 1.22.1 to 1.22.2. Restart zigbee2mqtt or switch off/on the thermostats.

### Zigbee2MQTT version

1.22.2-1

### Adapter firmware version

20201010

### Adapter

zStack30x

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7798077/log.txt)

Ivano62, 2022-01-02:
I've the same problem on BRT-100-TRV thermostat (TS0601 - _TZE200_b6wax7g0) with zigbee2mqtt v1.22.2.
Thanks

tauka84, 2022-01-02:
Same problem with moes BHT-002-GCLZB

aegjoyce, 2022-01-02:
> Same problem with moes BHT-002-GCLZB

Same model, same here. Going to try reverting to 1.22.1

Koenkk, 2022-01-02:
@hacker-cb could you check if this has something to do with https://github.com/Koenkk/zigbee-herdsman-converters/commit/27b4ef37377a1f3abd4071a882c47400810f0d8f ?

Koenkk, 2022-01-02:
Did an attempt to fix this, could you update to the latest dev branch and wait 24 hours to give the time to sync.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

aegjoyce, 2022-01-02:
I’m away for a week but if nobody has confirmed fixes then I will test when I get home. Thanks @Koenkk 👍🏻

m1nl, 2022-01-03:
Bumped zigbee-herdsman to 0.13.190 and zigbee-herdsman-converters to 14.0.369 and I confirm that MOES BRT-100-TRV now synchronizes time correctly.

WildeRNS, 2022-01-03:
Can confirm: on 1.22.2-dev commit: 23bdc18, time synchronization working fine (MOES BHT-002-GCLZB).

groenmarsmannetje, 2022-01-03:
Also with Moes HY368 TRV it seems so far that time synchronisation is working fine again so far after installing 1.22.2-dev commit: 23bdc18.

Koenkk, 2022-01-03:
thanks for confirming.

closed 2022-01-03 by Koenkk

shymchenko, 2022-01-03:
After installing 1.22.2-dev the problem is fixed! Thank you!

bambo86, 2022-01-06:
Same problem with TS0601 (Moes BHT-002-GCLZB).
Whet it will be fixed in "normal" version of zigbee2mqtt ?

pawol, 2022-01-06:
It seems to me that time on z2b 1.22.1 is synchronized properly but now on z2b 1.22.2 I have to set the time manually on my TS0601. Unfortunatelly I have updated my cc2652p in the same time. After downgrade of cc2652p problem still persists, so it looks I have confirmed bug in the z2b 1.22.2. When 1.22.3 will be released?

bdariusb, 2022-01-11:
huh, after updating to dev branch, the time synchronizes on all devices, but it is wrong time 🤷

groenmarsmannetje, 2022-01-11:
Time can be a complex concept. Please explain in detail what is wrong. And how did you verify if the time was correct or wrong. Did you check the display on the device itself?

bdariusb, 2022-01-11:
it's a HY08WE thermostat, has date and time display. Time is shifted forward by ~1.5 days as opposed to local machine time. Running `koenkk/zigbee2mqtt:latest-dev` docker image

pawol, 2022-01-11:
I use bare metal installation and the time is ok

tauka84, 2022-01-11:
In moes BHT-002-GCLZB time is synchronized correctly, but the day of the week is wrong. For example today (tuesday) it should show 2 but it display 1.
Even if i change manualy it back to wrong day after synchro

shymchenko, 2022-01-12:
> In moes BHT-002-GCLZB time is synchronized correctly, but the day of the week is wrong. For example today (tuesday) it should show 2 but it display 1. Even if i change manualy it back to wrong day after synchro.

I have a custom integration for  BHT-002-GCLZB with full range of parameters.

https://drive.google.com/file/d/1RSX8ImbfL7O_R95HbmRbxO4U-qDZtmUR/view?usp=sharing

I'm still working on it, but the time synchronization in it is normal

luk984, 2022-01-12:
When can we expect the update of the regular zigbee2mqtt vesrion with the solution from the dev? I would like to stay with the stable version and have the issue fixed. Is there any planned date for the update?

groenmarsmannetje, 2022-01-12:
A new version of zigbee2mqtt is released every month around the first day of the month. So next one is on February 1st I think.

KameDomotics, 2022-01-13:
I have a BHT-002, coordinator CC2652 (fw Zstack 20211217) and Zigbee2Mqtt v1.22.2.

Same problem.

I have to manually change the time and day of the week several times in a day.
Example: I manually set the time 19:45 and the day of the week number 4, after not even one hour I find the time 6:56 and day 7.
It changes automatically without a sense...

pakagi, 2022-01-14:
Hello, is this really fixed ?

I switched from stable to dev - says 1.22.2-dev (commit #57852fcd
), but after starting zigbee2mqtt I had to re-pair my thermostat valve (TS0601), and at that moment it screwed up the date and time on it again (set some random one).

bdariusb, 2022-01-14:
> Hello, is this really fixed ?
> 
> I switched from stable to dev - says 1.22.2-dev (commit #57852fcd ), but after starting zigbee2mqtt I had to re-pair my thermostat valve (TS0601), and at that moment it screwed up the date and time on it again (set some random one).

definitely not fixed. At least for some devices

groenmarsmannetje, 2022-01-15:
If it is not fixed for you please provide the manufacturer id that is shown in zigbee2mqtt front-end. Because there are many different versions that all use TS0601 as id. This might  help the developers to pinpoint the devices that are not fixed.

bdariusb, 2022-01-15:
My device id is _TZE200_znzs7yaw. As I mentioned time is shifted by ~1.5 days forward and it is the same on all devices

pakagi, 2022-01-15:
In my case, it still behaves wrong way on device "_TZE200_husqqvux".

groenmarsmannetje, 2022-01-16:
In my case the fix works with _TZE200_ckud7u2l and _TZE200_ywdxldoj both branded as Moes.

elBoz73, 2022-01-16:
Hi all, 
Same issue for me for my moes BHT-002-GCLZB.
Constructeur Zigbee _TZE200_aoclfnxz
Modèle Zigbee TS0601.

Is there a workaround to wait until the next release?
Thanks

bambo86, 2022-01-16:
> Hi all, Same issue for me for my moes BHT-002-GCLZB. Constructeur Zigbee _TZE200_aoclfnxz Modèle Zigbee TS0601.
> 
> Is there a workaround to wait until the next release? Thanks

bambo86, 2022-01-16:
> > Hi all, Same issue for me for my moes BHT-002-GCLZB. Constructeur Zigbee _TZE200_aoclfnxz Modèle Zigbee TS0601.
> > Is there a workaround to wait until the next release? Thanks

I came back to the previous version of zigbee2mqtt
Time and date are good
I'm waiting to the next release with update

elBoz73, 2022-01-16:
Hi Bambo86 ,
thanks for the workaround. 
I do not find how to donwgrade, is there a procedure or do I have to uninstall and reinstall the previsou version?
One question. If I have to uninstall the addon do I loose all my config ?

Thanks

bambo86, 2022-01-16:
I do a backup before each update.
If you don't, you may have a problem...

iksrd, 2022-01-17:
After updating the firmware, it does not work. Thermal head: _TZE200_cwnjrr7
[image]
2

elBoz73, 2022-01-17:
bambo86, 
I installed HA a month ago and switch to zigbee2mqtt only two weeks ago... So no back up as it is my first version.
But, I take the advise for all my futur update..... thanks for that.

groenmarsmannetje, 2022-01-17:
> bambo86,
> I installed HA a month ago and switch to zigbee2mqtt only two weeks ago... So no back up as it is my first version.
> But, I take the advise for all my futur update..... thanks for that.

Home Assistant OS offers daily backups.

groenmarsmannetje, 2022-01-17:
@Koenkk it seems that the fix doesn't work for all devices. See reports about it above. Should this report be reopened or a new one be created?

Koenkk, 2022-01-17:
@groenmarsmannetje are you sure you are on the latest dev? Could you provide me a screenshot of settings -> about like this:

[image]

groenmarsmannetje, 2022-01-17:
> @groenmarsmannetje are you sure you are on the latest dev? Could you provide me a screenshot of settings -> about like this:
> 
> [image]

@bdariusb,  @pakagi  and  @elBoz73 as you three reported that there are still issues. Please see request from Koen above.

tauka84, 2022-01-18:
> > @groenmarsmannetje are you sure you are on the latest dev? Could you provide me a screenshot of settings -> about like this:
> 
> @bdariusb, @pakagi, @tauka84 and @elBoz73 as you four reported that there are still issues. Please see request from Koen above.

@Koenkk i use this version
[image]

and i have this issue still 
> In moes BHT-002-GCLZB time is synchronized correctly, but the day of the week is wrong. For example today (tuesday) it should show 2 but it display 1. Even if i change manualy it back to wrong day after synchro.

The day of the week on moes is always one day back. Time is ok.

Koenkk, 2022-01-18:
@tauka84 did you check that the issue is fixed when downgrading to 1.22.1?

shymchenko, 2022-01-18:
Zigbee2MQTT version 1.22.2-dev commit: e6b0414
Coordinator type zStack30x
Coordinator revision 20201010
Frontend version 0.6.68

- MOES BHT-002-GCLZB
- AVATTO ME81H
- HKGK BAC-002-ALZB

Date, Time and day of week is ok!

tauka84, 2022-01-18:
My both MOES BHT-002-GCLZB set manual corectly, change one day back after synchro.
[image]
[image]

shymchenko, 2022-01-20:
Just updated Zigbee2MQTT
version 1.22.2-dev commit: d873479

MOES BHT-002-GCLZB
AVATTO ME81H
HKGK BAC-002-ALZB
Date, Time and day of week is ok!

tauka84, 2022-01-20:
I connect the MOES BHT-002-GCLZB with moes zb gate (tuya).
Its the same problem with one day back.
All the date on router are correct.
Question what happend with my moes 😲

shymchenko, 2022-01-20:
I don`t use a Tuya zIgbee gate, I connect thermostates to usb zigbee stick.

tauka84, 2022-01-20:
> I don`t use a Tuya zIgbee gate, I connect thermostates to usb zigbee stick.

I normally use usb zigbee stick too.
Connected tuya gate to check if it synchronizes the day correctly. Not.
That means it's not zigbee2mqtt fault. 😉

kremik, 2022-02-01:
This is not closed. I used the dev branch and time sync was working OK. After I saw there is new 1.22.2 master version I switched to the master version and the issue is back - time is not synced in 1.22.2 :-((((((

groenmarsmannetje, 2022-02-02:
> This is not closed. I used the dev branch and time sync was working OK. After I saw there is new 1.22.2 master version I switched to the master version and the issue is back - time is not synced in 1.22.2 :-((((((

That doesn’t surprise me, because it is mentioned in the release notes of release 1.23.0 which was releases yesterday.

kremik, 2022-02-02:
> > This is not closed. I used the dev branch and time sync was working OK. After I saw there is new 1.22.2 master version I switched to the master version and the issue is back - time is not synced in 1.22.2 :-((((((
> 
> That doesn’t surprise me, because it is mentioned in the release notes of release 1.23.0 which was released yesterday. Maybe install that version instead 😉

Yes, I confirm that it was fixed in 1.23.0. I didn't know that 1.22.2 and 1.23 were released so fast after each other.

elBoz73, 2022-02-04:
I updated mine, and I also confirm that the  1.23.0 fixed my issue so far.

Thanks a lot .

bdariusb, 2022-09-25:
hello. So I only recently updated to latest zigbee2mqtt and here I go again with un-synchronized time on the thermostat (_TZE200_znzs7yaw) Last working version is 1.21.1. If I downgrade from latest it instantly synchronizes time in all devices. What can I do to debug this?

Alex2030S, 2022-10-16:
Hi. For my device, the problem has not been solved on any versions. Constant time out of sync.
"manufacturerID":4098,"manufacturerName":"_TZE200_znzs7yaw","model":"HY08WE"
z2m 1.28.0
zStack3x0 20220928

sheaur, 2022-10-29:
hello, my device, the problem has not been solved: _TZE200_2ekuz3dz, z2m 1.28.0 zStack3x0 20220219

bdariusb, 2022-11-13:
I played around a bit and found out that with this external converter the time gets to synchronize on my devices after some time:
```
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const hy_set_time_request = {
    cluster: 'manuSpecificTuya',
    type: ['commandMcuSyncTime'],
    convert: async (model, msg, publish, options, meta) => {
        const OneJanuary2000 = new Date('January 01, 2000 00:00:00 UTC+00:00').getTime();
        const currentTime = new Date().getTime();
        const utcTime = Math.round((currentTime - OneJanuary2000) / 1000);
        const localTime = Math.round(currentTime / 1000) - (new Date()).getTimezoneOffset() * 60;
        const endpoint = msg.endpoint;
        const payload = {
            payloadSize: 8,
            payload: [
                ...tuya.convertDecimalValueTo4ByteHexArray(utcTime),
                ...tuya.convertDecimalValueTo4ByteHexArray(localTime),
            ],
        };
        await endpoint.command('manuSpecificTuya', 'mcuSyncTime', payload, {});
    },
}

const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ea = exposes.access;
const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_znzs7yaw'}],
    model: 'HY08WE',
    vendor: 'TuYa',
    description: 'Wall-mount thermostat (EXTERNAL)',
    fromZigbee: [fz.hy_thermostat, fz.ignore_basic_report, hy_set_time_request],
    toZigbee: [tz.hy_thermostat],
    exposes: [exposes.climate().withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET)
        .withLocalTemperature(ea.STATE)
        .withSystemMode(['off', 'auto', 'heat'], ea.STATE_SET).withRunningState(['idle', 'heat'], ea.STATE)],
};

module.exports = definition;

```
Now what exactly was wrong I am not sure, maybe it's the payload format but at least the thermostats are usable now

pabloruizsoria, 2022-12-11:
I have the same problem with my Moes BHT-002-GCLZB.
I'm using a raspberry with a zigbee usb stick where I have installed (not dockerized) hassio OS.
I think I'm up to date:

```
Home Assistant 2022.12.1
Supervisor 2022.11.2
Operating System 9.3
Frontend 20221208.0 - latest
```

and it says zigbee (settings > about):

```
Zigbee2MQTT version
[1.28.4](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.4) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coordinator type
zStack12
Coordinator revision
20180507
Coordinator IEEE Address
0x00124b0014d9b237
```

Thanks in advance

AoSpades, 2023-11-06:
Daylight saving time ended a week ago, and now I have encountered the same issue with a bunch of VALE SMART THERMOSTAT TV05-ZG that report as TS0601 _TZE200_mudxchsu