Koenkk/zigbee2mqtt #10491: [Problem]: IKEA E1744 with legacy = false don't report correctly

roumano, 2022-01-01:
### What happened?

I use this device : IKEA E1744
After re-reading the documentation ( https://www.zigbee2mqtt.io/devices/E1744.html ), i have disable the legacy support

But with `legacy: false`, it's not reporting correctly the change

If i continuously turn the volume of the device relativity quickly it's only send one packet `brightness_move_up`
but i see lot of this line (exact same line) :  

```
info MQTT publish: topic 'zigbee2mqtt/Volume', payload '{"action":null,"battery":34,"brightness":255,"linkquality":255,"rate":195,"update":{"state":"idle"},"update_available":false}'
```

My full configuration of the device is now : 
```
  '0xec1bbdfffe...':
    friendly_name: Volume
    legacy: false
    simulated_brightness:
      delta: 20
      interval: 200
```

If i turn slow the device (like 5 secondes for 1 turn, it's working as execpted as it's send these packet :  
```
brightness_move_up
brightness_stop
brightness_move_up
brightness_stop
brightness_move_up
brightness_stop
...
```

### What did you expect to happen?

Should send several packet brightness_move_up

### How to reproduce it (minimal and precise)

* Pair the device IKEA E1744
* Disable legacy support
* Turn quickly the volume of the device 

### Zigbee2MQTT version

1.22.2 commit: 414c51f5

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Debug log

[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/7798239/debug.txt)

roumano, 2022-02-01:
Issue still exist

YodaDaCoda, 2022-02-11:
I'm using zigbee2mqtt v1.23.0 & a CC2652R  (zig-a-zig-ah! / zzh) USB stick with firmware 20211217.

I'm experiencing what I think is the same issue. To clarify, when I rotate the device slowly, I get a `brightness_move_(up|down)` followed immediately by a `brightness_stop` every ~100ms (I'm estimating the frequency based on what I can see coming from the zigbee2mqtt logs). If I enable legacy mode for the device, I get the same behaviour but with `rotate_(left|right)`/`rotate_stop` events.

If I rotate the device a little more quickly, I get a single `brightness_move_(up|down)`->`brightness_stop` after I stop rotating. Doesn't matter if I rotate for 1s or 1m, I only get the events after I stop rotation.

In what is likely a separate issue, if I rotate for long enough (longer than about 5s), I actually get `brightness_move_up`/`brightness_move_down`/`brightness_move_up`/`brightness_stop` (when I rotate clockwise; swap the up/down events for rotating anticlockwise). In this case, I think the down event and the second up event are both spurious.

I tried the simulated brightness stuff, but since the events only occur after rotation stops, it didn't achieve anything useful. (Also, the `brightness` attribute didn't change with the `brightness_move` events, surely it should?)

The interesting part, is the events emitted by z2m appear to match the zigbee messages received from the device, leading me to think this is perhaps an issue with the device, not with z2m - assuming the `Received Zigbee message` lines in the log accurately reflect a zigbee message being received, and aren't an abstraction. I did update the device to the latest firmware version, `2.3.075`, so maybe something has changed in its behaviour between versions. Happy to test different firmware versions if I can figure out how to downgrade (though OTA takes a couple hours...)

closed 2022-03-25 by github-actions[bot]