Koenkk/zigbee2mqtt #10492: [Problem]: Ikea  LED2005R5 Group Transition not working when using Home Assistant Scenes.

tungmeister, 2022-01-01:
### What happened?

I've got 14 of these spotlights in a group for my dining room lighting. I've set the transition to 1 at each light and on the group however when changing scenes the transition isn't respected other than when turning off the lights.
I've also tried defining a transition when selecting the scenes but that also has no impact. 

### What did you expect to happen?

the group should use the transition period. 

### How to reproduce it (minimal and precise)

define a group with the lights in it, set a transition on the group,  then set the brightness/on state/colour temp from a scene in Hass. No transition will be observed

### Zigbee2MQTT version

 1.22.2-dev commit: 15f1df8

### Adapter firmware version

20210708

### Adapter

Slaeshs

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7798244/log.txt)

tungmeister, 2022-01-01:
Just looked into this further and I've seen that this will be due to my scenes providing brightness and color_temp. I'll have to split this out to get the transitions working.

closed 2022-01-01 by tungmeister

Shivy011, 2023-01-06:
Also got this issue.