Koenkk/zigbee2mqtt #10493: No data in Domoticz on new installed BlitzWolf SHP-15

WBoerhoop, 2022-01-01:
### What happened?

After installing 1.22.2, pairing was not a problem anymore with blitzWolf SHP-15 and Zigbee2mqtt on Domoticz.
After installation I discovered that no data was imported in Domoticz.

i connected for a test a powerbank with build in light with light on to test on BWPL03 between 22:45 and 23:15 ( see log)

[image]

### What did you expect to happen?

After connecting a device that need power to the plug, that i can read a power consumption inside Domoticz to perform related actions.

### How to reproduce it (minimal and precise)

connect for test powerbank

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20

### Adapter

Slaesh's CC2652RB

### Debug log

 info  2022-01-01 22:31:29: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:31:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:31:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":1.07,"energy":78.78,"linkquality":45,"power":208,"state":"ON","voltage":233}'
info  2022-01-01 22:31:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":69,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:31:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":1.07,"energy":78.78,"linkquality":42,"power":208,"state":"ON","voltage":233}'
info  2022-01-01 22:32:54: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":1.07,"energy":78.79,"linkquality":72,"power":208,"state":"ON","voltage":233}'
info  2022-01-01 22:32:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":1.07,"energy":78.79,"linkquality":72,"power":208,"state":"ON","voltage":233}'
info  2022-01-01 22:32:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":75,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:32:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:32:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":1.07,"energy":78.79,"linkquality":72,"power":208,"state":"ON","voltage":233}'
info  2022-01-01 22:33:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":75,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:33:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.89,"energy":78.79,"linkquality":39,"power":150,"state":"ON","voltage":234}'
info  2022-01-01 22:33:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:33:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.89,"energy":78.79,"linkquality":42,"power":150,"state":"ON","voltage":234}'
info  2022-01-01 22:34:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:34:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.88,"energy":78.79,"linkquality":42,"power":148,"state":"ON","voltage":233}'
info  2022-01-01 22:34:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:34:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.88,"energy":78.79,"linkquality":39,"power":148,"state":"ON","voltage":233}'
info  2022-01-01 22:35:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.9,"energy":78.79,"linkquality":42,"power":150,"state":"ON","voltage":233}'
info  2022-01-01 22:35:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.9,"energy":78.79,"linkquality":42,"power":150,"state":"ON","voltage":233}'
info  2022-01-01 22:35:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:35:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:36:27: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:36:47: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.9,"energy":78.8,"linkquality":69,"power":150,"state":"ON","voltage":233}'
info  2022-01-01 22:36:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":233}'
info  2022-01-01 22:36:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.9,"energy":78.8,"linkquality":72,"power":150,"state":"ON","voltage":234}'
info  2022-01-01 22:36:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":233}'
info  2022-01-01 22:36:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.9,"energy":78.8,"linkquality":72,"power":150,"state":"ON","voltage":234}'
info  2022-01-01 22:37:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":69,"power":0,"state":"ON","voltage":233}'
info  2022-01-01 22:37:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.9,"energy":78.8,"linkquality":45,"power":150,"state":"ON","voltage":234}'
info  2022-01-01 22:37:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":233}'
info  2022-01-01 22:37:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.9,"energy":78.8,"linkquality":48,"power":150,"state":"ON","voltage":234}'
info  2022-01-01 22:38:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":233}'
info  2022-01-01 22:38:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.9,"energy":78.8,"linkquality":39,"power":152,"state":"ON","voltage":234}'
info  2022-01-01 22:38:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":233}'
info  2022-01-01 22:38:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.9,"energy":78.8,"linkquality":39,"power":152,"state":"ON","voltage":234}'
info  2022-01-01 22:39:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":232}'
info  2022-01-01 22:39:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.88,"energy":78.8,"linkquality":39,"power":150,"state":"ON","voltage":233}'
info  2022-01-01 22:39:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":232}'
[... 13 lines cut ...]
info  2022-01-01 22:42:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":75,"power":0,"state":"ON","voltage":231}'
info  2022-01-01 22:42:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.9,"energy":78.81,"linkquality":72,"power":148,"state":"ON","voltage":232}'
info  2022-01-01 22:43:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":72,"power":0,"state":"ON","voltage":233}'
info  2022-01-01 22:43:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.9,"energy":78.81,"linkquality":72,"power":150,"state":"ON","voltage":234}'
info  2022-01-01 22:43:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":75,"power":0,"state":"ON","voltage":233}'
info  2022-01-01 22:43:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.9,"energy":78.81,"linkquality":72,"power":150,"state":"ON","voltage":234}'
[... cut ...]
info  2022-01-01 22:47:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.86,"energy":78.82,"linkquality":39,"power":140,"state":"ON","voltage":234}'
info  2022-01-01 22:47:57: MQTT publish: topic 'zigbee2mqtt/BWPL01-Slaapkamer', payload '{"current":0,"energy":3.82,"linkquality":75,"power":0,"state":"ON","voltage":233}'
info  2022-01-01 22:47:57: MQTT publish: topic 'zigbee2mqtt/BWPL02-Zolder', payload '{"current":0.86,"energy":78.82,"linkquality":39,"power":140,"state":"ON","voltage":234}'

WBoerhoop, 2022-01-02:
on https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html  
In the options I read that the plug must be actively approached to read the electrical data

Options
How to use device type specific configuration

measurement_poll_interval: This device does not support reporting electric measurements so it is polled instead. The default poll interval is 60 seconds, set to -1 to disable. The value must be a number with a minimum value of -1

WBoerhoop, 2022-01-02:
The function of the switch (on/off) is the only action that works.

WBoerhoop, 2022-01-02:
Part of DB
{"id":13,"type":"Router","ieeeAddr":"0x540f57fffefdc4ee","nwkAddr":30287,"manufId":4098,"manufName":"**_TZ3000_u5u4cakc**","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":

Koenkk, 2022-01-02:
Looking at your logs zigbee2mqtt publishes values to MQTT: `'{"current":0.89,"energy":78.82,"linkquality":75,"power":148,"state":"ON","voltage":233}'`, so this seems to be a Domoticz issue to me.

WBoerhoop, 2022-01-02:
Check, that's right, because BWPL01 and 02 are not SHP-15 but SHP-13 and have been running for a while and working as expected and so is this data.

The 3 new Blitzwolf SHP-15 are new for 2 days, there is no data in the log.
And after I pair them, the only activity that works is turning on and off via the device switch.

Perhaps it have something to do with what i read on https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html

In the options I read that the plug must be actively approached to read the electrical data

"Options
Use device type specific configuration

Measurement_poll_interval: This device does not support electrical measurement reporting, so it is polled instead. The default poll interval is 60 seconds, set to -1 to disable. The value must be a number with a minimum value of -1"

Koenkk, 2022-01-02:
@WBoerhoop does any value show up when going to the device in the z2m frontend -> exposes -> and refresh the power or energy value?

WBoerhoop, 2022-01-02:
Will do a test with copy through the laser printer connected to BWPL03 that's a SHP-15, picture voor and after the copy
[image]
[image]

No difference at all

Koenkk, 2022-01-03:
Could you provide the zigbee2mqtt debug log at the moment you press the refresh button for power in the z2m frontend?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

WBoerhoop, 2022-01-03:
Ohhh, first got an error message that zigbee was already started, then I started domoticz all over again and data came in from the SHP-15 plugs. Please keep an eye on it to see if it continues to work normally, thank you very much in any case.

WBoerhoop, 2022-01-06:
Everything is working properly, so close the issue

closed 2022-01-06 by WBoerhoop