Koenkk/zigbee2mqtt #10494: [Feature request]: Xiaomi SP-EUC01 (aqara EU smart plug) add power_outage_memory and overload_protection

Vanntag, 2022-01-01:
### Is your feature request related to a problem? Please describe

Aqara SP-EUC01 don’t have « power_outage_memory » and « overload_protection »

### Describe the solution you'd like

Sorry, but for the Xiaomi SP-EUC01 (aqara EU smart plug), is it possible to add « power_outage_memory » and « overload_protection »? Because if With aqara bridge, it’s possible to manage those things. In advance, thanks for consider this. 

### Describe alternatives you've considered

Hum…..change for an other plug?

### Additional context

I want to plug my freezer on it, and if power went down, when it restart, it would be cool if the plug also turn on.

rdelville, 2022-01-13:
Hello @Vanntag,

I found another issue regarding this request (https://github.com/Koenkk/zigbee-herdsman-converters/issues/3411). Unfortunately, this one is closed

thschenk, 2022-02-06:
I have the power_outage_memory implemented here:
https://github.com/thschenk/zigbee-herdsman-converters/commit/ac9d7a55e6cdfb0de41e86064ebf62d71a981364

I tested it on my SP-EUC01 plug (firmware date 09-22-2020) and it works.

PierreMacherel, 2022-03-11:
@thschenk could you submit a PR?

marsounjan, 2022-03-20:
PR done ;)
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4026

Koenkk, 2022-03-20:
Added thanks to @marsounjan 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-03-20 by Koenkk