Koenkk/zigbee2mqtt #10496: [Problem]: Z2M 1.22.2 → "Check for new updates" yields "..(No image available for imageType '285')" for Philips Hue devices

SHxKM, 2022-01-02:
### What happened?

When I updated to Z2M, all my Hue devices showed a new firmware version (1.93.3) is available. I updated most of them.

For all of them now, when I click on "Check for new updates", I get this error:

> Failed to check if update available for '[[DEVICE NAME]]' (No image available for imageType '285')

The `ImageType '285'` is constant.

### What did you expect to happen?

If there is no update, and everything is OK with the device, get the "no update available"/"device up to date" message.

### How to reproduce it (minimal and precise)

1. Upgrade Philips Hue firmware version (`1.93.3`) on a device
2. Go to OTA tab in Z2M UI
3. Click on "Check for new updates"

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

0x26580700

### Adapter

Conbee II

### Debug log

Failed to check if update available for 'outdb_pantry_window_impress_l' (No image available for imageType '285')

TheJulianJES, 2022-01-02:
What are the device models that are requesting image type 285?

SHxKM, 2022-01-02:
> What are the device models that are requesting image type 285? Do all of your Hue devices request that image type now?

@TheJulianJES  - Seems like that. I've had an Econic, an Impress, and an Aurelle do that. Let me know if you need model IDs of each.

SHxKM, 2022-01-02:
@TheJulianJES - your question made me wonder if this seems to be scoped to a specific device class. I haven't tested ALL my devices that have this new firmware but it seems like any device with a GU10 does _not_ exhibit these problems. All Econics I've checked do, all Impresses do, same for the Adores, and Aurelles.

So it seems that somehow the GU10s are unaffected, but only them..

TheJulianJES, 2022-01-02:
Do you know if your GU10s support Bluetooth?

SHxKM, 2022-01-02:
> Do you know if your GU10s support Bluetooth?

Yes they do.

TheJulianJES, 2022-01-02:
I've just checked again and Philips/Signify doesn't provide any firmware updates with image type 285.
For now, I wouldn't worry about this error. It's possible that when Signify releases another batch of updates, they'll provide an update with image type 285 for the models requesting that.

SHxKM, 2022-01-02:
> I've just checked again and Philips/Signify doesn't provide any firmware updates with image type 285. For now, I wouldn't worry about this error. It's possible that when Signify releases another batch of updates, they'll provide an update with image type 285 for the models requesting that.
> 
> Basically, it seems like some models now request a different image type for updates.

Can you elaborate a bit more? I'm not super-familiar with how OTA updates are checked. I thought Z2M is checking against its own-repository of available images, so if Philips aren't providing 285 (and we know that), why would this yield an error? Maybe it should be suppressed for now?

I feel quite uneasy when it comes to firmware related errors TBH. But maybe it's just ignorance.

TheJulianJES, 2022-01-02:
Yes, Z2M basically checks all devices against this file: https://github.com/Koenkk/zigbee-OTA/blob/master/index.json
As of right now, this also contains links to the latest Hue OTA updates.

The simplified version of Zigbee OTAs is as following:
- Z2M sends image_notify command to the bulb when using the "Check for new updates" buton
- bulb responds with a question: "Is an image with manufacturer code 4107 (Philips) and image type xxx available?"
- Z2M checks if there are any (newer) OTA files for that manufacturer code and image type (and possibly initiates the update)

I'm not actually using Z2M but the "no image available..." error seems like it should be a warning instead. It's possible that it does that for all device with no OTA updates available at all or that Z2M just doesn't expect that the bulb asks for a difference image type at one point.

In either case, "the error" isn't directly coming from the device but rather Z2M. It's not really an issue and we'll see how Signify handles their next batch of updates.

Koenkk, 2022-01-02:
@TheJulianJES is right, we have to wait for Philips to start providing ota for these and once they do at them to https://github.com/Koenkk/zigbee-OTA/blob/master/index.json . So not something we can solve from z2m now.

SHxKM, 2022-01-02:
@Koenkk @TheJulianJES Thanks both, if you say it's a non-issue I'm fine with this. Closing.

closed 2022-01-02 by SHxKM

alexw1982, 2022-02-21:
I am having the same issue on a hue ensis lamp - which is not necessarily a new lamp but has been on the market for 1+ years. The updates are available via bluetooth but I get the same error within Z2M. I am not an expert on the subject matter - but are we sure that the issue is with Philips?