Koenkk/zigbee2mqtt #10497: [Problem]: 3A/Nue 2-gang wall switches now being identified as double power point

bigfeetneedbigboots, 2022-01-02:
### What happened?

I rebuilt my Zigbee2Mqtt server and re-joined all of my home's devices however the 2-gang wall switches that were previously added correctly as HGZB-42 devices, have now all added themselves in as NUE-AUWZO2 devices. The 1-gang (HGZB-41) and 3-gang (HGZB-43) have all joined correctly - only the 2-gang have issues.

### What did you expect to happen?

Expected HGZB-42 to be added as a HGZB-42 (not NUE-AUWZO2) as they did when I first built my Zigbee2Mqtt server about 14-18 months ago.

### How to reproduce it (minimal and precise)

Join HGZB-42 device.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Debug log

Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: Currently 55 devices are joined:
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: off_n_3sw (0x00158d0004578965): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: hal_e_1sw (0x00158d000476ae42): HGZB-41 - Nue / 3A Smart one gang wall switch (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: off_se_cdl (0x00158d000457aef1): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: off_sw_cdl (0x00158d00045786d7): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: off_ne_cdl (0x00158d0004579bc3): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: off_nw_cdl (0x00158d00046269cc): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: liv_se_2sw (0x00158d000476f56a): NUE-AUWZO2 - Nue / 3A Smart Zigbee double power point (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: kit_n_thl (0xcc86ecfffed117c4): LCZ030 - TuYa Temperature & humidity & illuminance sensor with display (EndDevice)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: ent_nw_3sw (0x00158d000449cfbc): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: din_ne_3sw (0x00158d00045789d7): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: pow_s_2sw (0x00158d000476f624): NUE-AUWZO2 - Nue / 3A Smart Zigbee double power point (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: rum_w_3sw (0x00158d0004629806): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: bat_e_3sw (0x00158d00045784b2): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: rum_sw_2sw (0x00158d00045785cf): NUE-AUWZO2 - Nue / 3A Smart Zigbee double power point (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: tex_nw_2sw (0x00158d00044a76ad): NUE-AUWZO2 - Nue / 3A Smart Zigbee double power point (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: the_sw_3sw (0x00158d0004578a85): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: rum_nw_cdl (0x00158d000457ae55): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: the_sw_cdl (0x00158d000457b5a4): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: the_ne_cdl (0x00158d000476b2f6): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: kit_ne2_3sw (0x00158d0004578af5): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: but_ne_1sw (0x00158d000457780e): HGZB-41 - Nue / 3A Smart one gang wall switch (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: kit_ne1_3sw (0x00158d00045787c5): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: hal_w3_3sw (0x00158d000449cf3a): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: hal_w2_1sw (0x00158d0004767fe9): HGZB-41 - Nue / 3A Smart one gang wall switch (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: stu_n_3sw (0x00158d000449dbd0): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: sto_n_1sw (0x00158d000449dd70): HGZB-41 - Nue / 3A Smart one gang wall switch (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: hal_w1_3sw (0x00158d00045786f9): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: lau_se_2sw (0x00158d0004622421): NUE-AUWZO2 - Nue / 3A Smart Zigbee double power point (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: gar_n_1sw (0x00158d000476b3e7): HGZB-41 - Nue / 3A Smart one gang wall switch (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: gar_e_3sw (0x00158d000457876b): HGZB-43 - Nue / 3A Smart light switch - 3 gang v2.0 (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: hal_w1_cdl (0x00158d0004578408): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: hal_e2_cdl (0x00158d000457b55a): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: hal_w2_cdl (0x00158d000476b2fb): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: hal_w3_cdl (0x00158d000448c057): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: hal_e1_cdl (0x00158d0004578494): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: stu_e_cdl (0x00158d00045784b3): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: stu_w_cdl (0x00158d0004622131): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: kit_n_cdl (0x00158d0004578b26): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: kit_sw_cdl (0x00158d000457b611): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: kit_w_cdl (0x00158d00045786b5): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: kit_nw_cdl (0x00158d0004579ba0): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: spa_w_1sw (0x00158d000457abe1): HGZB-41 - Nue / 3A Smart one gang wall switch (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: gue_s_2sw (0x00158d00044a769d): NUE-AUWZO2 - Nue / 3A Smart Zigbee double power point (Router)
[... cut ...]
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: men_w_cdl (0x00158d000457ac91): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: bat_th (0xec1bbdfffee4bf5a): TS0201 - TuYa Temperature & humidity sensor with display (EndDevice)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: men-th (0xec1bbdfffee4c306): TS0201 - TuYa Temperature & humidity sensor with display (EndDevice)

Koenkk, 2022-01-02:
@danps1 (https://github.com/Koenkk/zigbee-herdsman-converters/pull/3477) seems that this model is reused for different devices, could you both share your `data/database.db` entry of this device?

danps1, 2022-01-02:
Hi @Koenkk,

Yes sure. Hopefully the manufacturer hasn't been using the same modelId on multiple different devices. I now have 3 of these, here are the db entries...

```
{"id":56,"type":"Router","ieeeAddr":"0xXXXXXXXXXXXXXXXX","nwkAddr":16565,"manufId":0,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"LXN-2S27LX1.0","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":0,"inClusterList":[0,4,3,6,5,4096,8],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"LXN-2S27LX1.0","manufacturerName":"3A Smart Home DE","powerSource":1,"zclVersion":3,"appVersion":9,"dateCode":"20210703","swBuildId":"1.1.1"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":0,"inClusterList":[0,4,3,6,5,4096,8],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":9,"dateCode":"20210703","swBuildId":"1.1.1","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1641120608038,"useImplicitCheckin":true}
{"id":57,"type":"Router","ieeeAddr":"0xXXXXXXXXXXXXXXXX","nwkAddr":49368,"manufId":0,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"LXN-2S27LX1.0","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":0,"inClusterList":[0,4,3,6,5,4096,8],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"LXN-2S27LX1.0","manufacturerName":"3A Smart Home DE","powerSource":1,"zclVersion":3,"appVersion":9,"dateCode":"20210703","swBuildId":"1.1.1"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":0,"inClusterList":[0,4,3,6,5,4096,8],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":9,"dateCode":"20210703","swBuildId":"1.1.1","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1641120652821,"useImplicitCheckin":true}
{"id":60,"type":"Router","ieeeAddr":"0xXXXXXXXXXXXXXXXX","nwkAddr":11953,"manufId":0,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"LXN-2S27LX1.0","epList":[1,2],"endpoints":{"1":{"profId":49246,"epId":1,"devId":0,"inClusterList":[0,4,3,6,5,4096,8],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"LXN-2S27LX1.0","manufacturerName":"3A Smart Home DE","powerSource":1,"zclVersion":3,"appVersion":9,"dateCode":"20210703","swBuildId":"1.1.1"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":0,"inClusterList":[0,4,3,6,5,4096,8],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":9,"dateCode":"20210703","swBuildId":"1.1.1","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1641120602464,"useImplicitCheckin":true}
```

bigfeetneedbigboots, 2022-01-02:
In trying to workaround the issue, I created a .js file for the 2-gang wall switches and setup an external converter. It worked! Thanks very much for this awesome bit of software and all the work you all do in order to maintain it. 

Here is the working .js file content for the 2-gang wall switches which I hope is helpful to you/others:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LXN-2S27LX1.0'],
    model: 'HGZB-42',
    vendor: 'Nue / 3A',
    description: 'Smart 2-gang wall switch',
    extend: extend.switch(),
    exposes: [e.switch().withEndpoint('top'), e.switch().withEndpoint('bottom')],
    endpoint: (device) => {
        return {'top': 11, 'bottom': 12};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(11), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(12), coordinatorEndpoint, ['genOnOff']);
    }
};

module.exports = definition;`

Koenkk, 2022-01-02:
@bigfeetneedbigboots can you share the data/database.db entry of your device?

bigfeetneedbigboots, 2022-01-02:
[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/7799798/database.txt)
Renamed from .db to .txt so I could upload it.

Koenkk, 2022-01-02:
thanks, fixed

@bigfeetneedbigboots @danps1 please check if your device is recognised correctly in the latest dev

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

vatsanvasu, 2022-01-03:
Hi Guys,

I have the same problem and noticed that all the 2 gang switches have joined back as NUE-AUWZO2 instead of HGZB-42 after the latest zigbee2mqtt add on update on home assistant. I am a novice with HA and wanted to know what is the easiest way to resolve this? Is there a way to fix it without changing to the edge version of zigbee2mqtt?

danps1, 2022-01-03:
> thanks, fixed
> 
> @bigfeetneedbigboots @danps1 please check if your device is recognised correctly in the latest dev
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

So I just checked with Edge install of HA Add-on `1.22.2-dev commit: 23bdc18` - and all 3 of my PowerPoints are showing up as PowerPoints

Thanks @Koenkk

bigfeetneedbigboots, 2022-01-03:
> https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

Updated to 1.22.2-dev commit: 23bdc186, tested and working fine. Thank you @Koenkk and @danps1 for your help.

closed 2022-01-03 by bigfeetneedbigboots

danps1, 2022-01-03:
> Hi Guys,
>
> I have the same problem and noticed that all the 2 gang switches have joined back as NUE-AUWZO2 instead of HGZB-42 after the latest zigbee2mqtt add on update on home assistant. I am a [...]

Hi @vatsanvasu,

Sorry no easy way without running the Edge version, but temporarily running Edge is easy, and will fix the problem...

1. Install the Edge HA Addon from the same repository
2. Copy the config from zigbee2mqtt into clipboard
3. Stop zigbee2mqtt
4. Paste config to Edge
5. Start Edge
6. Check that devices are recognised correctly

Once the changes are merged into main, you just switch back.

vatsanvasu, 2022-01-03:
Hi @danps1 
Thanks for the response. Just want clarification on step 2 i.e copy the config from zigbee2mqtt.
What config do I need to copy? Is it just the configuration.yaml file?

danps1, 2022-01-03:
Hi @vatsanvasu,

Just go into the Configuration tab of the add-on and copy everything there...

[image]

lukima61, 2022-02-05:
Hello,
I have a very similar problem with a shutter switch. It also has the model number 'LXN-2S27LX1.0'. I can pair it successfully, but it is not supported :-(
In the nue_3a.js file 'LXN-2S27LX1.0' is entered, but not for a shutter switch. I then tried to integrate the shutter switch according to the instructions "Support new devices". Unfortunately my attempt fails directly after the restart of the container, because it aborts directly.
Can someone help me?

lukima61, 2022-03-15:
Hello,
I am now using zigbee2mqtt version 1.22.15, the container starts again. However, I still can't solve my problem, and have also opened the ticket [.](https://github.com/Koenkk/zigbee2mqtt/issues/11823) It would be nice if someone could help me.
Best regards

olgru, 2022-09-28:
I do have the same issue thank @lukima61 - shutterswitch and no change to get this working...

lukima61, 2022-10-02:
Hello,
I have not been able to solve the problem yet, not even with the manufacturer. No feedback came from the Zigbee community unfortunately, too bad.

Best regards
lukima61