Koenkk/zigbee2mqtt #10498: [Problem]: 3A/Nue RGBW Downlights (LXN-2S27LX1.0) have joined as Strip Light Controllers

bigfeetneedbigboots, 2022-01-02:
### What happened?

I rebuilt my Zigbee2Mqtt server however when I re-joined my 3A/Nue RGBW downlights (Zigbee Model: LXT56-LS27LX1.7), they joined as Model 3A12S-15, not the Model HGZB-DLC4-N12B that they are,.

### What did you expect to happen?

Joining Zigbee Model: LXT56-LS27LX1.7 creates devices as Model HGZB-DLC4-N12B

### How to reproduce it (minimal and precise)

Join a 3A/Nue 12W RGBW downlight LXT56-LS27LX1.7

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Debug log

Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: off_se_cdl (0x00158d000457aef1): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: off_sw_cdl (0x00158d00045786d7): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: off_ne_cdl (0x00158d0004579bc3): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)
Jan 02 15:33:14 UpsOffEastZ2M npm[1455]: Zigbee2MQTT:info  2022-01-02 15:33:14: off_nw_cdl (0x00158d00046269cc): 3A12S-15 - Nue / 3A Smart Zigbee 3.0 strip light controller (Router)

Koenkk, 2022-01-02:
@zibous  seems this model is re-used across device (https://github.com/Koenkk/zigbee2mqtt/issues/7292) , can you both provide your data/database.db entry of this device?

zibous, 2022-01-02:
@Koenkk 

> @zibous seems this model is re-used across device (#7292) , can you both provide your data/database.db entry of this device?

```json
{"id":1,"type":"Coordinator","ieeeAddr":"0x00212effff060927","nwkAddr":0,"manufId":4405,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[25,10],"outClusterList":[1280],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":100,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}
{"id":2,"type":"EndDevice","ieeeAddr":"0x680ae2fffe81f382","nwkAddr":26759,"manufId":4476,"modelId":"TRADFRI remote control","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,1,3,32,4096,64636],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI remote control","zclVersion":3,"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190401","swBuildId":"2.3.014"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":34}}},"binds":[{"cluster":6,"groupID":901,"type":"group"},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00212effff060927","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190401","swBuildId":"2.3.014","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1104804361},"lastSeen":1641122168075}
{"id":3,"type":"Group","groupID":901,"members":[],"meta":{}}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00124b00226a2b97","nwkAddr":62315,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"TH01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"TH01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":5,"hwVersion":1,"dateCode":"20201026"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2257}},"msRelativeHumidity":{"attributes":{"measuredValue":3384}},"genPowerCfg":{"attributes":{"batteryVoltage":29,"batteryPercentageRemaining":150}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00212effff060927","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00212effff060927","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00212effff060927","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":5,"maxRepIntval":1800,"repChange":20},{"cluster":1029,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":5,"hwVersion":1,"dateCode":"20201026","zclVersion":1,"interviewCompleted":true,"meta":{"configured":15672516},"lastSeen":1641124621109}
{"id":5,"type":"EndDevice","ieeeAddr":"0x00124b00226b5a3a","nwkAddr":16972,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"TH01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"TH01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":5,"hwVersion":1,"dateCode":"20201026"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2264}},"genPowerCfg":{"attributes":{"batteryVoltage":29,"batteryPercentageRemaining":137}},"msRelativeHumidity":{"attributes":{"measuredValue":3482}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00212effff060927","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00212effff060927","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00212effff060927","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":5,"maxRepIntval":1800,"repChange":20},{"cluster":1029,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":5,"hwVersion":1,"dateCode":"20201026","zclVersion":1,"interviewCompleted":true,"meta":{"configured":15672516},"lastSeen":1641123424583}
{"id":6,"type":"EndDevice","ieeeAddr":"0x00124b002242a5e9","nwkAddr":60551,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"66666","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"66666","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":4,"hwVersion":1,"dateCode":"20200718"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":1862}},"msRelativeHumidity":{"attributes":{"measuredValue":4209}},"genPowerCfg":{"attributes":{"batteryVoltage":29,"batteryPercentageRemaining":148}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00212effff060927","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00212effff060927","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00212effff060927","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepInt [... cut ...] usterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_yw7cahqs","powerSource":3,"zclVersion":3,"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff060927","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1641123629516}
```

bigfeetneedbigboots, 2022-01-02:
[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/7799827/database.db.txt)
@Koenkk database attached as requested.

[nue3acdl.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/7799828/nue3acdl.js.txt)
Worked around issue using attached external converter file for incorrectly recognised HGZB-DLC4-N12B.

Koenkk, 2022-01-02:
@bigfeetneedbigboots this device has exactly the same properties, meaning we cannot distinguish the 3A12S-15 and   HGZB-DLC4-N12B from each other.

The external converter is indeed a fine workaround.

closed 2022-01-02 by Koenkk