Koenkk/zigbee2mqtt #10504: [Problem]: Keypad dis/arm mqtt messages are empty

Rockman18, 2022-01-02:
### What happened?

Since last update (1.22.2), my XHK1-EU keypad doesn't react to arming / disarming commands.
MQTT messages are empty. No action, action_code, action_transaction, action_zone.

[image]

### What did you expect to happen?

Fill action, action_code, action_transaction and action_zone like in 1.22.1

[image]

### How to reproduce it (minimal and precise)

Update to 1.22.2 and enter a code on the keypad.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210120

### Adapter

Electrolama zzh

### Debug log

1.22.1 logs :
[log_1.22.1.txt](https://github.com/Koenkk/zigbee2mqtt/files/7798963/log_1.22.1.txt)

1.22.2 logs :
[log_1.22.2.txt](https://github.com/Koenkk/zigbee2mqtt/files/7798964/log_1.22.2.txt)

ikke-zelf, 2022-01-02:
Same Problem here.
Tamper gets through but actions and modes don't.

Koenkk, 2022-01-02:
Did an attempt to fix it, please let me know if it works

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ikke-zelf, 2022-01-02:
Hi Koen, can you guide me in how to test this?
I'm running on Home Assistant and rolled-back the zigbee2mqtt latest version to the one before.

Koenkk, 2022-01-03:
@ikke-zelf switch to the edge version (uninstall normal, install edge (no repairing required))

ikke-zelf, 2022-01-03:
Just installed the edge version.
It seems to work, but some observations:

1. Arming the device gives two messages from the keypad towards the node-red. the first as expected, the second empty.

{"action":"arm_day_zones","action_code":"1234","action_transaction":27,"action_zone":23,"battery":100,"battery_low":false,"contact":true,"linkquality":15,"occupancy":true,"presence":null,"tamper":false,"temperature":15,"voltage":6100}
{"action":"","action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"battery_low":false,"contact":true,"linkquality":15,"occupancy":true,"presence":null,"tamper":false,"temperature":15,"voltage":6100}

2. Disarming the device gives two messages from the keypad towards the node-red. the first as expected, the second empty.

{"action":"disarm","action_code":"1234","action_transaction":30,"action_zone":23,"battery":100,"battery_low":false,"contact":true,"linkquality":15,"occupancy":true,"presence":null,"tamper":false,"temperature":15,"voltage":6100}
{"action":"","action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"battery_low":false,"contact":true,"linkquality":15,"occupancy":true,"presence":null,"tamper":false,"temperature":15,"voltage":6100}

3.  the disarm comes also with a warning message.

{"warning":{"duration":0,"mode":"stop","strobe":false}}

-----
I will now go back to the dev/naster version and test if the messages come in twice as well and to verify if the 'warning' message appears in that release also.

Koenkk, 2022-01-03:
@ikke-zelf that is expected when the home assistant integration is enabled. https://www.zigbee2mqtt.io/guide/faq/#why-is-the-action-property-always-empty

ikke-zelf, 2022-01-03:
okay, didn't knew that.
Then your change seems to do the trick.

are you going to put it in the next release (and if so, is that every month or...)

Koenkk, 2022-01-03:
The fix will be included on the next release on 1 February.

closed 2022-01-03 by Koenkk

t0ny-peng, 2022-01-09:
Same problem here with Xfinity keypad. I can see the debug output in Zigbee2Mqtt but no message in MQTT topic. The version is 1.22.2-1 as HA addon. 

[image]

Koenkk, 2022-01-24:
I've improved the fix a bit, can you guys check if it still works correctly on the latest-dev?

ikke-zelf, 2022-01-24:
Hi Koen, I installed and checked and all seems to be oke and working as far as I can see.

rpatel3001, 2022-02-09:
Hi @Koenkk, I just got this keypad and am having similar problems. Both 1.23.0 and latest-dev exposed values for battery percent, battery voltage, and occupancy, which were null on 1.22.2. However the remaining values are all null.

kbgators, 2024-01-14:
> Same problem here with Xfinity keypad. I can see the debug output in Zigbee2Mqtt but no message in MQTT topic. The version is 1.22.2-1 as HA addon.
> 
> [image]

@ikke-zelf - can you share your node red code please?  I'm pulling my hair out, having bought 3 keypads.  
I cannot get the keypad LED to show a color status - it's always Green and never changes to Red when armed.

I can arm and disarm from the keypads (using a bluepring and Alarmo), but no status is shown on the keypad itself

ikke-zelf, 2024-01-15:
@kbgators the issue/solution is that you need to send the code somewhat the same twice.
first send it with a transaction
then without.

`{
    "arm_mode": {
        "mode": "arm_night_zones",
        "transaction": 23
    }
}`

`{
    "arm_mode": {
        "mode": "arm_night_zones"
    }
}`

kbgators, 2024-01-15:
@ikke-zelf - Thank you for replying!  I might be doing something wrong. 
The LED stays green and doesn't change to red.
Are you using "set" as part of the MQTT Publish Topic?

Here is the Automation I just used:
[image]

and this is the JSON in Node Red for the service mqtt.publish:

[image]

[image]

[image]

ikke-zelf, 2024-01-15:
[{"id":"e797c1b28720f32d","type":"tab","label":"Flow 1","disabled":false,"info":"","env":[]},{"id":"0aea3fed5e5d3cec","type":"link in","z":"e797c1b28720f32d","name":"link_in_Keypads","links":["2e0c14ead2f1acbf","32fe0534ab3ad8e3","8d7d29d68c686f38","c6f9cf46803e1bfc","c71f42bf7e80e94f","e495f7cdc914bad7","f9139445cfa31234"],"x":685,"y":400,"wires":[["6a4a96b0b85bb71b","042c9cb5f4e82253"]]},{"id":"042c9cb5f4e82253","type":"mqtt out","z":"e797c1b28720f32d","name":"mqtt_KeypadBoven","topic":"zigbee2mqtt/Keypad_Boven/set","qos":"2","retain":"true","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"a7e95831.3143b8","x":840,"y":420,"wires":[]},{"id":"6a4a96b0b85bb71b","type":"debug","z":"e797c1b28720f32d","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":1120,"y":500,"wires":[]},{"id":"4b5b1f70ea657776","type":"api-call-service","z":"e797c1b28720f32d","name":"SetArmMode","server":"6d000dde.3e0b04","version":5,"debugenabled":true,"domain":"input_text","service":"set_value","areaId":[],"deviceId":[],"entityId":["input_text.armstatus"],"data":"{\"value\":msg.payload.arm_mode.mode}","dataType":"jsonata","mergeContext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","x":820,"y":520,"wires":[[]]},{"id":"3752a03f1afb6954","type":"comment","z":"e797c1b28720f32d","name":"Set arm mode in HA","info":"","x":840,"y":482,"wires":[]},{"id":"e30b4e59cee711b7","type":"comment","z":"e797c1b28720f32d","name":"Send Status and Mode to Keypads","info":"","x":800,"y":340,"wires":[]},{"id":"9ec944e9571693c7","type":"template","z":"e797c1b28720f32d","name":"Arm night zones pre","field":"payload","fieldType":"msg","format":"json","syntax":"mustache","template":"{\n    \"arm_mode\": {\n        \"mode\": \"arm_night_zones\",\n        \"transaction\": 23\n    }\n}","output":"json","x":110,"y":260,"wires":[["f561248959f7bff7","2e0c14ead2f1acbf"]]},{"id":"f561248959f7bff7","type":"template","z":"e797c1b28720f32d","name":"Arm night zones","field":"payload","fieldType":"msg","format":"json","syntax":"mustache","template":"{\n    \"arm_mode\": {\n        \"mode\": \"arm_night_zones\"\n    }\n}","output":"json","x":330,"y":300,"wires":[["2e0c14ead2f1acbf","5b0e9117ce101193"]]},{"id":"2e0c14ead2f1acbf","type":"link out","z":"e797c1b28720f32d","name":"send_ArmNightZones","mode":"link","links":["0aea3fed5e5d3cec"],"x":525,"y":260,"wires":[]},{"id":"5b0e9117ce101193","type":"link out","z":"e797c1b28720f32d","name":"ArmMode HA","links":["9977f5fae431ff37"],"x":585,"y":300,"wires":[]},{"id":"c8b47707ae3ec2de","type":"comment","z":"e797c1b28720f32d","name":"Debugging","info":"","x":1110,"y":460,"wires":[]},{"id":"9977f5fae431ff37","type":"link in","z":"e797c1b28720f32d","name":"link_in_HomeAssistant","links":["128eeedbaffb2293","3708c4bd2da32d55","3c647a3d41b3a454","74a3e38daf61e633","86c376fc4ce311d3","5b0e9117ce101193","b4e9383f07ba06f0"],"x":685,"y":520,"wires":[["4b5b1f70ea657776"]]},{"id":"7397cedf181d5211","type":"comment","z":"e797c1b28720f32d","name":"MQTT link","info":"","x":500,"y":220,"wires":[]},{"id":"d7101c760ca4a16c","type":"comment","z":"e797c1b28720f32d","name":"Set Status and Modes","info":"","x":120,"y":200,"wires":[]},{"id":"d8485bfd0aeda5c7","type":"comment","z":"e797c1b28720f32d","name":"ArmMode link","info":"","x":630,"y":260,"wires":[]},{"id":"a7e95831.3143b8","type":"mqtt-broker","name":"MasterPi","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":false,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""},{"id":"6d000dde.3e0b04","type":"server","name":"Home Assistant","version":5,"addon":true,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":false,"cacheJson":true,"heartbeat":false,"heartbeatInterval":30,"areaSelector":"friendlyName","deviceSelector":"friendlyName","entitySelector":"friendlyName","statusSeparator":"at: ","statusYear":"hidden","statusMonth":"short","statusDay":"numeric","statusHourCycle":"h23","statusTimeFormat":"h:m","enableGlobalContextStore":true}]

ikke-zelf, 2024-01-15:
@kbgators can you do something with the exported json for one state ?

kbgators, 2024-01-15:
Not sure of your question @ikke-zelf.  Can you explain?
I tried to import that text into node-red but it gives an error

ikke-zelf, 2024-01-15:
`[
    {
        "id": "e797c1b28720f32d",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "0aea3fed5e5d3cec",
        "type": "link in",
        "z": "e797c1b28720f32d",
        "name": "link_in_Keypads",
        "links": [
            "2e0c14ead2f1acbf",
            "32fe0534ab3ad8e3",
            "8d7d29d68c686f38",
            "c6f9cf46803e1bfc",
            "c71f42bf7e80e94f",
            "e495f7cdc914bad7",
            "f9139445cfa31234"
        ],
        "x": 685,
        "y": 400,
        "wires": [
            [
                "6a4a96b0b85bb71b",
                "042c9cb5f4e82253"
            ]
        ]
    },
    {
        "id": "042c9cb5f4e82253",
        "type": "mqtt out",
        "z": "e797c1b28720f32d",
        "name": "mqtt_KeypadBoven",
        "topic": "zigbee2mqtt/Keypad_Boven/set",
        "qos": "2",
        "retain": "true",
        "respTopic": "",
        "contentType": "",
        "userProps": "",
        "correl": "",
        "expiry": "",
        "broker": "a7e95831.3143b8",
        "x": 840,
        "y": 420,
        "wires": []
    },
    {
        "id": "6a4a96b0b85bb71b",
        "type": "debug",
        "z": "e797c1b28720f32d",
        "name": "",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 1120,
        "y": 500,
        "wires": []
    },
    {
        "id": "4b5b1f70ea657776",
        "type": "api-call-service",
[... 191 lines cut ...]
    },
    {
        "id": "6d000dde.3e0b04",
        "type": "server",
        "name": "Home Assistant",
        "version": 5,
        "addon": true,
        "rejectUnauthorizedCerts": true,
        "ha_boolean": "y|yes|true|on|home|open",
        "connectionDelay": false,
        "cacheJson": true,
        "heartbeat": false,
        "heartbeatInterval": 30,
        "areaSelector": "friendlyName",
        "deviceSelector": "friendlyName",
        "entitySelector": "friendlyName",
        "statusSeparator": "at: ",
        "statusYear": "hidden",
        "statusMonth": "short",
        "statusDay": "numeric",
        "statusHourCycle": "h23",
        "statusTimeFormat": "h:m",
        "enableGlobalContextStore": true
    }
]

ikke-zelf, 2024-01-15:
@kbgators try that one (first one was 'compact'. this one 'formatted')

kbgators, 2024-01-15:
@ikke-zelf  - sorry.  Same error.  I assume you want me to import that into node red?

[image]

ikke-zelf, 2024-01-15:
[flows (1).json](https://github.com/Koenkk/zigbee2mqtt/files/13942962/flows.1.json)

and l ike this?
yes I want you to import so you can compare... cause I also am using 'set'

ikke-zelf, 2024-01-15:
`[{"id":"e797c1b28720f32d","type":"tab","label":"Flow 1","disabled":false,"info":"","env":[]},{"id":"0aea3fed5e5d3cec","type":"link in","z":"e797c1b28720f32d","name":"link_in_Keypads","links":["2e0c14ead2f1acbf","32fe0534ab3ad8e3","8d7d29d68c686f38","c6f9cf46803e1bfc","c71f42bf7e80e94f","e495f7cdc914bad7","f9139445cfa31234"],"x":685,"y":400,"wires":[["6a4a96b0b85bb71b","042c9cb5f4e82253"]]},{"id":"042c9cb5f4e82253","type":"mqtt out","z":"e797c1b28720f32d","name":"mqtt_KeypadBoven","topic":"zigbee2mqtt/Keypad_Boven/set","qos":"2","retain":"true","respTopic":"","contentType":"","userProps":"","correl":"","expiry":"","broker":"a7e95831.3143b8","x":840,"y":420,"wires":[]},{"id":"6a4a96b0b85bb71b","type":"debug","z":"e797c1b28720f32d","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":1120,"y":500,"wires":[]},{"id":"4b5b1f70ea657776","type":"api-call-service","z":"e797c1b28720f32d","name":"SetArmMode","server":"6d000dde.3e0b04","version":5,"debugenabled":true,"domain":"input_text","service":"set_value","areaId":[],"deviceId":[],"entityId":["input_text.armstatus"],"data":"{\"value\":msg.payload.arm_mode.mode}","dataType":"jsonata","mergeContext":"","mustacheAltTags":false,"outputProperties":[],"queue":"none","x":820,"y":520,"wires":[[]]},{"id":"3752a03f1afb6954","type":"comment","z":"e797c1b28720f32d","name":"Set arm mode in HA","info":"","x":840,"y":482,"wires":[]},{"id":"e30b4e59cee711b7","type":"comment","z":"e797c1b28720f32d","name":"Send Status and Mode to Keypads","info":"","x":800,"y":340,"wires":[]},{"id":"9ec944e9571693c7","type":"template","z":"e797c1b28720f32d","name":"Arm night zones pre","field":"payload","fieldType":"msg","format":"json","syntax":"mustache","template":"{\n    \"arm_mode\": {\n        \"mode\": \"arm_night_zones\",\n        \"transaction\": 23\n    }\n}","output":"json","x":110,"y":260,"wires":[["f561248959f7bff7","2e0c14ead2f1acbf"]]},{"id":"f561248959f7bff7","type":"template","z":"e797c1b28720f32d","name":"Arm night zones","field":"payload","fieldType":"msg","format":"json","syntax":"mustache","template":"{\n    \"arm_mode\": {\n        \"mode\": \"arm_night_zones\"\n    }\n}","output":"json","x":330,"y":300,"wires":[["2e0c14ead2f1acbf","5b0e9117ce101193"]]},{"id":"2e0c14ead2f1acbf","type":"link out","z":"e797c1b28720f32d","name":"send_ArmNightZones","mode":"link","links":["0aea3fed5e5d3cec"],"x":525,"y":260,"wires":[]},{"id":"5b0e9117ce101193","type":"link out","z":"e797c1b28720f32d","name":"ArmMode HA","links":["9977f5fae431ff37"],"x":585,"y":300,"wires":[]},{"id":"c8b47707ae3ec2de","type":"comment","z":"e797c1b28720f32d","name":"Debugging","info":"","x":1110,"y":460,"wires":[]},{"id":"9977f5fae431ff37","type":"link in","z":"e797c1b28720f32d","name":"link_in_HomeAssistant","links":["128eeedbaffb2293","3708c4bd2da32d55","3c647a3d41b3a454","74a3e38daf61e633","86c376fc4ce311d3","5b0e9117ce101193","b4e9383f07ba06f0"],"x":685,"y":520,"wires":[["4b5b1f70ea657776"]]},{"id":"7397cedf181d5211","type":"comment","z":"e797c1b28720f32d","name":"MQTT link","info":"","x":500,"y":220,"wires":[]},{"id":"d7101c760ca4a16c","type":"comment","z":"e797c1b28720f32d","name":"Set Status and Modes","info":"","x":120,"y":200,"wires":[]},{"id":"d8485bfd0aeda5c7","type":"comment","z":"e797c1b28720f32d","name":"ArmMode link","info":"","x":630,"y":260,"wires":[]},{"id":"a7e95831.3143b8","type":"mqtt-broker","name":"MasterPi","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":false,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""},{"id":"6d000dde.3e0b04","type":"server","name":"Home Assistant","version":5,"addon":true,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":false,"cacheJson":true,"heartbeat":false,"heartbeatInterval":30,"areaSelector":"friendlyName","deviceSelector":"friendlyName","entitySelector":"friendlyName","statusSeparator":"at: ","statusYear":"hidden","statusMonth":"short","statusDay":"numeric","statusHourCycle":"h23","statusTimeFormat":"h:m","enableGlobalContextStore":true}]`

ikke-zelf, 2024-01-15:
probably this is better.... the damn editor removes a backslash in front of the v for value

kbgators, 2024-01-15:
got the .json file to work.   imported to Node Red and changed the mqtt broker, the keypad name, and mode to test my flow to "arm_day_zones"
[image]

I ran that flow and it beeped the keypad. But no color change.  Grrr.

here is the log in Z2M after executing your flow:

Debug 2024-01-15 16:07:37Received MQTT message on 'zigbee2mqtt/Alarm_Keypad_Bedroom/set' with data '{"arm_mode":{"mode":"arm_day_zones","transaction":23}}'
Debug 2024-01-15 16:07:37Publishing 'set' 'arm_mode' to 'Alarm_Keypad_Bedroom'
Debug 2024-01-15 16:07:37Received MQTT message on 'zigbee2mqtt/Alarm_Keypad_Bedroom/set' with data '{"arm_mode":{"mode":"arm_day_zones"}}'
Debug 2024-01-15 16:07:37Publishing 'set' 'arm_mode' to 'Alarm_Keypad_Bedroom'

kbgators, 2024-01-15:
@ikke-zelf - This device.  It shows 3400 on the back.   When I paired it in Z2M, it calls it 3400-D
[image]

[image]

Thanks for continuing to help!

ikke-zelf, 2024-01-15:
Then most likely that's the issue.
I am using the XHK1-EU keypad which I know i supported also with setting the lights.
I think only @Koenkk can tell us if yours is also supported with status lights.

vjonas, 2024-03-04:
> Then most likely that's the issue. I am using the XHK1-EU keypad which I know i supported also with setting the lights. I think only @Koenkk can tell us if yours is also supported with status lights.

How did you manage to pair the device to z2m?
I did all the steps, the tamper button is blinking 3 times, then a pause, then again 3 times.
z2m is on and in ‘enable join’ mode. still nothing is showing up on z2m. the device is not pairing.
What am I missing?

vjonas, 2024-03-05:
For some reason, the sonoff dongle P does not pick up the keypad to connect.
I tried via zha and a conbee II and it worked...