Koenkk/zigbee2mqtt #10505: [Problem]: Non-sensical (?) constantly sent from one device after update

SHxKM, 2022-01-02:
### What happened?

After updating the firmware for a Philips Hue Single Pilar light (I have 3 of them), when I turn on the debug logs I see these messages every 30 seconds:

> Received Zigbee message from 'up_ph_pillar_single_ceiling_1', type 'attributeReport', cluster 'genBasic', data '{"32":"0:PWRON@0","33":10046}' from endpoint 11 with groupID null

### What did you expect to happen?

Not to see this weird payload

### How to reproduce it (minimal and precise)

I'm not sure.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

0x26580700

### Adapter

Conbee II

### Debug log

> Received Zigbee message from 'up_ph_pillar_single_ceiling_1', type 'attributeReport', cluster 'genBasic', data '{"32":"0:PWRON@0","33":10046}' from endpoint 11 with groupID null

SHxKM, 2022-01-02:
I just found out the device supposedly has a bound like this:
[image]

However any attempt to remove this bind fails, either via the UI, or through the MQTT endpoint.

SHxKM, 2022-01-02:
I finally decided to unpair and repair the device. The problem went away. This can be closed I’ll just leave it to stale.

Koenkk, 2022-01-02:
Not sure what happend, seems to be a bug on the device side.

closed 2022-01-02 by Koenkk