Koenkk/zigbee2mqtt #10510: [Problem]: after updating slaesh stick + ha addon, zigbee network notw workiung, lot of errors

kpcz, 2022-01-02:
### What happened?

Upgraded Slashes coordinator to CC2652RB_coordinator_20211217.zip and HA addon to 1.22.2, zigbee switches and lights stopped workint, temperature sensors still gives info

### What did you expect to happen?

work as expected.

### How to reproduce it (minimal and precise)

Update Slaesh stick to latest fw.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

Slash USB stick

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7799116/log.txt)

kpcz, 2022-01-02:
```
Zigbee2MQTT:error 2022-01-02 15:22:18: Failed to configure '0x90fd9ffffe8f1fe5', attempt 3 (Error: Bind 0x90fd9ffffe8f1fe5/1 genOnOff from '901' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))
Zigbee2MQTT:error 2022-01-02 15:22:31: Publish 'set' 'state' to '0x000b57fffea39cee' failed: 'Error: Command 0x000b57fffea39cee/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":10}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 55471 - 1 - 16 - 8 - 11 after 10000ms)'
Zigbee2MQTT:error 2022-01-02 15:24:01: Publish 'set' 'state' to '0x000b57fffea39cee' failed: 'Error: Command 0x000b57fffea39cee/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 55471 - 1 - 20 - 6 - 11 after 10000ms)'
```

Is it safe to downgrade adapter to version that was working?  As it looks like the errors started after upgrading the adapter, not the HA addon. Flashed it via ZigSTar GW Multi tool

kpcz, 2022-01-02:
Reflashed stick with SmartRF FLaSh programmer 2 (as I did previous upgrades), no changes, same errors.

kpcz, 2022-01-02:
Reverted back to before working FW  CC2652RB_coordinator_20211217 and HA addon  1.22.2-1 but still not working and getting the same errors.  Only xiaomi temperature/humidity sensors works. Dont have any clues what broke and how to fix now. Please help.

I am afraid that I would need to repair the devices, as they are around 60..

davidramiro, 2022-01-02:
Same issue here. In addition to the error message @kpcz posted above, I sometimes see this one as well:

`Error: Command 0xec1bbdfffe1c231c/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":0,\"transtime\":5}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SREQ '--> ZDO - extRouteDisc - {\"dstAddr\":65400,\"options\":0,\"radius\":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'","meta":{"friendly_name":"GU10 RGB 1"},"type":"zigbee_publish_error"}`

`0xc7: NWK_TABLE_FULL` seems to be mentioned [here](https://www.zigbee2mqtt.io/guide/faq/#common-error-codes), although I can rule out frequency occupation pretty surely.

I'll do some reverting and report back if I find a working state.

kpcz, 2022-01-02:
Powercycled bulbs, and other devices that works like routers, and looks like the zigbee network slowly comes back to life. I think I read somewhere that after fW upgrade(?) some time must pass so the devices connect/create mesh correctly back again.

ArtJames, 2022-01-02:
How i fxed it:
Update Z2M to 1.22.2 and the 2652P stick to  20211217
After updating just leave Z2M alone for at least one hour.
Send a command now and then, or press a button on a remote.
Powercycle devices if possible.
After a while all is back to normal.

kpcz, 2022-01-02:
Thanks everyone. All working good with latest FW and addon. Sorry for panic :)

closed 2022-01-02 by kpcz

Danit2, 2022-01-02:
I had the same problem. 
You must unplug and replug some router devices. 
After them it works again.