Koenkk/zigbee2mqtt #10511: [Problem]: new "current_level_startup" not reflected in state unless read manually from cluster

SHxKM, 2022-01-02:
### What happened?

The `current_level_startup` attribute (`hue_power_on_behavior`) does not get updated in state unless it is explicitly read.

### What did you expect to happen?

Especially if I update this attribute, I want the new value to be available in state so I can act on it _programatically_.

### How to reproduce it (minimal and precise)

- I've set the `hue_power_on_behavior` of a Philips light to `recover`.
- I then sent a `get` command, and got this reply: (no `level_config` at all)
```
MQTT publish: topic 'zigbee2mqtt/down_ph_centura_1', payload '{"brightness":84,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_mode":"color_temp","color_temp":366,"color_temp_startup":65535,"linkquality":215,"state":"OFF","update":{"state":"idle"},"update_available":false}'
```
- I then tried reading the attribute from HA like this and got `None`:
```
{{ states.light.down_ph_centura_1.attributes['level_config']}}
```
- I then went to the device in Z2M UI, went to endpoint `11`, cluster `LevelCtrl`, attribute `startUpCurrentLevel`, hit "Read" and got:
```
Read result of 'genLevelCtrl': {"startUpCurrentLevel":255}
```

Subsequently, this attribute was now available in the Z2M `State` tab, in Home Assistant, and in the `get` responses sent to MQTT, here's a response (this is actually a payload of another device, I'm not reading the cluster for the device above on purpose because I want to get it to work programatically):

```
MQTT publish: topic 'zigbee2mqtt/bedroom_ceiling', payload '{"brightness":254,"color_mode":"color_temp","color_temp":366,"color_temp_startup":65535,"level_config":{"current_level_startup":"previous"},"linkquality":167,"state":"OFF","update":{"state":"idle"},"update_available":false}'
```

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

0x26580700

### Adapter

Conbee II

### Debug log

> MQTT publish: topic 'zigbee2mqtt/down_ph_centura_1', payload '{"brightness":84,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_mode":"color_temp","color_temp":366,"color_temp_startup":65535,"linkquality":215,"state":"OFF","update":{"state":"idle"},"update_available":false}'

SHxKM, 2022-01-02:
It's possible to refresh the state with:

```
zigbee2mqtt/device_name/get/level_config
```

SHxKM, 2022-01-05:
@Koenkk even though I found a way it's a bit cumbersome to call this endpoint repeatedly. Will Z2M update HA of this attribute change if we don't call `zigbee2mqtt/device_name/get/level_config`? right now even Z2M UI is not updating this value if we don't call `zigbee2mqtt/device_name/get/level_config`

Koenkk, 2022-01-05:
@SHxKM it seems https://github.com/Koenkk/zigbee-herdsman-converters/blob/dbd523def852a9ab902e43001423c5f9b58aa8dc/converters/toZigbee.js#L526 returns the wrong state, could you debug this with some `console.log` to see what goes wrong?

SHxKM, 2022-01-05:
> @SHxKM it seems https://github.com/Koenkk/zigbee-herdsman-converters/blob/dbd523def852a9ab902e43001423c5f9b58aa8dc/converters/toZigbee.js#L526 returns the wrong state, could you debug this with some `console.log` to see what goes wrong?

I'm not sure how this can be done without installing Z2M in a separate environment or installing it locally, which I can't afford to do right now. So I can only guess semi-blindly after trying to understand the code:

1. The `hue_power_on_behavior` segment is the only one that doesn't have a `convertGet`, I wonder why..(and not sure at all it has anything to do with the problem)
2. Maybe something with the disparity between `recover` (what we send in the update) and `previous` (what gets displayed).

I'll keep looking.

SHxKM, 2022-01-05:
I think I may be onto something, it may be a short in the dark but what the hell.

When we update "hue_power_on_behavior", we reach its key [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/dbd523def852a9ab902e43001423c5f9b58aa8dc/converters/toZigbee.js#L1837). This method, _unlike_ others (like [`hue_motion_sensitivity`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/dbd523def852a9ab902e43001423c5f9b58aa8dc/converters/toZigbee.js#L1941), [`RTCGQ13LM_motion_sensitivity`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/dbd523def852a9ab902e43001423c5f9b58aa8dc/converters/toZigbee.js#L1965), [`hue_motion_led_indication`](https://github.com/Koenkk/zigbee-herdsman-converters/blob/dbd523def852a9ab902e43001423c5f9b58aa8dc/converters/toZigbee.js#L1952)), does not return anything.

So @Koenkk  the problem may not be with the `get` (because it clearly works and refreshes the state), but with the set that doesn't. I'm not sure at all since I'm still learning the very basics of how Z2M works.

Koenkk, 2022-01-06:
> but with the set that doesn't return anything

That is indeed a problem, to return an (optimistic) state see e.g. https://github.com/Koenkk/zigbee-herdsman-converters/blob/dbd523def852a9ab902e43001423c5f9b58aa8dc/converters/toZigbee.js#L1941 , anything that is returned here will be published to MQTT

SHxKM, 2022-01-06:
So just so I understand, you’re saying returning something would be a problem? Or that we found the culprit? If so, is it planned to be fixed?

Koenkk, 2022-01-06:
should be fixed now, whenever `hue_power_on_behavior` is set it will be reflected in the published MQTT payload

SHxKM, 2022-01-06:
> should be fixed now, whenever `hue_power_on_behavior` is set it will be reflected in the published MQTT payload

Are you sure this will work? When the value == “recover” we don’t seem to manipulate it at all when we publish the MQTT payload? And we overwrite it sometime later if another property is changed…(again, I could be wrong here).

Koenkk, 2022-01-07:
It will just publish to MQTT whatever you set to `hue_power_on_behavior`, so if you e.g. set `{"hue_power_on_behavior": "recover"}` it will publish that to zigbee2mqtt/MY_DEVICE

SHxKM, 2022-01-07:
> It will just publish to MQTT whatever you set to `hue_power_on_behavior`, so if you e.g. set `{"hue_power_on_behavior": "recover"}` it will publish that to zigbee2mqtt/MY_DEVICE

But I guess that will not update `level_config/startup_current_level`? So we’d have to read it manually still? Or will that be called subsequently? 

My goal is to have state[‘level_config’][‘startup_current_level’] refresh properly after an update, without having to read it explicitly.

Koenkk, 2022-01-07:
Could you provide me a debug log when setting `level_config/startup_current_level`?

SHxKM, 2022-01-07:
> Could you provide me a debug log when setting `level_config/startup_current_level`?

Hmm, well I am never really setting that one. Not directly at least. It’s not documented. 

The docs say you can send `hue_power_on_behavior: “recover”` to alter that specific behavior, and I coincidentally noticed that after that payload is sent, both the Z2M device state and the HA state have this `level_config: {startup_current_level: previous}`, but **only** after a GET is sent. 

So flow is: 

- send the update with payload to hue_power_on_behavior
- send GET to device/level_config
- both Z2M state and HA state now show this attribute 

My whole goal and only reason I got here is that I wrote a cron script that tries to find out whether ANY Hue light is not configured with this setting (previous/recover), and if such a device is found, update it to this setting. It runs on a schedule so if devices lose this setting we can re-apply it. 

Anyway, I’m still confused now. It seems like we have different endpoints for writing and reading the same attr? Or two attributes holding the same information? Or maybe one attribute  is specific and the other isn’t?

Koenkk, 2022-01-07:
`hue_power_on_behavior` is specific while `startup_current_level` is a generic zigbee thing, I wasn't aware that setting the `hue_power_on_behavior` influences the `startup_current_level`.

> It runs on a schedule so if devices lose this setting we can re-apply it.

Why just not rely on the `hue_power_on_behavior` attribute in the payload and check if it is set to recover?

SHxKM, 2022-01-07:
> `hue_power_on_behavior` is specific while `startup_current_level` is a generic zigbee thing, I wasn't aware that setting the `hue_power_on_behavior` influences the `startup_current_level`.

It seems like it does because they read from the same cluster..?

> > It runs on a schedule so if devices lose this setting we can re-apply it.
> 
> Why just not rely on the `hue_power_on_behavior` attribute in the payload and check if it is set to recover?

Which payload? The “set” does/did not return any state, and I didn’t even know we could do a get on hue_power_on_behavior.

Koenkk, 2022-01-07:
> This attribute isn’t on any of my devices’ state. It’s simply not there.

I've added this in https://github.com/Koenkk/zigbee-herdsman-converters/commit/f2e0815fc8e807a9bbf8076537fc015f47bb82e5 so this is only available in the dev branch yet. Note that you will need to set it again in order for this to appear

SHxKM, 2022-01-07:
> > This attribute isn’t on any of my devices’ state. It’s simply not there.
> 
> I've added this in [Koenkk/zigbee-herdsman-converters@f2e0815](https://github.com/Koenkk/zigbee-herdsman-converters/commit/f2e0815fc8e807a9bbf8076537fc015f47bb82e5) so this is only available in the dev branch yet. Note that you will need to set it again in order for this to appear

Got it. But am I right that this will only “refresh” if I read/write it? Meaning: there is no periodic refresh or something (say on restart) of this cluster, so if for some reason Z2M and the device lose sync, I will either have to SET or GET (explicitly) to find out?

Koenkk, 2022-01-07:
> But am I right that this will only “refresh” if I read/write it? Meaning: there is no periodic refresh or something (say on restart) of this cluster, so if for some reason Z2M and the device lose sync, I will either have to SET or GET (explicitly) to find out?

The state is updated when you read/write or when the device report the attributes itself. It is not polled

> also hate to say this after your hard work but I think we should NOT publish power_on_behavior if we already have level_config - don’t you agree?

the `hue_power_on_behavior` could contain more than the brightness, e.g. `hue_power_on_color_temperature` and `hue_power_on_color`

SHxKM, 2022-01-07:
Fair enough. Appreciate your hard work. I guess I should set-up reporting (though last I tried it said UNREPORTABLE or something.

closed 2022-01-07 by SHxKM

SHxKM, 2022-01-07:
@Koenkk would you be willing to also implement a convertGet for hue_power_on_behavior? I can try that myself (probably tomorrow) if you’re too busy.

Koenkk, 2022-01-07:
@SHxKM would be good if you try, the convertget is a mirror of the converset, it should read all the attributes that are written in the convertset

SHxKM, 2022-01-08:
> @SHxKM would be good if you try, the convertget is a mirror of the converset, it should read all the attributes that are written in the convertset

Will do today. Is the endpoint created automagically? (device/get/power_on_behavior)

Koenkk, 2022-01-08:
> Is the endpoint created automagically? (device/get/power_on_behavior)

yes, adding the convertget is enough