Koenkk/zigbee2mqtt #10514: 1.22.2-dev: SJCGQ11LM - "water_leak": null

zibous, 2022-01-02:
### What happened?

I get different MQTT messages with the SJCGQ11LM. One time "water_leak" is: null and the tamper attribute is missing. 

### What did you expect to happen?

Inital, if no water_leak is reported, then "water_leak" is: NULL 

### How to reproduce it (minimal and precise)

restart zigbee2mqtt

```json
{
    "battery": 100,
    "battery_low": null,
    "device": {
        "applicationVersion": 4,
        "dateCode": "20170721",
        "friendlyName": "waterleak-bad",
        "hardwareVersion": 2,
        "ieeeAddr": "0x00158d00028208ae",
        "manufacturerID": 4151,
        "manufacturerName": "LUMI",
        "model": "SJCGQ11LM",
        "networkAddress": 14883,
        "powerSource": "Battery",
        "softwareBuildID": "3000-0001",
        "stackVersion": 2,
        "type": "EndDevice",
        "zclVersion": 1
    },
    "elapsed": 3022520,
    "last_seen": "2022-01-02T13:54:08+01:00",
    "linkquality": 167,
    "temperature": 31,
    "voltage": 3005,
    "water_leak": null
}
```

```json
{
    "battery": 86,
    "battery_low": false,
    "device": {
        "applicationVersion": 4,
        "dateCode": "20170721",
        "friendlyName": "waterleak-washer",
        "hardwareVersion": 2,
        "ieeeAddr": "0x00158d00028079c0",
        "manufacturerID": 4151,
        "manufacturerName": "LUMI",
        "model": "SJCGQ11LM",
        "networkAddress": 60332,
        "softwareBuildID": "3000-0001",
        "type": "EndDevice"
    },
    "elapsed": 1569521,
    "last_seen": "2022-01-02T13:49:12+01:00",
    "linkquality": 183,
    "tamper": false,
    "temperature": 25,
    "voltage": 2975,
    "water_leak": false
}
```

### Zigbee2MQTT version

1.22.2-dev 

### Adapter firmware version

0x26660700

### Adapter

ConBee2/RaspBee2

### Debug log

[image]

[image]

Koenkk, 2022-01-02:
The sensors needs to be triggered at least once to have a `water_leak` value.

jalalhobbs, 2022-01-03:
Could this be something that zigbee2mqtt can handle as it does the temperature calibration?

closed 2022-02-11 by github-actions[bot]