Koenkk/zigbee2mqtt #10516: [New device support]:

jebabi, 2022-01-02:
### Link

https://www.lidl.dk/p/p55665

### Database entry

{"id":50,"type":"Router","ieeeAddr":"0x60a423fffe69fe82","nwkAddr":39252,"manufId":4098,"manufName":"_TZ3000_kohbva1f","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3000_kohbva1f","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":2,"currentX":0,"currentY":0,"currentSaturation":254,"colorTemperature":454,"currentHue":0}},"genLevelCtrl":{"attributes":{"currentLevel":140}},"genOnOff":{"attributes":{"onOff":1}},"msTemperatureMeasurement":{"attributes":{}},"genAnalogValue":{"attributes":{}},"genMultistateOutput":{"attributes":{}},"greenPower":{"attributes":{}},"genTime":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1641138038854}

### Comments

Vendor: `Lidl`
Model: `14153706L`
Description: `Livarno Smart LED Ceiling Light`
It's the round version from the provided link.

The fingerprint matches [this bulb from Aldi](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/aldi.js#L16) and I can't figure out how to add support when the fingerprint is already taken.

The current match fails due to the attribute `enhancedCurrentHue` which should be `currentHue`.

### Supported color modes

brightness, color_temp, color_xy

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

Koenkk, 2022-01-02:
@BMOD89  could you confirm that the Aldi L122CB63H11A9.0W uses `_TZ3000_kohbva1f` as the manufacturer name?

BMOD89, 2022-01-03:
@Koenkk unfortunately I cannot find that bulb anymore... 
The bulb was working with that settings, so I guess it is correct.  
But another hint could be the log in that issue https://github.com/Koenkk/zigbee2mqtt/issues/6340. 
So it seems that both devices have the same model ID as manufacturer name...

Koenkk, 2022-01-03:
@BMOD89 thanks

@jebabi so unfortunately we cannot distinguish these lights. I've added it to the generic TuYa TS0505B instead and marked both the Aldi and Lidl as a whitelabel (so they will appear as such on the supported device page)

jebabi, 2022-01-03:
@Koenkk thank you for looking into this issue.

Just to be sure, the best way for me to add support for this device is by using an external converter, correct?

Koenkk, 2022-01-03:
@jebabi this device is already supported, there is no need to add an external converter. (it should work out-of-the-box despite the wrong model/description)

jebabi, 2022-01-03:
@Koenkk It's not working as expected.

`color_temp` only changes the brightness, not the temperature. And `brightness` seems to be a percentage of the brightness level set by `color_temp`. For instance  `color_temp`: 470 and `brightness`: 254 is approximately the same brightness and temperature as `color_temp`: 160 and `brightness`: 10

When setting `color_xy` only half the lamp is in the right color.

Koenkk, 2022-01-03:
@jebabi I see, that is bad news since the device doesn't use the standard commands then.

Can you try with the following external converter?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0505B'],
    model: '14153706L',
    vendor: 'Lidl',
    description: 'Smart LED Ceiling Light',
    exposes: [e.light_brightness_colortemp()],
    fromZigbee: [fz.on_off, fz.tuya_led_controller],
    toZigbee: [tz.tuya_led_controller, tz.ignore_transition, tz.ignore_rate],
};

module.exports = definition;

```

jebabi, 2022-01-03:
@Koenkk I'm using your changes from [this commit](https://github.com/Koenkk/zigbee-herdsman-converters/commit/7d74703684a39e33e71f72f86b22711e9a2435b1) (btw there is an extra `,` in line 172)

Added the external converter to `/app/data/14153706L.js` and enabled it in `configuration.yaml`

```
external_converters:
  - 14153706L.js
```

1. Removed the device
2. Restarted Zigbee2MQTT
3. Paired the device

It shows up as:
Manufacturer TuYa
Model TS0505B

Same issues as described above.

Koenkk, 2022-01-04:
@jebabi then I'm afraid you will have to reverse engineer how to control this bulb using the original TuYa gateway; some resources which can help:
- https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531
- https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

jebabi, 2022-01-04:
@Koenkk That's okay, I assumed as much going into this.

Do I need the gateway Lidl sells og will any TuYa gateway work? I'm not really into how apps and gateways from different vendors pair up.

Thank you so much for the support.

Koenkk, 2022-01-05:
@jebabi I believe any tuya gateway should work.