Koenkk/zigbee2mqtt #10517: [Problem]: GS361A-H04: Randomly 'MAC transaction expired' (240) and 0x10: MEM_ERROR after some days

alexanderwwagner, 2022-01-02:
### What happened?

In my smarthome I have installed two GS361A-H04/Essentials 120112 (Siterwell).
https://www.zigbee2mqtt.io/devices/GS361A-H04.html

They are connected via zigbee to my raspberrypi (CC2531 USB Stick).
Right after the pairing all is working as excpected
- get all actualvalues
- can set all setvalues via zigbee2mqtt

But after some time. (Sometimes four days sometimes one day) the two devices are only readonly.
If I try to set the temperature over zigbee2mqtt there I can see one of this errors:

`Zigbee2MQTT:error 2022-01-02 14:32:12: Publish 'set' 'current_heating_setpoint' to 'a1_therm_officestudio_NR4' failed: 'Error: Command 0x086bd7fffec83e29/1 manuSpecificTuya.setData({"status":0,"transid":1,"dp":2,"datatype":2,"length_hi":0,"length_lo":4,"data":[0,0,0,105]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> AF - dataRequest - {"dstaddr":26701,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":107,"options":0,"radius":30,"len":13,"data":{"type":"Buffer","data":[17,5,0,0,1,2,2,0,4,0,0,0,105]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)'))'`

`Zigbee2MQTT:error 2022-01-02 14:33:04: Publish 'set' 'current_heating_setpoint' to 'a2_therm_officestudio_NR4' failed: 'Error: Command 0x14b457fffe3cb20e/1 manuSpecificTuya.setData({"status":0,"transid":0,"dp":2,"datatype":2,"length_hi":0,"length_lo":4,"data":[0,0,0,105]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'`

To solve this problem the only way is to poweroff the two devices and to do a repairing of the devices.

To improve the stability of the zigbee network I connected the CC2531 with a USB extension cable and Changed channel to 20. Changed pan_id to 6755 and did a repairing of all devices.
This was described in a other issue... but that solved my problem not.
https://github.com/Koenkk/zigbee2mqtt/issues/6177

My configuration.yml
`homeassistant: false
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost
serial:
  port: /dev/ttyACM0
advanced:
  pan_id: 6755
  channel: 20
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
device_options:
  legacy: false
devices:
  '0x00158d000486e53b':
    friendly_name: s_eco_officestudio_NR4
  '0x00158d000548a10d':
    friendly_name: s_eco_livingroom_NR1
  '0x00158d0006e48f29':
    friendly_name: s_eco_bathroom_NR2
  '0x00158d000533c4f4':
    friendly_name: s_eco_sleepingroom_NR3
  '0x00158d0006e3b65e':
    friendly_name: s_eco_garden_NR6
  '0x00158d0003d4e721':
    friendly_name: s_window_1_officestudio_NR4
  '0x14b457fffe3cb20e':
    friendly_name: a2_therm_officestudio_NR4
  '0x086bd7fffec83e29':
    friendly_name: a1_therm_officestudio_NR4
  '0x0017880109a7c011':
    friendly_name: ms_hue_livingroom_NR1
  '0x0c4314fffedf85e3':
    friendly_name: a1_led_livingroom_NR1
  '0x04cf8cdf3c802a87':
    friendly_name: a1_switch_cleaner_NR3`

- There are 5 aquara humidity sensors
- There are 2 essential thermostats (GS361A-H04)
- There  is  1 aquara window sensor
- There  is  1 Xiaomi Mi power plug (router)
- There  is  1 Philips Hue Wireless Dimming switch
- There is 1 innr led (Router) (not connected)

Because I use docker-compose with the official zigbee2mqtt docker image I do not know how to enable 
the herdesmann debugging with docker-compose...?
https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging

Do you need some other informations from me?

### What did you expect to happen?

Set the temperature values without any problems.

### How to reproduce it (minimal and precise)

Look at: What happened

### Zigbee2MQTT version

koenkk/zigbee2mqtt:1.22.1

### Adapter firmware version

20190608

### Adapter

CC2531 ZigBee USB-Stick

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7799445/log.txt)

Koenkk, 2022-01-02:
`MEM_ERROR` is a sign of corruption on the coordinator side, since you are using a CC2531 I highly recommend to update to one of the recommended coordinators: https://www.zigbee2mqtt.io/guide/adapters/#recommended (the CC2531 is a weak coordinator)

alexanderwwagner, 2022-01-04:
Thank you for your fast answer.
I orderd now a recommended coordinator... I hope the delivery time is not very long because there are some chip problems...
I will give here an answer if I tested it with the new coordinator!

The most time there I can see the "Data request failed with error: 'MAC transaction expired' (240)" error and not the "Mem_error". But maybe the Mac transaction error is a result of the mem error?

Is there any way how I can enable the Zigbee-herdsman debugging through a docker-compose.yml file?
Through my OS Its not just possible to have access to the docker run command...
https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging

alexanderwwagner, 2022-01-21:
In the meantime I have the "Electrolama zig-a-zig-ah!" for about one week in operation.
The first days all worked fine, but today I I can not set the temperature via zigbee2mqtt because there are two errors. (One for each device)

`greencube_zigbee2mqtt  Zigbee2MQTT:error 2022-01-21 08:17:51: Publish 'set' 'current_heating_setpoint' to 'a2_therm_officestudio_NR4' failed: 'Error: Command 0x14b457fffe3cb20e/1 manuSpecificTuya.dataRequest({"seq":46,"dp":2,"datatype":2,"length_hi":0,"length_lo":4,"data":[0,0,0,205]}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
 greencube_zigbee2mqtt  Zigbee2MQTT:error 2022-01-21 08:17:54: Publish 'set' 'current_heating_setpoint' to 'a1_therm_officestudio_NR4' failed: 'Error: Command 0x086bd7fffec83e29/1 manuSpecificTuya.dataRequest({"seq":49,"dp":2,"datatype":2,"length_hi":0,"length_lo":4,"data":[0,0,0,220]}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'`

There are some of this errors too:
https://github.com/Koenkk/zigbee2mqtt/issues/9477
`greencube_zigbee2mqtt  zigbee-herdsman-converters:tuyaThermostat: Unrecognized DP #17 with data {"seq":2048,"dp":17,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[0]}}
 greencube_zigbee2mqtt  zigbee-herdsman-converters:tuyaThermostat: Unrecognized DP #17 with data {"seq":2048,"dp":17,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[0]}}
 greencube_zigbee2mqtt  zigbee-herdsman-converters:tuyaThermostat: Unrecognized DP #19 with data {"seq":2304,"dp":19,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[0]}}
 greencube_zigbee2mqtt  zigbee-herdsman-converters:tuyaThermostat: Unrecognized DP #19 with data {"seq":2304,"dp":19,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[0]}}
 greencube_zigbee2mqtt  zigbee-herdsman-converters:tuyaThermostat: Unrecognized DP #19 with data {"seq":2304,"dp":19,"datatype":4,"fn":0,"data":{"type":"Buffer","data":[0]}}`

(koenkk/zigbee2mqtt:1.22.1/
 Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20211217,"transportrev":2},"type":"zStack3x0"}')

At the moment I don´t get it to work.
Repairing of the devices does not help!
Do you have an idea what the problem could be?

alexanderwwagner, 2022-01-22:
Reflashing the coordinator did not help.
I used the default firmware of the zigbee2mqtt website.

Hmm what can I do to get it working again?
And what I have to do, to have a stable zigbee network?

alexanderwwagner, 2022-01-23:
I summerize:
The problem of this issue I could solve through the notice of Koenkk to use a other coordinator.
I switched from CC2531 to Electrolama zig-a-zig-ah!

Since I use the new coordinator the two errors never come back to me:
- MAC transaction expired (240) 
- MEM_Error 

Thanks to Koenkk!
For the new problem with the "no network route" I will open a new issue, because that has nothing to do with my first problem.

closed 2022-01-23 by alexanderwwagner