Koenkk/zigbee2mqtt #10526: [New device support]: Nue / 3A Single Power Outlet with 2 USB Ports

bigfeetneedbigboots, 2022-01-02:
### Link

https://www.lectory.com.au/NUE-Smart-ZigBee-USB-Ports-Power-Point-Wall-Mount-Single-Power-Point-2-USB-Ports

### Database entry

{"id":61,"type":"Router","ieeeAddr":"0x00158d0001769f7b","nwkAddr":9985,"manufId":4478,"manufName":"3A Smart Home DE","powerSource":"Mains (single phase)","modelId":"LXN56-0S27LX1.1","epList":[11],"endpoints":{"11":{"profId":49246,"epId":11,"devId":0,"inClusterList":[0,4,3,6,5,4096,8],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LXN56-0S27LX1.1","manufacturerName":"3A Smart Home DE","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190312","swBuildId":"1000-0001"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20190312","swBuildId":"1000-0001","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1641156803267,"useImplicitCheckin":true}

### Comments

I've created an external converter file for this device and it is working correctly based on my testing. I am not confident doing a PR etc to make it available to others so asking for help with that part
[nue3a1po.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/7799907/nue3a1po.js.txt)
.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LXN56-0S27LX1.1'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'NUE-AUWZO1',
    vendor: 'Nue / 3A',
    description: 'Smart Zigbee 1-gang Power Outlet with 2 USB',
    extend: extend.switch()
};

module.exports = definition;
```

Koenkk, 2022-01-03:
Do you get state updates when controlling the switch physically? If not try with the following below `extend`:

```js
configure: async (device, coordinatorEndpoint, logger) => {
    const endpoint = device.getEndpoint(11);
    await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
    await reporting.onOff(endpoint);
},
```

bigfeetneedbigboots, 2022-01-03:
And that's why you're a god amongst mere Zigbee mortals like me @Koenkk! Thanks mate. 

Adding the lines of code you suggested below `extend` worked perfectly and the use case you raised now works.

Koenkk, 2022-01-04:
It seems this device is already supported out-of-the-box (recognised as a HGZB-20-UK power plug), can you confirm it works correclty?

bigfeetneedbigboots, 2022-01-05:
Yes @Koenkk, it turns on/off from MQTT commands and reflects back states to MQTT when turned on/off physically when I let the device set itself up as a HGZB-20-UK power plug. Functionally it is fine, aesthetically it is an incorrect model, description and picture. Is 3A using the same Zigbee model numbers ("LXN56-0S27LX1.1" in this case) for different types of devices?

Koenkk, 2022-01-05:
>  Is 3A using the same Zigbee model numbers ("LXN56-0S27LX1.1" in this case) for different types of devices?

Yes they do, meaning we cannot distinguish them. I believe you can add an `icon` property  (e.g. below `model`) to your external converter with an url to the image

closed 2022-01-27 by bigfeetneedbigboots